FreeCAD/FreeCAD #7878: [Issue]

johannespfister, 2022-11-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=73418&p=641076

### Version

0.20 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods: 
  * Assembly4
  * sheetmetal
```

### Subproject(s) affected?

OpenSCAD

### Issue description

I have a OpenSCAD File with 2 Modules (and some submodules in this modules) that works in OpenSCAD and FreeCAD 0.19.
But when i open them in FreeCAD 0.20.1 or 0.21, one of the Modules is not correctly placed next to the other Module, as it is in FreeCAD 0.19.

Steps to reproduce:
- Use a PC with Freecad 0.20.1 or 0.21 installed and OpenSCAD installed.
- Open FreeCAD 0.20.1 or 0.21.
- Create a new Document.
- Import the attached OpenSCAD File.
- See that there are 2 Modules which are away from each other (way more than 20mm).
- Modify the Attached OpenSCAD remove one of the sections with the comment "remove ... to avoid the bug" (choose any of them).
- Create a new Document in FreeCAD.
- Import the modified OpenSCAD.
- See that the Position of the 2 Modules is different. They should be close to each other (about 20mm space in between them).

Alternatively, you can open the same file in FreeCAD 0.19 or OpenSCAD without modifying it, and you should see that the 2 Modules are near (about 20mm) to each other.

### Anything else?

[OpenSCAD File with this error](https://forum.freecadweb.org/download/file.php?id=207086&sid=bdb1f23a9f30189a47f783a0b771cd10)

OpenSCAD version:
```
OpenSCAD Version: 2021.01
System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 4 CPUs 7.70 GB RAM
User Agent: OpenSCAD/2021.01 (Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833)
Compiler: GCC "7.4.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.3.7
CGAL version, kernels: 4.11, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.13.0
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.2
fontconfig version: 2.13.1
freetype version: 2.10.1
harfbuzz version: 2.4.0
cairo version: 1.16.0
lib3mf version: 1.8.0
Application Path: C:/Program Files/OpenSCAD
Documents Path: C:\Users\Urs Jost\Documents
User Documents Path: C:\Users\Urs Jost\Documents
Resource Path: C:/Program Files/OpenSCAD
User Library Path: C:/Users/Urs Jost/Documents/OpenSCAD/libraries
User Config Path: C:\Users\Urs Jost\AppData\Local/OpenSCAD
Backup Path: C:/Users/Urs Jost/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
C:/Users/Urs Jost/Documents/OpenSCAD/libraries
C:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
C:/WINDOWS/fonts
C:/Users/Urs Jost/.local/share/fonts
C:/Users/Urs Jost/.fonts
GLEW version: 1.12.0
OpenGL Version: 4.0.0 - Build 10.18.10.4425
GL Renderer: Intel(R) HD Graphics 4000
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QGLWidgetGL Extensions:
GL_3DFX_texture_compression_FXT1
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_draw_indirect
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robustness
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counters
GL_ARB_shader_bit_encoding
GL_ARB_shader_objects
GL_ARB_shader_precision
GL_ARB_shader_subroutine
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_tessellation_shader
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_compiled_vertex_array
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shader_integer_mix
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_buffer
GL_EXT_texture_compression_s3tc
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_texture_swizzle
GL_EXT_transform_feedback
GL_IBM_texture_mirrored_repeat
GL_INTEL_map_texture
GL_INTEL_performance_queries
GL_INTEL_performance_query
GL_KHR_blend_equation_advanced
GL_KHR_debug
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_ARB_buffer_region
WGL_ARB_create_context
WGL_ARB_create_context_profile
WGL_ARB_create_context_robustness
WGL_ARB_extensions_string
WGL_ARB_framebuffer_sRGB
WGL_ARB_make_current_read
WGL_ARB_multisample
WGL_ARB_pbuffer
WGL_ARB_pixel_format
WGL_ARB_pixel_format_float
WGL_EXT_create_context_es2_profile
WGL_EXT_create_context_es_profile
WGL_EXT_depth_float
WGL_EXT_extensions_string
WGL_EXT_pixel_format_packed_float
WGL_EXT_swap_control
WGL_EXT_swap_control_tear
WGL_NV_DX_interop
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

KeithSloan, 2022-11-25:
Seems to be okay with the Alternate_Openscad_Importer https://github.com/KeithSloan/OpenSCAD_Alt_Import
[image]

FlachyJoe, 2022-11-26:
@KeithSloan can you please assign a descriptive title to this issue ?

KeithSloan, 2022-11-26:
@FlachyJoe Not my issue, was just testing with the Alternate OpenSCAD importer.

FlachyJoe, 2022-11-26:
@KeithSloan whoops !
@johannespfister can you please update the title ?

chennes, 2022-11-30:
It looks like the final grouping, "Group001", is getting a placement put on it, and it shouldn't. If I zero out that transform everything looks fine.

chennes, 2022-12-01:
Fixed by 3f118b52bc62db6a5ca85daafaa2c021411e7177.

closed 2022-12-01 by chennes

johannespfister, 2022-12-01:
@chennes Thank you for fixing it.