FreeCAD/FreeCAD #7880: [Issue] PartDesign pad in custom direction fail in UpToFace case

FlachyJoe, 2022-11-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=73674&p=642646#p642646

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.1 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.31097 (Git)
Build type: Unknown
Branch: master
Hash: c23ee7ad88433a0f00e84afae0a62fe738be0c09
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: French/France (fr_F
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

Error "SketchBased: Up to face: Must not be parallel to extrusion direction!" is raised when the base Sketch is perpendicular to the chosen end face even if the custom direction is set.

The axis parallelism check doesn't take care of the direction of extrusion.

I propose a fix here: https://github.com/FreeCAD/FreeCAD/pull/7879

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-12-04:
Closing per #7879

FlachyJoe, 2022-12-04:
#7943 invalids the forum test file.
This issue is not solved yet.

luzpaz, 2023-01-15:
Any more traction on this issue?

FlachyJoe, 2023-01-15:
Fixed by #8026

FlachyJoe, 2023-01-15:
Thank you @luzpaz to wake me up :smile:

closed 2023-01-15 by FlachyJoe