FreeCAD/FreeCAD #7881: Curves WB Interpolate Failed

cropduster32, 2022-11-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 20.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Curves 0.6.1
  * Pyramids-and-Polyhedrons
[/code]
```

### Problem description

Trying to create a curve around a simple sketch with no result. Report view shows:
Running the Python command 'Interpolate' failed:
Traceback (most recent call last):
  File "/home/brianw/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/interpolate.py", line 339, in Activated
    source = self.parseSel(s)
  File "/home/brianw/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/interpolate.py", line 322, in parseSel
    sen = obj.SubElementNames

'Sketcher.SketchObject' object has no attribute 'SubElementNames'

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2022-11-26:
Hi,
can you share the problematic file or at least describe your workflow ?

cropduster32, 2022-11-26:
Yes, I was following the MangoJelly video 'Curves Workbench 16: Boat / Ship building with Interpolate Curves Possible solution?' and got to the point of using the Interpolate Curve tool and go the message in Report View shown above. I can zip the file here if necessary but as there are only a few lines it's probably not worth it.

FlachyJoe, 2022-11-26:
[Curves_Interpolate](https://wiki.freecadweb.org/Curves_Interpolate) works as expected here
```
OS: Ubuntu 22.04.1 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.31097 (Git)
Build type: Unknown
Branch: master
Hash: c23ee7ad88433a0f00e84afae0a62fe738be0c09
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: French/France (fr_FR)
Installed mods: 
  * Curves 0.6.1
```

It doesn't seem a code issue but an usage one. Please [explore the forum](https://forum.freecadweb.org/viewtopic.php?f=3&t=22675) and ask there if needed.

luzpaz, 2022-11-28:
Closing this ticket for the time being.

closed 2022-11-28 by luzpaz