FreeCAD/FreeCAD #7882: [Problem] Toolbar customizations problem

mskf1383, 2022-11-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Debian GNU/Linux bookworm/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.1.
Build type: Release
Python 3.10.7, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: Persian/Iran (fa_IR) ]
Installed mods: 
  * kicadStepUpMod.bak
  * pcb
  * kicadStepUpMod
```

### Subproject(s) affected?

Core

### Problem description

I customized toolbar to this:
[image]

But after close and reopening FreeCAD, It became this:
[image]

### Anything else?

My system language is Persian which is RTL (Right-To-Left). Also when I was in GNOME, FreeCAD was LTR anyway. But in KDE is true.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-11-29:
Does the customization never work or is this a one time thing?

mskf1383, 2022-11-29:
It always happen. Also it is important that when I move a toolbox vertical, it save. but when I move it horizontal, It reset at next startup. I think this is a problem with RTL.

> Are other Qt applications working in this regard?

I don't know. QT apps I have are from KDE and don't let user to move toolboxes.

> Is this an installation from the debian repos?

Yes

luzpaz, 2022-11-30:
This is a bigger issue. This needs to be discussed on the UI/UX subforum as a dedicated thread for RTL UI/UX

luzpaz, 2023-03-05:
@mskf1383 can you open a forum thread to generate more exposure for this ticket, please?

mskf1383, 2023-03-06:
I created an account, but after signing in, there is no forum:

[image]

luzpaz, 2023-03-06:
Weird. What username did you choose? Maybe try a hard browser refresh ?

mskf1383, 2023-03-06:
mskf1383
Nothing changed.
I ask in matrix room.

mskf1383, 2023-03-13:
https://forum.freecad.org/viewtopic.php?t=76754

xtemp09, 2023-03-14:
@abdullahtahiriyo, might I ask if you can check Right-to-left User Interface?

By default Qt uses `Qt::LeftToRight`. The [documentation says](https://doc.qt.io/qt-6/qguiapplication.html#layoutDirection-prop) that it's possible to set `Qt::LayoutDirection` to `Qt::LayoutDirectionAuto` with `QGuiApplication::setLayoutDirection()` method and let Qt decide the direction based on the application's language.

My proposition is simple ― just one-line addition to `/src/Main/MainGui.cpp` with the following:

```c++
app.setLayoutDirection ( Qt::LayoutDirectionAuto );
```

I have never used RTL, so I don't know what kind of damage it can cause with different themes and with Python especially that already utilizes `Qt::RightToLeft` in some places.

abdullahtahiriyo, 2023-03-20:
@xtemp09 

Sorry I have no experience in RTL.

mskf1383, 2023-03-23:
> Can someone share a mockup of what a correct RTL UI looks like ?

The problem is that when I start FreeCAD, it is LTR:
[image]

After opening and closing preferences it become RTL (That is correct but has some small problems):
[image]

Fixes (The start page itself should be completely reversed, so I removed it):
[image]

luzpaz, 2023-10-06:
[image]

https://doc.qt.io/qt-6/qt.html#LayoutDirection-enum

luzpaz, 2025-02-01:
@mskf1383 gave us some low hanging fruits to solve in https://github.com/FreeCAD/FreeCAD/issues/7882#issuecomment-1480880664

Anyone want to spearhead this task ?