FreeCAD/FreeCAD #7904: [Issue]

riopar10, 2022-11-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Spain (es_ES)
Installed mods: 
  * Lithophane
[/code]
```

### Subproject(s) affected?

Part

### Issue description

PART - Incorrect export
Hello, good morning, a problem when exporting, whatever the file is, stl, 3mf, etc, always or almost always, but quite frequently, freecad extrudes it well, image 1 but the export does not do it well, image 2 You know, you have proof or the solution to the problem.
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-11-29:
can you share a file where you experienced this problem? are you sure you are selecting _only_ the shape you want to export and not any of the previous shapes?

riopar10, 2022-12-04:
Hi, I'm sorry. not having answered before, for work and also trying possible solutions, but it has worked.
How can I upload .stl and FCstd files, to see the problems, thanks

luzpaz, 2022-12-04:
Put them in zip archives

riopar10, 2022-12-04:
envío los archivos para que los podais examinar, gracias
[Sin título-6.zip](https://github.com/FreeCAD/FreeCAD/files/10149397/Sin.titulo-6.zip)

adrianinsaval, 2022-12-04:
what do you select when exporting? by selecting the correct object I get exactly the expected result when exporting
[Sin título-6-Cut001.zip](https://github.com/FreeCAD/FreeCAD/files/10149737/Sin.titulo-6-Cut001.zip)
see https://forum.freecadweb.org/viewtopic.php?p=170906#p170906
this is most likely just user error and can be closed