FreeCAD/FreeCAD #7910: [Problem] Sketch exportto DXF

ningjiazun, 2022-11-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 1909
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Chinese/China (zh_CN)
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

0.20.1 Export 2D graphics error，
But redrawing the same graphics in 0.19.3 works fine。

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-11-30:
Please add an example file or we will close this ticket. Note: It's best to ask on the forum before opening a ticket.

adrianinsaval, 2022-11-30:
@luzpaz IMO issues like this should be closed directly and then reopened only if proper reproduction steps are provided, otherwise who knows if someone will remember to actually close this later, makes for easier maintenance IMO.
also note that the forum first policy is no longer in place so I don't know if we should keep recommending that, tbh sometimes it complicates things when you have to go back to the forum to get the full picture of the report.

luzpaz, 2022-12-01:
I'm still not used to the new contributions protocol. Thanks.

closed 2022-12-01 by luzpaz

adrianinsaval, 2022-12-01:
It's a work in progress! :)