FreeCAD/FreeCAD #7913: PATH-pocket made with splines fails

acp693, 2022-11-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?f=15&t=74016&sid=f67514feaa82f32fe9f6fa93062a4e95

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.31109 (Git)
Build type: Release
Branch: master
Hash: c23ee7ad88433a0f00e84afae0a62fe738be0c09
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * CurvedShapes 1.0.4
  * Curves 0.5.3
  * FeedsAndSpeeds 0.4.0
  * Marz 0.0.26-alpha
  * Silk 1.0.0
[/code]
```

### Subproject(s) affected?

Path

### Issue description

Path generated for a pocket made with splines fails to generate path that reaches the bottom of the pocket

### Anything else?

[BSplinePocket.zip](https://github.com/FreeCAD/FreeCAD/files/10125680/BSplinePocket.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-21:
Looks like in last dev situation with curve and spline is the same.
To get full depth need to set little value for **Finish Depth**

[image]

maxwxyz, 2024-12-21:
So is the issue resolved?

tarman3, 2024-12-21:
It doesn't look like it.
It seems the behavior has changed.

closed 2024-12-21 by maxwxyz