FreeCAD/FreeCAD #7914: Unsupported sub shape type, appears in versions after 30988

mowley-bob, 2022-11-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
Working Version

[code]
OS: Arch Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.21.30988 (Git) AppImage
Build type: Release
Branch: master
Hash: 0679c40d26ab2f48ca0f8cd0c34ac3bdbdacec54
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * Help 1.0.3
  * BIM 2021.12.0
[/code]

Non-working version

[code]
OS: Arch Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.21.31109 (Git) AppImage
Build type: Release
Branch: master
Hash: c23ee7ad88433a0f00e84afae0a62fe738be0c09
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * Help 1.0.3
  * BIM 2021.12.0
[/code]
```

### Issue description

The Dimension on the Technical Drawing called Battery-and-Holder reports an Unsupported sub-shape type on versions of FreeCAD after v21.30988.  

[Hailong-battery-and-holder.FCStd.gz](https://github.com/FreeCAD/FreeCAD/files/10126317/Hailong-battery-and-holder.FCStd.gz)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mowley-bob, 2022-12-01:
Screenshot of affected drawing.

[image]

mowley-bob, 2022-12-01:
Working in revision 30988
[image]

luzpaz, 2022-12-01:
Can confirm that upon opening the attached file in latest FreeCAD (version below) I see 
`07:21:01  <Exception> Unsupported sub-shape type` 

```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.31153 (Git) Snap 464
Build type: Release
Branch: master
Hash: 3f118b52bc62db6a5ca85daafaa2c021411e7177
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * TabBar
  * ExtremeProDark 2.5.4
  * FreeCAD-BlenderBIM 2022.10.20
  * PieMenu 1.2.3
```

WandererFan, 2022-12-02:
should be fixed by #7926

luzpaz, 2022-12-02:
Closing issue. If there are still remaining issues, please re-open.

closed 2022-12-02 by luzpaz