FreeCAD/FreeCAD #7918: Path:  Be able to replace command or parameter strings in the "refactored" postprocesors

LarryWoestman, 2022-11-30:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=73915

### Subproject(s) affected?

Path

### Idea description

It would help a lot if the "refactored" postprocessors could be configured to replace the command strings ("G01", for example) and/or the parameter strings ("Q", for example) with user-specified strings without having to modify the code that is shared between all of the "refactored" postprocessors.
A specific example would be replacing "Q" with "I" in the drilling G-code commands (G73 and G78).
In the non-refactored postprocessors a few lines of code need to be changed to do this, but in the "refactored" postprocessors there isn't an easy place to make a similar change without affecting all of the "refactored" postprocessors.

### Anything else?

I intend to figure out a way to do this to the "refactored" postprocessors eventually, but don't want the idea to be forgotten.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ShamanTcler, 2025-01-15:
Are you basically asking for an alias command? For Example if the alias command for G01 is F27, and everywhere the post processor would put a G01 it becomes an F27. That would be "possible" with a routine that evaluated the GCode file (not individual toolpaths) and did global text substitutions.

LarryWoestman, 2025-01-15:
This discussion triggered the filing of this issue:  https://forum.freecad.org/viewtopic.php?t=73915

I think there are cases where a relatively simple re-processing of the postprocessor output might work.
Other cases that I have run into require more logic and "internal" knowledge of the postprocessor state and might not be so easy to do as a post-postprocessor step.

I have been thinking about this general area while refactoring the postprocessor shared code.  The changes to be able to swap out routines that process the "parameters" (X, F, ...) open up some new possibilities.  I try to keep in mind being able to customize a postprocessor to output svg, dxf, hpgl (plotter g-code equivalent), and laser while I am doing my refactoring.  Odds are that I won't account for all of those capabilities, but I want the end result to stay "flexible" for more than just g-code output.  There are a lot more postprocessors that I would like to look and and create "refactored" versions of, and I expect to learn even more about the kinds of customization flexibility that is needed.

My future plans for the refactored postprocessor code base include being able to do a "string swap" for things like "G1" and refactoring the main processing "loop" for an internal path data structure to call a list of functions from a data structure similar to the way the the parameter processing loop is now done.  I suspect that I will need three more lists, one for "pre-processing" (headers and preambles, for example), one for "post-processing" (postambles, for example) and one for the main "processing loop" that generates the g-code lines involved in actually moving around and turning the cutter.
Some of the existing checks and functions for the postprocessors that I started from seem to be unique to a single postprocessor and probably need to move back out of the shared code base into the customizations for a given postprocessor once sufficient customization ability is available.

I also want to make sure that it is possible to "inherit" from a refactored postprocessor if you want to build on a postprocessor that already exists rather than copy and existing postprocessor and modify it.