FreeCAD/FreeCAD #7924: Pointcloud-positioning-issue

CalzitPLA, 2022-12-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu 20.04.5 LTS (Budgie:GNOME/budgie-desktop)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: German/Germany (de_DE)
Installed mods: 
  * InventorLoader 1.3.0
  * FEMbyGEN 1.1.0
[/code]
```

### Subproject(s) affected?

None

### Issue description

In CloudCompare arranged pointclouds don't import correct

1.  Multicloud pointcloud loosing their zeropoint . Weird results after importing.
2. FreeCAD offsets imported pointcloud

Files are checked with different software and are not corrupt.
Error is reproducable.
By merging multicloud the second error gets a workaround and dont occure then.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-12-01:
Need a file to reproduce and preferably some screenshots.

CalzitPLA, 2022-12-02:
Pct with dxf in CC:
[image]+

Same files in FreeCAD:
[image].

Subsampled the clouds. One merged and one as multicloud.
!!!!!!!!!!!!!!!!! I had to change the ending from e57 to txt !!!!!!!!!!!!!!!!!!!!!!!!
!!!!!!!!!!!!!!!!! Change back to e57, or it won't work !!!!!!!!!!!!!!!!!!!!!!!!!!

[Try.multicloud.merged.txt](https://github.com/FreeCAD/FreeCAD/files/10138683/Try.multicloud.merged.txt)

[Try.multicloud.txt](https://github.com/FreeCAD/FreeCAD/files/10138685/Try.multicloud.txt)

luzpaz, 2022-12-03:
Why not just zip them in to a zip archive?

luzpaz, 2022-12-20:
@wwmayer do you have a moment to weigh-in on this ?

wwmayer, 2022-12-21:
The E57 format defines a placement for each data3D node which isn't considered in FreeCAD.

commit f0a4ec82 (2022-12-21): Points: fixes #7924: Pointcloud import anomalies
closed 2022-12-21 by wwmayer