FreeCAD/FreeCAD #7945: [Problem] 2D Offset with a complex imported sketch

Igor-Misic, 2022-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu 20.04.5 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Part

### Problem description

When I imported SVD and then converted it to sketch I got this: 

[image]

The issue is when I try to use Part -> 2D-Offset I got this:
```
14:52:01  <Exception> makeOffset2D: result of offsetting is null!
14:52:01  Recompute failed! Please check report view.
```

Just to mention that 2D Offset works properly if I sketch something on my own.  What could be the issue? to complex sketch or some else? 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-12-04:
Please attach SVG

Igor-Misic, 2022-12-04:
[image]
SVG attached. 
To reproduce use the first "Path"
[image]

grubertechnic, 2022-12-08:
svg is a "crap" format for any type of CAD. It have not the preciseness, which is needed for that, besides the shape is too cluttered (see the points positions) and complex (awaiting intersections) in combination. I am sure that pretty every CAD will give not valid offset (in term of CAD, no nearing method) in this. Should be closed.

Igor-Misic, 2022-12-08:
Thanks for the explanation.

closed 2022-12-08 by Igor-Misic