FreeCAD/FreeCAD #7957: [Bug] fail to parse ifcopenshell version from non numbers

yoguut, 2022-12-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
OS: Debian GNU/Linux 11 (bullseye) (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Python version: 3.9.2
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.1
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Issue description

When exporting a build model to `IFC` file format. The following error occurred:

```plaintext,ignore
17:43:14  Traceback (most recent call last):
  File "<string>", line 4, in <module>
  File "/usr/share/freecad/Mod/Arch/exportIFC.py", line 212, in export
    preferences = getPreferences()
  File "/usr/share/freecad/Mod/Arch/exportIFC.py", line 167, in getPreferences
    elif hasattr(ifcopenshell, "version") and (float(ifcopenshell.version[:3]) >= 0.6):
<class 'ValueError'>: could not convert string to float: 'v0.'
```

This is likely caused by the version string of `ifcopenshell` having non number characters.
So I ran the following in python repl to confirm:

```python,ignore
import ifcopenshell
ifcopenshell.version
>>> 'v0.7.0-fa6bbf2d'
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-12-06:
Can you reproduce on v0.20.x ?

yoguut, 2022-12-11:
Just tried running it on a new machine with Python3.10 with the following FreeCAD installed via flatpak
```plaintext
[code]
OS: KDE Flatpak runtime (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.6, Qt 5.15.7, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.2
Locale: English/United States (en_US)
[/code]
```

Got this error when testing it inside of the FreeCAD's python console:
```plaintext
>>> import ifcopenshell
Traceback (most recent call last):
  File "/app/lib/python3.10/site-packages/ifcopenshell/__init__.py", line 54, in <module>
    from . import ifcopenshell_wrapper
  File "/app/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/app/lib/python3.10/site-packages/ifcopenshell/ifcopenshell_wrapper.py", line 13, in <module>
    from . import _ifcopenshell_wrapper
  File "/app/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 142, in _import
    return original_import(name, *args, **kwargs)
ImportError: /app/lib/python3.10/site-packages/ifcopenshell/_ifcopenshell_wrapper.so: undefined symbol: _ZTV13HdfSerializer

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/app/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/app/lib/python3.10/site-packages/ifcopenshell/__init__.py", line 62, in <module>
    raise ImportError("IfcOpenShell not built for '%s'" % python_distribution)
ImportError: IfcOpenShell not built for 'linux/64bit/python3.10'
```

Since I installed it via flatpak, my FreeCAD is looking at a path like within the flatpak container `/app/lib/python3.10/site-packages/ifcopenshell/__init__.py`.

And since I installed `ifcopenshell` via my system's default package manager `apt`. It's obvious to me that FreeCAD won't be able to import ifcopenshell from within `flatpak`

The easy solution is to uninstall my current FreeCAD and then install it again via `apt install freecad`. That should allow FreeCAD to adopt the `ifcopenshell` at the path that `ifcopenshell` is installed.

So my question is that is there a way to install `ifcopenshell` and somehow make it available for the flatpak version of FreeCAD? Because I think (other than just me) there are people who use flatpak version of FreeCAD and needed to export their IFC files?

Any tips on this one?

luzpaz, 2022-12-11:
I think you need to open a separate Flatpak  issue at https://github.com/flathub/org.freecadweb.FreeCAD/issues

luzpaz, 2022-12-11:
This issue is really a downstream 3rd party dependency problem. If the PPA was working correctly it would be the best solution (next to Flatpak and Snap).

closed 2022-12-11 by luzpaz