FreeCAD/FreeCAD #7960: [Issue] Geometry errors in solid created by scripting

acolomitchi, 2022-12-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * CurvedShapes 1.0.4
  * freecad.gears 1.0.0
  * lattice2 1.0.0
  * sheetmetal 0.2.59
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

Uses Part, PartDesign, InvoluteFeature in heavy python scripting - https://github.com/acolomitchi/FreeCAD-wormdrive.
To repro, take the [throated-worm-drive.GeomErrors.py](https://github.com/acolomitchi/FreeCAD-wormdrive/blob/main/throated-worm-drive.GeomErrors.py) and execute it in the FreeCAD's console.

The `Wheel_hobt1` in the "Wheel" body shows geometry errors (Part/Check geometry) , the  `Wheel_hobt2` is FUBARed.

[image]

### Anything else?

Context - I'm trying to get a complete worm-drive (gear and wheel) following the path of https://imgur.com/a/UzLSB

The worm-gear function and the resulted part are fine.

The worm-wheel procedure (`makeThroatedWormWheel`) - I believe correctly coded - produces solid parts with geometry errors on the way.

The procedure goes along the line of:
- obtain a worm-drive part
- use it repeatedly to cut a "gear blank" (flat cylinder), rotating the latter one tooth at a time

The first step (`Wheel_hobt0`) is fine, the second step (`Wheel_hobt1`) results in a part that apparently looks fine but has geometry errors, the third step (`Wheel_hobt2`) is badly damaged (assumed due to the geom errors in the prev step).

I tried the maintaining the wheel fixed and moving the "hob" around but the results are the same (the second step produces solids with geom errors)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-12-20:
Opened a [thread](https://forum.freecadweb.org/viewtopic.php?f=22&t=74602) on the FC forums

tomate44, 2022-12-21:
IMHO, when attempting several successive BOPs on complex shapes like in this script, it is not surprising that we hit one of already known limits of OCCT's BOPs.
So I don't think it is a valid bug report.

acolomitchi, 2022-12-21:
@tomate44 

> So I don't think it is a valid bug report.

sudden attack of pedantry - the bug is valid, only the project against which is raised against is wrong :)

OK, feel free to close it with a WONTFIX resolution, I'll see if I can get it transported against OCCT

luzpaz, 2022-12-21:
Please feel free to x-post the bug report you make upstream here so we can track it

closed 2022-12-21 by luzpaz