FreeCAD/FreeCAD #7961: On click at freecad file into the freecad program i lose all toolbar settings.

gorillale, 2022-12-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods: 
  * fasteners
[/code]
```

### Subproject(s) affected?

Core

### Issue description

On click at freecad file into the freecad program i lose all toolbar settings. 

When i start freecad alone, alle toolbar is ok, but after load a free3d file an start programm alone toolbar lose always...

Why ist the toolbar settings not stored?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-12-19:
@chennes what's the approach we take to tickets like this now?

chennes, 2022-12-20:
I tried to reproduce this, but I couldn't reliably get the toolbar to mess up. Sometimes it did, and sometimes it didn't, and it didn't seem tied to whether I was opening FreeCAD by double-clicking on a file or loading it directly. I think it makes sense to ask on the forums and see if anyone has a good test procedure to reproduce this issue.

adrianinsaval, 2023-05-04:
very unclear bug report, what actually happens to the toolbars? did OP try clearing all settings? Is OP sure he's launching the same version on both situations?

adrianinsaval, 2023-05-23:
I'm going to assume this is a duplicate of #5941 and will therefore close in favor of it since it's much better described with reproduction steps and confirmed by others

closed 2023-05-23 by adrianinsaval