FreeCAD/FreeCAD #7965: Main Menu is having a problem[Issue] [regression]

iplayfast, 2022-12-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Linux Mint 21 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31240 (Git)
Build type: Unknown
Branch: master
Hash: d08ccbfb77dd70b15e87a4589273c7d8d0b9707e
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/Canada (en_CA)
Installed mods: 
  * Plot 1.0.4
  * dummy.py
  * fcgear.backup1663084939.8135655 1.0.0
  * BOLTSFC
  * BIM 2021.12.0
  * CycloidGearBox
  * Design456.backup1663084959.3295681 0.0.1 (Disabled)
  * gbNov8.stl
  * workfeature
  * ThreadProfile.backup1663090363.0629296 0.12.0
  * fcgear 1.0.0
  * BOLTSFC.backup1663084916.1449692
  * FeedsAndSpeeds.backup1663084947.2214124 0.4.0
  * parts_library
  * OSE3dPrinter.backup1663090347.963575
  * FreeCAD_assembly3 0.11.2
  * parts_library.backup1663084960.3254437
  * cycdisk.stl
  * PieMenu.backup1663090349.8951852
  * CurvedShapes 1.0.4
  * Manipulator 1.4.9
  * gbNov8.FCStd
  * Manipulator.backup1663090346.481476 1.4.5
  * SelectorToolbar
  * EM
  * Assembly4 1.0.4
  * ThreadProfile 0.12.4
  * sheetmetal 0.2.59
  * pindisk.stl
  * testdisk.log
  * 3D_Printing_Tools
  * Lithophane.backup1663090409.4460692
  * es.stl
  * kicadStepUpMod 10.16.5
  * ose-workbench-core
  * ExplodedAssembly
  * A2plus.backup1663084907.2462485 0.4.56a
  * FCGear 1.0.0
  * fasteners.backup1663084935.08665 0.3.38
  * OSE3dPrinter
  * RemBench
  * Assembly4.backup1663084910.0541313 1.0.3
  * retr3d
  * retr3d.backup1663090354.7622156
  * A2plus 0.4.60d
  * ExplodedAssembly.backup1663084933.417286
  * RemBench.backup1663090353.5145144
  * CurvedShapes.backup1663084917.9989994 1.0.3
  * Plot.backup1663090350.9766667 1.0.3
  * LCInterlocking
  * BIM.backup1663084913.9426856 2021.12.0
  * Mechatronic
  * Design456 0.0.1
  * sheetmetal.backup1663090360.8238878 0.2.49
  * PieMenu
  * LCInterlocking.backup1663084958.1891425
  * drivdisk.stl
  * Mechatronic1
  * symbols_library
  * Help 1.0.3
  * SelectorToolbar.backup1663090359.6344967
  * FeedsAndSpeeds 0.5.0
  * Pyramids-and-Polyhedrons
  * Help.backup1663084950.2632241 1.0.3
  * workfeature-macro
  * ose-piping-workbench
  * Nikra-DAP
  * fasteners 0.4.23
  * Lattice2 1.0.0
[/code]
```

### Subproject(s) affected?

None

### Issue description

Clicking on File or Tool in the main menu doesn't do anything. This issue is not present in [code]
OS: Linux Mint 21 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/Canada (en_CA)
Installed mods: 
  * Plot 1.0.4
  * dummy.py
  * fcgear.backup1663084939.8135655 1.0.0
  * BOLTSFC
  * BIM 2021.12.0
  * CycloidGearBox
  * Design456.backup1663084959.3295681 0.0.1 (Disabled)
  * gbNov8.stl
  * workfeature
  * ThreadProfile.backup1663090363.0629296 0.12.0
  * fcgear 1.0.0
  * BOLTSFC.backup1663084916.1449692
  * FeedsAndSpeeds.backup1663084947.2214124 0.4.0
  * parts_library
  * OSE3dPrinter.backup1663090347.963575
  * FreeCAD_assembly3 0.11.2
  * parts_library.backup1663084960.3254437
  * cycdisk.stl
  * PieMenu.backup1663090349.8951852
  * CurvedShapes 1.0.4
  * Manipulator 1.4.9
  * gbNov8.FCStd
  * Manipulator.backup1663090346.481476 1.4.5
  * SelectorToolbar
  * EM
  * Assembly4 1.0.4
  * ThreadProfile 0.12.4
  * sheetmetal 0.2.59
  * pindisk.stl
  * testdisk.log
  * 3D_Printing_Tools
  * Lithophane.backup1663090409.4460692
  * es.stl
  * kicadStepUpMod 10.16.5
  * ose-workbench-core
  * ExplodedAssembly
  * A2plus.backup1663084907.2462485 0.4.56a
  * FCGear 1.0.0
  * fasteners.backup1663084935.08665 0.3.38
  * OSE3dPrinter
  * RemBench
  * Assembly4.backup1663084910.0541313 1.0.3
  * retr3d
  * retr3d.backup1663090354.7622156
  * A2plus 0.4.60d
  * ExplodedAssembly.backup1663084933.417286
  * RemBench.backup1663090353.5145144
  * CurvedShapes.backup1663084917.9989994 1.0.3
  * Plot.backup1663090350.9766667 1.0.3
  * LCInterlocking
  * BIM.backup1663084913.9426856 2021.12.0
  * Mechatronic
  * Design456 0.0.1
  * sheetmetal.backup1663090360.8238878 0.2.49
  * PieMenu
  * LCInterlocking.backup1663084958.1891425
  * drivdisk.stl
  * Mechatronic1
  * symbols_library
  * Help 1.0.3
  * SelectorToolbar.backup1663090359.6344967
  * FeedsAndSpeeds 0.5.0
  * Pyramids-and-Polyhedrons
  * Help.backup1663084950.2632241 1.0.3
  * workfeature-macro
  * ose-piping-workbench
  * Nikra-DAP
  * fasteners 0.4.23
  * Lattice2 1.0.0
[/code]
and seems to have started happening in the last month. 
To get around it I can click on any of the other menu headers (View Tools Macro ...) to pop up the menu and then move the mouse to the File or Edit menu and their respective menus appear, or I can use Ctrl-F to bring up the file menu, and Ctrl-E to bring up the edit menu. 

Just using the mouse seems to not work.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2022-12-23:
I can confirm, I also experience this bug on recent builds.

luzpaz, 2022-12-23:
@obelisk79 [About](https://wiki.freecad.org/About) info ?

obelisk79, 2022-12-24:
```
OS: Manjaro Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31351 +3 (Git)
Build type: Release
Branch: PR7888
Hash: a2f235cd2a0a4f4c56c8011ea736376259a5160d
Python 3.10.8, Qt 5.15.7, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * IconThemes
  * Behave-Dark-Colors 0.1.1
  * ProDarkThemePreferencePack 1.0.0
  * Assembly3 0.11.4
  * CubeMenu
  * Manipulator 1.4.9
  * Silk 1.0.0
  * lattice2 1.0.0
  * fasteners 0.4.27
  * Lithophane
  * Curves 0.6.5
  * PieMenu 1.2.4
  * freecad.gears 1.0.0
  * A2plus 0.4.60g
  * Render 2022.3.0
  * Dracula 0.0.2
  * CfdOF 1.20.6

```

iplayfast, 2023-01-11:
I've done a git bisect and it appears that the commit causing troubles is
06d65c73a38466e1103c39606ffe7761e08aca61 is the first bad commit
commit 06d65c73a38466e1103c39606ffe7761e08aca61
Author: 0penBrain <48731257+0penBrain@users.noreply.github.com>
Date:   Thu Jun 16 09:03:24 2022 +0200

    [BUGFIX] Gui: correctly restore message type enabling in report view

 src/Gui/ReportView.cpp | 3 +++
 1 file changed, 3 insertions(+)

luzpaz, 2023-01-12:
CC @0penBrain

0penBrain, 2023-01-12:
First I can't reproduce with latest master. ;)
Then there is no chance the pointed commit can be responsible for that. :)

I'm most thinking there is an issue either with an addon or with some configuration issues.

Can one reproduce with blank configuration ?
To get one, all this folders shall be renamed if exist (on Linux)
```
~/.local/share/FreeCAD
~/.config/FreeCAD
~/.cache/FreeCAD
~/.FreeCAD
```

iplayfast, 2023-01-14:
Yes, I couldn't see how those changes affected menus either. But figured I was being stupid.
Sure enough, can't reproduce with a blank configuration. 

After much installing of workbenches (fresh!) I've discovered the source of the problem is RemBench. Which is a very nice workbench addon.

luzpaz, 2023-01-14:
@iplayfast do you mind opening a ticket on [FreeCAD-addons](https://github.com/FreeCAD/FreeCAD-addons) to track this bug ?

bstott, 2023-01-14:
I have similar, not the same, issue on Windows. See #8169 .  I don't use Rem.  However, I have used the FreeCAD feature to add Sketcher to another Work Bench Toolbar. I've forgotten which WB.  Part? Curves?  I will need to get back to my desk later today, 1/14/23.

iplayfast, 2023-01-14:
> @iplayfast do you mind opening a ticket on [FreeCAD-addons](https://github.com/FreeCAD/FreeCAD-addons) to track this bug ?

Done.

iplayfast, 2023-01-14:
> I have similar, not the same, issue on Windows. See #8169 . I don't use Rem. However, I have used the Fr

I narrowed it down by renaming my Mod directory to Mod1, and then reinstalling the addon's I used one by one using the AddonManager.  This had the benefit of not installing obsolete addons or weird things that have developed over the years.

bstott, 2023-01-14:
Great... Thank you.  Now, to fix? (smile)

Syres916, 2023-02-02:
@iplayfast with relatively high confidence I believe the regression is actually caused by https://github.com/FreeCAD/FreeCAD/commit/7458912b3340a428cd5fcb763859659ebb8b76ce which logically does look more reasonable. Tomorrow I will look into debugging the Accessories menu of PieMenu (and other Triplus authored Addons) with relation to this code to see if there's a simple solution.

Syres916, 2023-02-03:
Update: There seems to be (in my simplistic view) a clash between https://github.com/FreeCAD/FreeCAD/blob/master/src/Gui/ToolBarManager.cpp#L184 and https://github.com/mdkus/PieMenu/blob/master/InitGui.py#L66
I'll try and re-write the Pie Menu configuration to use the existing `View` menu rather than create a new Accessories menu and see if it's stable on Linux.

Syres916, 2023-02-04:
@realthunder I would appreciate some of your valuable time to look at this please. I notice your LinkDaily AppImage of 31/01/2023 loads the PieMenu 1.2.4 addon without causing the side effects noted here. Is there something that needs updating in `ToolBarManager.cpp` in order to correct the regression?

iplayfast, 2023-02-05:
I'm no longer able to compile as I'm getting an unrelated crash (something to do with pivy and coin)
[code]
FreeCAD: /home/chris/github/freecad-source/src/Mod/Part/Gui/TaskDimension.cpp:1343: static void PartGui::ArcEngine::initClass(): Assertion `SoEngine::getClassTypeId() != SoType::badType()' failed.
[/code]
But, the last time I was able to get it going this error was gone with the un-installation of that addon. So as far as I can tell this issue as been resolved.

wwmayer, 2023-03-13:
According to https://forum.freecad.org/viewtopic.php?t=76756 it seems it's caused by the Pie-Menu

luzpaz, 2023-03-13:
CC @mdkus (piemenu dev)

wwmayer, 2023-03-13:
I have installed the Pie-Menu and can conform that File and Edit don't react on clicks.

mdkus, 2023-03-13:
@luzpaz 
@wwmayer
I couldn't reproduce this in version 0.21.0.30492 (Windows 11). I will try it with the actual developer version...

wwmayer, 2023-03-13:
The ghost widget that is shown here is this one: https://github.com/FreeCAD/FreeCAD/blob/master/src/Gui/ToolBarManager.cpp#L182
When I change the code and set an object name _fc_action_widget_ and remove the Pie-Menu add-on then everything is fine. I can access this widget and make it visible with:
```
from PySide import QtWidgets
mw = Gui.getMainWindow()
aw = mw.findChild(QtWidgets.QWidget, "_fc_action_widget_")
aw.pos() # (0, 0)
aw.size() # (100, 30)
# This raises then the problem independent of the Pie-Menu
aw.setAutoFillBackground(True)
aw.raise_() # makes it visible
```
So, the problem already is in FreeCAD but the Pie-Menu code somehow makes this problem visible (maybe caused by the function showAtMouse).

https://github.com/FreeCAD/FreeCAD/pull/8608 fixes the problem.

closed 2023-03-13 by wwmayer

mdkus, 2023-03-25:
@luzpaz 
@wwmayer 
I analyzed the code of my predecessor 'triplus' and found out, that it is a timing problem. The problem comes up when PieMenu is too fast when adding the 'AccessoriesMenu'. So triplus implemented a timer for waiting on FC to have loaded the default workbench with its initial menu bar. Addons which are real workbenches can fortunatly use workbenches method 'addMenuBar', which avoids the problem somehow. So I reimplemented this timer in PieMenu version 1.2.7 and it should work even without the patch. Please confirm.

bstott, 2023-03-25:
Great. Excellent sleuthing.  Thanks. Many will be able to re-install the PieMenu soon.  Thanks, again.

bstott, 2023-03-25:
Problem SOLVED.

The PieMenu timing change WORKS!!!

I updated/upgraded PieMenu on 3/25/23 and tested within FreeCAD 0.21.0, and realthunders 1/2023 FreeCAD, which both had the problem.    IT IS FIXED!!!   The PieMenu timing change - WORKED!   The File, Edit, and View Menus work.

Congratulations.

commit 48ca8d52 (2023-03-13): [Gui] Fix regressions from PR7771...

specifically https://github.com/FreeCAD/FreeCAD/commit/7458912b3340a428cd5fcb763859659ebb8b76ce fixes #7965 #8440