FreeCAD/FreeCAD #7987: Lables & attributes of Model not shown good if step file encoding not UTF-8 [Issue]

fxliang, 2022-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Chinese/China (zh_CN)
[/code]
```

### Subproject(s) affected?

None

### Issue description

open a step file export by solidworks, assembly type, with part labeled with Chinese characters.
if open step file as text, for example there is a line like bellow.
```#108530 = SHAPE_REPRESENTATION ( '海绵2', ( #51170 ), #27577 ) ;``` 
labels not displayed properly like
[image]

I checked the step file, and found that the step file is encoded in GB18030 not UTF-8
[image]

so i tried to change the file encoding to UTF-8, and saved a copy, and then opened the copy , it works fine.
[image]

it seems because in these lines UTF-8 is predefined, not optional.
[lines](https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Import/Gui/AppImportGuiPy.cpp#L404-L406)

so is it possible a feature to auto detect the file encoding, or an encoding option in file open dialog ?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2022-12-08:
The support of utf8 from the step importer of OCCT is pretty "new". I do not know if this is already considered in FreeCAD. 

2017: https://forum.freecadweb.org/viewtopic.php?t=20815
almost  2021: https://www.opencascade.com/open-cascade-technology-7-5-0-released/

But yes, that is an issue.

daanmichiels, 2022-12-10:
@fxliang - Do you have a minimal step file to test this with?

fxliang, 2022-12-11:
> @fxliang - Do you have a minimal step file to test this with?

later tomorrow ( beijing time)

fxliang, 2022-12-12:
> @fxliang - Do you have a minimal step file to test this with?

pls kindly check the attached file bellow. four files inside.
files named with ```_utf8``` have been changed encoding with vim (command ```:set fenc=utf-8```

when open with vim as text file, cp936 detected.
[image]

formats both ap203 and ap214 have been included.

[image]

[steps.zip](https://github.com/FreeCAD/FreeCAD/files/10203752/steps.zip)

luzpaz, 2023-03-15:
@realthunder would you have a moment to look at this ?

luzpaz, 2023-04-13:
bumping for visibility

Kuzma30, 2024-05-26:
I found way how to set file encoding (ITS ONLY IDEA). I test this on OCCT 7.8.1. We need ReadFile function with parameters (some version of OCCT have only ReadFile (const Standard_CString theFileName);)
```
  Standard_EXPORT IFSelect_ReturnStatus ReadFile(const Standard_CString theFileName,
                                                 const StepData_ConfParameters& theParams);
```

Patch file. here format hardcoded to Resource_FormatType_CP1251
```
diff --git a/src/Mod/Import/App/ReaderStep.cpp b/src/Mod/Import/App/ReaderStep.cpp
index 47cbb48f5b..9e93d5b73e 100644
--- a/src/Mod/Import/App/ReaderStep.cpp
+++ b/src/Mod/Import/App/ReaderStep.cpp
@@ -31,6 +31,7 @@
 #include <XSControl_WorkSession.hxx>
 #endif
 
+#include <StepData_StepModel.hxx>
 #include "ReaderStep.h"
 #include <Base/Exception.h>
 #include <Mod/Part/App/encodeFilename.h>
@@ -51,7 +52,10 @@ void ReaderStep::read(Handle(TDocStd_Document) hDoc)  // NOLINT
     aReader.SetNameMode(true);
     aReader.SetLayerMode(true);
     aReader.SetSHUOMode(true);
-    if (aReader.ReadFile(name8bit.c_str()) != IFSelect_RetDone) {
+    Handle(StepData_StepModel) aStepModel = new StepData_StepModel;
+    aStepModel->InternalParameters.InitFromStatic();
+    aStepModel->SetSourceCodePage(Resource_FormatType_CP1251);
+    if (aReader.ReadFile(name8bit.c_str(),aStepModel->InternalParameters) != IFSelect_RetDone) {
         throw Base::FileException("Cannot read STEP file", file);
     }
```
 Values for format from Resource_FormatType.hxx You can use any in your test.
```
enum Resource_FormatType
{
  Resource_FormatType_SJIS,         //!< SJIS (Shift Japanese Industrial Standards) encoding
  Resource_FormatType_EUC,          //!< EUC (Extended Unix Code) multi-byte encoding primarily for Japanese, Korean, and simplified Chinese
  Resource_FormatType_NoConversion, //!< format type indicating non-conversion behavior
  Resource_FormatType_GB,           //!< GB (Guobiao) encoding for Simplified Chinese
  Resource_FormatType_UTF8,         //!< multi-byte UTF-8 encoding
  Resource_FormatType_SystemLocale, //!< active system-defined locale; this value is strongly NOT recommended to use

  // Windows-native ("ANSI") 8-bit code pages
  Resource_FormatType_CP1250,       //!< cp1250 (Central European) encoding
  Resource_FormatType_CP1251,       //!< cp1251 (Cyrillic) encoding
  Resource_FormatType_CP1252,       //!< cp1252 (Western European) encoding
  Resource_FormatType_CP1253,       //!< cp1253 (Greek) encoding
  Resource_FormatType_CP1254,       //!< cp1254 (Turkish) encoding
  Resource_FormatType_CP1255,       //!< cp1255 (Hebrew) encoding
  Resource_FormatType_CP1256,       //!< cp1256 (Arabic) encoding
  Resource_FormatType_CP1257,       //!< cp1257 (Baltic) encoding
  Resource_FormatType_CP1258,       //!< cp1258 (Vietnamese) encoding

  // ISO8859 8-bit code pages
  Resource_FormatType_iso8859_1,    //!< ISO 8859-1 (Western European) encoding
  Resource_FormatType_iso8859_2,    //!< ISO 8859-2 (Central European) encoding
  Resource_FormatType_iso8859_3,    //!< ISO 8859-3 (Turkish) encoding
  Resource_FormatType_iso8859_4,    //!< ISO 8859-4 (Northern European) encoding
  Resource_FormatType_iso8859_5,    //!< ISO 8859-5 (Cyrillic) encoding
  Resource_FormatType_iso8859_6,    //!< ISO 8859-6 (Arabic) encoding
  Resource_FormatType_iso8859_7,    //!< ISO 8859-7 (Greek) encoding
  Resource_FormatType_iso8859_8,    //!< ISO 8859-8 (Hebrew) encoding
  Resource_FormatType_iso8859_9,    //!< ISO 8859-9 (Turkish) encoding

  // Addition code pages
  Resource_FormatType_CP850,        //!< ISO 850 (Western European) encoding
  Resource_FormatType_GBK,          //!< GBK  (UnifiedChinese) encoding
  Resource_FormatType_Big5,         //!< Big5 (TradChinese) encoding

  // old aliases
  Resource_FormatType_ANSI = Resource_FormatType_NoConversion,
  Resource_SJIS = Resource_FormatType_SJIS,
  Resource_EUC  = Resource_FormatType_EUC,
  Resource_ANSI = Resource_FormatType_ANSI,
  Resource_GB   = Resource_FormatType_GB,
};
```

What we need
1. Make Dialog where we can select correct SourceCodePage 
2. pass this code page to ReaderStep::read function

Kuzma30, 2024-05-28:
https://forum.freecad.org/viewtopic.php?t=87950

closed 2024-06-17 by chennes