FreeCAD/FreeCAD #7996: [Issue] TempPath is no longer respected

PLChris, 2022-12-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=73868

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.31155 (Git)
Build type: Release
Branch: master
Hash: 6f1b29769b33676edca2bd883b80bed25a0758c8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * DynamicData 2.46.0
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.4.0
  * fcgear 1.0.0
  * fasteners 0.4.16
  * sheetmetal 0.2.59
  * ExplodedAssembly
  * Curves 0.6.1
[/code]
```

### Subproject(s) affected?

None

### Issue description

(Not knowing where this will find attention here, if in "Issues" or "Problems", I create a ticket in both, because it is both, an issue and a problem. Someone knowing the difference can close or delete one of them, and please report when to create which. I tried to find out here https://forum.freecadweb.org/viewtopic.php?p=643735 in the forum, but there is no answer so far.)

The parameter BaseApp→Preferences→General→TempPath is no longer respected. It points to the directory, where the auto recovery files are stored.

How to reproduce:

- From the menu open Tools>Edit parameters
- Navigate to BaseApp->Preferences->General.
- Create a string type with the name TempPath and set it to an existing directory path. Use the expanded path and not ~/
- Check Preferences->General->Document->Storage->"Save AutoRecovery information every"
- Set the value of Preferences->General->Document->Storage->"Save AutoRecovery information every" to 2 min
(- I'm not sure if you have to restart FreeCAD here)
- Create a new document and add a Cube in Part workbench
- Watch the directory specified in TempPath

-> Existing behaviour: No backup file appears
-> Expected behaviour: there should appear a backup file in the TempPath. In 0.19 there is a directory created immediately for each file, and after the timeout there are two files inside of that directory, the FreeCAD file itself and an additional xml file 

This is a regression. It was still working in
OS: macOS 10.16
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: (HEAD detached at 0.19.2)
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

It is also not working in 0.20
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * DynamicData 2.46.0
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.4.0
  * fcgear 1.0.0
  * fasteners 0.4.16
  * sheetmetal 0.2.59
  * ExplodedAssembly
  * Curves 0.6.1
[/code]

### Anything else?

It would be good to know the difference between issues and problems, see [https://forum.freecadweb.org/viewtopic.php?p=643735](url)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-12-12:
No need to duplicate as it doesn't really make a difference which template you use, all are issues and end up in the same list, I suggest closing this one.

luzpaz, 2022-12-19:
TempPath variableis defined via CMaKe right? @berniev any idea?

berniev, 2022-12-19:
Can't do tests at the moment. Any log entries re saving?

luzpaz, 2022-12-20:
Code references? https://github.com/FreeCAD/FreeCAD/blob/9869a2c3d7bdacdfac605db318c73e3b2d71245c/src/App/Application.cpp#L2597-L2602

adrianinsaval, 2022-12-27:
seems this parameter is not used anymore https://forum.freecadweb.org/viewtopic.php?p=649760#p649760
@wwmayer is there any point in having the above code then?

wwmayer, 2022-12-27:
> is there any point in having the above code then?

When using this path for other purposes than the recovery files a user may still want to customize this directory to avoid to delete the files when rebooting the system.

luzpaz, 2023-03-22:
Whoever tackles this, please also take a look at #7998 
TIA!

luzpaz, 2023-04-11:
bumping for visibility

adrianinsaval, 2023-04-12:
werner mentioned that this is no longer used for auto recovery files: https://forum.freecad.org/viewtopic.php?p=649760#p649760
so this is likely not actually a bug, @PLChris can you confirm?

luzpaz, 2023-04-15:
@PLChris ping

luzpaz, 2023-12-17:
Waiting on @PLChris

adrianinsaval, 2023-12-18:
As werner pointed out before, this is not a bug but expected behavior. Transient files are stored in `QStandardPaths::GenericCacheLocation` which is os dependent. [Qt docs](https://doc.qt.io/qt-6/qstandardpaths.html) says it's these:

Path type | mac | linux and other unix like | windows
---|---|---|---
GenericCacheLocation | "~/Library/Caches" | "\~/.cache" | "C:/Users/<USER>/AppData/Local/cache"

closed 2023-12-18 by adrianinsaval