FreeCAD/FreeCAD #7997: clone rounded rectangle - version 20.1 - 29410

cnc50, 2022-12-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
Tried to clone a rectangle with rounded edges from the menu in scetcher workbench. Doesn`t work, if I change the width or height from the original rectangle. Only changes in radius work.

version 20.1 - 29410
```

### Subproject(s) affected?

None

### Issue description

[clone rounded rectangle.zip](https://github.com/FreeCAD/FreeCAD/files/10193291/clone.rounded.rectangle.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adiazubierna, 2022-12-11:
I could reproduce your issue with the version 0.20.1.29410

First I tried with your file and then I create a new one from scratch. According to the documentation, all the constrains in the clone should change when the constrains of the original rectangle are changed.

chennes, 2022-12-18:
I cannot reproduce in 0.21_dev: when I clone I get perfectly matching constraints. Can you retest with the latest development version?
```
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31337 +1 (Git)
Build type: Release
Branch: addonManagerExtractInstaller
Hash: 430d8600603c764e44b7a33ff19a15dd8507cd35
Python 3.8.6+, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.5.3
Locale: English/United States (en_US)
```

cnc50, 2022-12-18:
I found out  to be very careful of selecting the item ( including all
necessary constraints)  which should be cloned.
Must take a closer look.
Thanks a lot.

luzpaz, 2022-12-29:
@cnc50 ping

luzpaz, 2023-01-08:
@cnc50 please follow-up testing on the development build https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

cnc50, 2023-01-08:
Thanks a lot.
What is the difference between py38 and py310 7z-file?

luzpaz, 2023-01-08:
@cnc50 one is built against python v3.8 and the other python v3.10

luzpaz, 2023-01-19:
@cnc50 ping

luzpaz, 2023-11-26:
@cnc50 ping

maxwxyz, 2024-03-24:
I'll close this as the new tool was merged and there was no further feedback.

closed 2024-03-24 by maxwxyz