FreeCAD/FreeCAD #8016: [Bug] Inconsitent behavior of DiffuseColor after changing the number of faces

Roy-043, 2022-12-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.21.31155 (Git)
Build type: Release
Branch: master
Hash: 6f1b29769b33676edca2bd883b80bed25a0758c8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Part

### Issue description

The DiffuseColor property is handled inconsistently if the number of faces of an object changes. Initially all faces of the object will then receive the first color in the DiffuseColor list, but after saving and reopening the file the faces assume the ShapeColor of the object.

Test:
1. Open the file from the attaches zip: [DiffuseColor_Test.zip](https://github.com/FreeCAD/FreeCAD/files/10202036/DiffuseColor_Test.zip)
2. Change the Base of the Extrude to Sketch001.
3. Result: All faces of the object are now red.
4. Save and reopen the file.
5. Result: All faces are now grey.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-12-27:
Issue still occurs.

New file for V1.1 (to handle the transparency-alpha switch): 
[DiffuseColor_Test_1.1.FCStd.zip](https://github.com/user-attachments/files/18262274/DiffuseColor_Test_1.1.FCStd.zip)
Remove the .zip extension.

```
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

luzpaz, 2025-12-03:
Can anyone propose a solution ?