FreeCAD/FreeCAD #8018: [Feature Request] Implement G84 Canned Tapping Cycle

luvtofish, 2022-12-11:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=74171&sid=cfddc1f3af4bdffdb3cf1eded7005299

### Subproject(s) affected?

Path

### Idea description

Tapping on a CNC is every bit as crucial as the drilling cycles. Rigid tapping aside, just having the ability to perform a rudimentary G84 canned cycle with a floating tap holder tool would close a huge functionality gap in Path.

Needed functionality would include: 

- ability to enter tap size (eg. 1/4-20, M6-1, etc)
- spindle speed (coupled with tap size, used calculate feed rate in/pm & mm/pm)
- ability to set spindle rotation for LH or RH threads 
- Z depth
- R Plane (clearanceHeight) and optional O plane (safeHeight) for clearing obstacles
- optional dwell entry in seconds.
 
The canned cycle would need to understand how to change spindle direction, halt the spindle at Z depth, dwell (optional), and reverse spindle on the retract. Once returning to R depth, reverse the spindle back to normal rotation. Advance to next hole, etc.

### Anything else?

typical GCODE example:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2022-12-17:
Thanks for the feature request.  A couple of thoughts about this off the top of my head:

A user would create a tap tool using the toolbit system.  We don't have a standard tap tool shape but the toolbit system will support everything needed to create one.  This is pretty simple and is a good milestone task.

Spindle speed is declared in the tool controller so nothing should be needed there.  Calculating the feed rate is also an independent unit of work from the tapping operation. I would consider it out of scope for this feature.

The R plane and O plane should be derived from the safe height and clearance height properties common to all operations.  Some logic to derive it may be necessary but we should not need to add any properties to the operation.

That means most of what's required is already done.   A tapping operation should work on essentially the same kind of geometry as a drilling operation.  So we can copy src/Mod/Path/Path/Op/Drilling.py and use it as the starting point.  

The real logic, and it isn't very complicated, will be in a new rigid tapping generator.  I would start by copying /src/Mod/Path/Path/Base/Generator/drill.py.   The generator is actually quite simple.  The main part is a single pure function that takes everything needed to produce a single tapping cycle.  (Ignore feed rates for now).   The most important part is the edge.  This is where the tap will go.  One end of the edge is the start point and the other is the endpoint.  

The generator is very thoroughly tested.  In the case of drilling, the tests are in /src/Mod/Path/PathTests/TestPathDrillGenerator.py   Again, we can begin by copying that file as a starting point.  Writing the test cases first would be incredibly helpful.  To see how it works, you can do this at the command line:

`./bin/FreeCADCmd -t TestPathApp.TestPathDrillGenerator`

 So here's a work plan:

1.  copy the drillgenerator tests and adapt them to test a hypothetical tap generator.  Tests will fail.
2. Write the tapgenerator to make the tests pass.
3. Once the generator works, copy the Drill.py operation to RigidTap.py and adapt it to drive the generator.
4. Copy /src/Mod/Path/Path/Op/Gui/Drilling.py and adapt it as RigidTapping.py. If an icon is needed, do it now.
5. Tweak the cmake files to make all the new stuff install where it needs to be.

luvtofish, 2022-12-18:
Great Info Brad! With your details, I jumped in and started testing some
things. I found a couple items in the drilling generator tests that will
not apply (peck and chip breaking). I now have working tests minus those
two items.

I’ll keep plugging away at this and see how far I can take it.

Thanks
Dan

sliptonic, 2022-12-19:
We generally try to build our operations to support LinuxCNC as a reference implementation.  Whenever possible, we push the quirks of gcode dialect into the postprocessors. Looking closer at the [LinuxCNC gcode reference](http://linuxcnc.org/docs/html/gcode/g-code.html). 

I see three related gcodes that we should probably support. 
G33.1 is Rigid tapping.
G74 Left hand tapping with a floating chuck
G84 Right hand tapping with a floating chuck.

I think all three of these can probably be addressed in the same generator.

See especially the warning in G33.1 about the coordinate position.  This is something the generator can check for and throw an error.  For example, the edge has different XY coordinates between the start and end points.   This shouldn't happen in practice and will be caught upstream as well but the generator can check and error if it gets through.

G74/G84 looks pretty simple.  The only difference is the righthand/left hand which controls spindle direction.  There are two ways to handle this.  The operation could have a right/left enumeration property.  The better way is for the tapping tool to have that property. 

The operation will need a dwell time property

G33.1 rigid tapping is different
it has two additional words
K - distance per revolution.  The generator should be able to derive this from the feed and speed.

I - optional spindle speed multiplier for faster return move.  To support this, we would need another property in the operation.  I'm inclined to NOT support this in the first version and wait for someone to complain.

At this point, I see the tapping tool being similar to a drill but also having a pitch, and handedness property.  Taps also have properties like whether they are bottoming.  Let's not add that now.

luvtofish, 2022-12-19:
Rigid Tapping G33.1 requires three specific hardware capabilities in the CNC:

- DC Spindle drive with encoder
- Spindle encoder Control board 
- Rigid Tapping Option in the controller (CNC has to support it)

G74/G84 cycles require a floating tap holder and manual feed/speed callout.

CNC controls that support these canned cycles handle the spindle direction by virtue of the call itself. They understand the need to dwell (optional) at the bottom of the tapping cycle, reverse spindle rotation until arriving at R height, then reverse rotation back to normal. The Dwell time is optional and is a needed input along with the G call and movements. 

I now have a working test solution for G84. It's very basic, no feed or speed calculations, just the ability to create a tapping operation on a series of holes with the ability to supply a dwell time. I removed the peck and chip breaker references as they do not apply. Path doesn't need to do anything much here, just spit out the G codes and dwell along with the movements. The Control handles the spindle requirements as mentioned. 

Including both G74/G84. I envision a user selectable drop down box with either option listed and output the code requested code. A better option would be to grab the user input and set the spindle rotation of the selected tapping tool by updating the tool controller direction. Would think this could also be done for feed rate and spindle speed? It's left up to the user to decide proper speed/feed rates to properly execute the tapping sequence based on thread pitch/spindle speed. 

G33.1 Rigid Tapping seems much more involved to me. Different command parameters have to be coded along with the required movements.

sliptonic, 2022-12-19:
> I now have a working test solution for G84. It's very basic, no feed or speed calculations, just the ability to create a tapping operation on a series of holes with the ability to supply a dwell time. 

Awesome!  Feel free to create a PR in draft mode any time.  You can continue to push to your own branch to update the PR until it's ready.  The sooner others can see it, the better.

> Including both G74/G84. I envision a user selectable drop down box with either option listed and output the code requested code. A better option would be to grab the user input and set the spindle rotation of the selected tapping tool by updating the tool controller direction. 

Agreed.  This logic should be added to the tool controller.  I don't like adding the G74/84 option to the operations.  Then the user could set G74 (left hand) but choose a right hand tool.  I can't imagine there's ever a case where that makes sense.

So the ToolController should have some logic to check if the tool has directionality and set the spindle direction accordingly and throw an error if the user sets it wrong.

> G33.1 Rigid Tapping seems much more involved to me. Different command parameters have to be coded along with the required movements.

[It doesn't look too bad](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g33.1).   We already have XYZ from the input edge.  I and $ words are optional and can be skipped for now.  That leaves the K word to be calculated.  That's either taken from a pitch parameter of the toolbit or calculated from the Feed/Speed of the toolcontroller.  The interesting thing would be what to do if both a thread pitch and feed/speed are available but don't agree.   If you want to leave off G33.1 rigid tapping for now, that's fine.  Your call.

sliptonic, 2022-12-19:
Regardless of whether G33.1 is done now or later, we should consider the operation to be a generic 'tapping' operation rather than a specific flavor.   In other words, we should build the gui and the logic assuming it'll be extended to other flavors of tapping later.

luvtofish, 2022-12-19:
Okay, I'll shoot for getting a PR opened for a generic Tapping Operation with the idea it could be extended to include other functionality including Rigid Tapping in the future. I'll get a new source tree fork going for this -- I have a PR open now for adding a new Dynapath_4060_Post processor -- I think that one is waiting on a review from you. Coincidentally, the existing "Dynapath" post processor that is included in the source is broken -- not sure if you were aware. It's missing many items that are needed to make it usable. I'm okay with replacing that one with the one I wrote -- your call. It was easier for me to start over than to pick up where it is at.

sliptonic, 2022-12-21:
Draft PR is at https://github.com/FreeCAD/FreeCAD/pull/8069

sliptonic, 2025-07-05:
This appears to be resolved by the merged PR.  Closing

closed 2025-07-05 by sliptonic