FreeCAD/FreeCAD #8021: [Issue] Redundant Z move to clearance height after every G80 cancel call

luvtofish, 2022-12-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=74371

### Version

0.21 (Development)

### Full version info

```shell
The Drilling.py module issues a G0 Z0.118 (clearanceHeight) after every G80 (cancel drill cycle) operation. This is simply not necessary since each canned cycle always returns the tool to either the clearance or safe heights. The following Drilling.py code snippet should be modified to remove the rapid Z move append after the G80 command:
```

### Subproject(s) affected?

Path

### Issue description

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GeneGH, 2022-12-11:
Any motion code (G1, G2, etc) after a canned cycle automatically cancels the canned cycle. In general G80 is never needed anywhere.

luvtofish, 2022-12-11:
The issue is the need to remove the Z rapid move after the G80 is outputted. It’s redundant and not needed.

luzpaz, 2023-01-08:
Any follow-up on this? Also please edit the OP and add your [Full About](https://wiki.freecad.org/About) info for completion sake.

closed 2025-01-12 by github-actions[bot]