FreeCAD/FreeCAD #8029: [Problem]

ger08061959, 2022-12-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31300 (Git)
Build type: Unknown
Branch: master
Hash: ce8816685b78785f951a6443cb4e0ceb79c7611a
Python 3.10.6, Qt 5.15.3, Coin 4.0.1, Vtk 9.1.20220328, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Estimate 0.1.2
  * sheetmetal 0.2.58
  * Glass
  * BEM
  * nurbs
  * Trails 2022.1.0
  * 3D_Printing_Tools
  * Ship 2022.4.11
  * WebTools
  * Nodes 0.1.1
  * BIMxBEM
  * Render 2022.2.0
  * bimtester
  * ose-piping
  * BIMBots
  * Manipulator 1.4.9
  * BIM 2021.12.0
  * dodo 1.0.0
  * frame 0.1.0
  * workfeature
  * SteelColumn
  * Assembly3 0.1.1
[/code]
```

### Subproject(s) affected?

Addon Manager

### Problem description

Network Manager Error: __reply_finished not called by a Qt signal
Traceback (most recent call last):
  File "/../FreeCAD/build/Mod/AddonManager/addonmanager_connection_checker.py", line 79, in cancel_network_check
    self.connection_checker.success.disconnect(self.launch)
AttributeError: 'ConnectionCheckerGUI' object has no attribute 'launch'

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2022-12-17:
Thanks for the report, fixed now.

closed 2022-12-17 by chennes

mconsidine, 2023-01-27:
Actually, I'm getting this error using code pulled today (via 'git pull').  I double checked to ensure that the change identified in
https://github.com/FreeCAD/FreeCAD/commit/403e0dc4770cf2989c6cf7d4aa77b824dcb419ec
is there and it is.  This is using a debug version of 0.21.  Happy to provide more info or run other tests if that can be pointed out to me.  I don't know if/where any logs are kept ...
mconsidine

mconsidine, 2023-01-27:
I just posted on the forum that my issue related to AddonManager and QtWebEngineWidgets has been resolved by removing a pip-installed version of PySide2 from a /.local folder.  It is unclear to me if this problem here is related to that.  I am still getting segmentation faults but will look for the Network Manager error...

commit 403e0dc4 (2022-12-17): Addon Manager: Correct func names on disconnect

Fixes #8029.