FreeCAD/FreeCAD #8034: [Issue] Program received signal SIGSEGV at startup

marcogiusti, 2022-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
FreeCAD 0.20.1, Libs: 0.20.1R (Debian testing package)

and

FreeCAD 0.20.1 Revision: 29410 (Git) (latest AppImage)
```

### Subproject(s) affected?

None

### Issue description

FreeCAD work fine until I plug/unplug an external monitor. After I do this I cannot start FreeCAD unless I reboot. I use [mons](https://github.com/Ventto/mons) to easily control xrandr.

The stacktrace using the debian package, when using the AppImage the output is similar:

```
$ freecad
FreeCAD 0.20.1, Libs: 0.20.1R
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3bf90) [0x7f206745af90]
#1  0x7f2067f00b00 in QBackingStore::beginPaint(QRegion const&) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xd0
#2  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x174099) [0x7f2068574099]
#3  0x7f20685a4fa8 in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x468
#4  0x7f2068562f5e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#5  0x7f2069c26388 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x88
#6  0x7f20678b1718 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#7  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x175155) [0x7f2068575155]
#8  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x177bc0) [0x7f2068577bc0]
#9  0x7f206858ea80 in QWidget::repaint(QRect const&) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x80
#10  0x7f206858eada in QWidget::repaint() from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x3a
#11  0x7f2068711709 in QSplashScreen::repaint() from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x9
#12  /usr/lib/freecad-python3/lib/libFreeCADGui.so(+0x773909) [0x7f2069f73909]
#13  0x7f206971583d in Base::ConsoleSingleton::NotifyLog(char const*) from /usr/lib/freecad-python3/lib/libFreeCADBase.so+0xbd
#14  0x7f2069715b13 in Base::ConsoleSingleton::Log(char const*, ...) from /usr/lib/freecad-python3/lib/libFreeCADBase.so+0x213
#15  0x7f2069bbba81 in messageHandler(QtMsgType, QMessageLogContext const&, QString const&) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x51
#16  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0xc3ae0) [0x7f20676c3ae0]
#17  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0xc3bf8) [0x7f20676c3bf8]
#18  0x7f206769051d in QMessageLogger::warning(char const*, ...) const from /lib/x86_64-linux-gnu/libQt5Core.so.5+0xaf
#19  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x615ba) [0x7f20618b55ba]
#20  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x62582) [0x7f20618b6582]
#21  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x643ab) [0x7f20618b83ab]
#22  0x7f2067f00567 in QBackingStore::resize(QSize const&) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xa7
#23  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x17358f) [0x7f206857358f]
#24  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x174a5a) [0x7f2068574a5a]
#25  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c116c) [0x7f20685c116c]
#26  0x7f2068562f5e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#27  0x7f2069c26388 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x88
#28  0x7f20678b1718 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#29  0x7f2067d3e73d in QGuiApplicationPrivate::processExposeEvent(QWindowSystemInterfacePrivate::ExposeEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x8d
#30  0x7f2067d11e1c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#31  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x6dc7a) [0x7f20618c1c7a]
#32  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x299) [0x7f206591f7a9]
#33  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x54a38) [0x7f206591fa38]
#34  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f206591facc]
#35  0x7f20679094b6 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x66
#36  /usr/lib/freecad-python3/lib/libFreeCADGui.so(+0x773909) [0x7f2069f73909]
#37  0x7f206971583d in Base::ConsoleSingleton::NotifyLog(char const*) from /usr/lib/freecad-python3/lib/libFreeCADBase.so+0xbd
#38  0x7f2069715b13 in Base::ConsoleSingleton::Log(char const*, ...) from /usr/lib/freecad-python3/lib/libFreeCADBase.so+0x213
#39  0x7f206971622a in Base::ConsoleSingleton::sPyLog(_object*, _object*) from /usr/lib/freecad-python3/lib/libFreeCADBase.so+0x6a
#40  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128413) [0x7f2068d28413]
#41  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8e) [0x7f2068cdebce]
#42  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x90fc) [0x7f2068c78cfc]
#43  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c70c6) [0x7f2068dc70c6]
#44  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xb9) [0x7f2068dc1b09]
#45  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x211899) [0x7f2068e11899]
#46  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyRun_StringFlags+0x75) [0x7f2068e129a5]
#47  0x7f206972b7a8 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/lib/freecad-python3/lib/libFreeCADBase.so+0x68
#48  0x7f2069bbc0ff in Gui::Application::runInitGuiScript() from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x3f
#49  0x7f2069bc1bd9 in Gui::Application::runApplication() from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x10c9
#50  freecad(+0x4c01) [0x558ffef3cc01]
#51  /lib/x86_64-linux-gnu/libc.so.6(+0x2718a) [0x7f206744618a]
#52  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f2067446245]
#53  freecad(+0x5ad1) [0x558ffef3dad1]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2022-12-17:
This is happening pretty deep inside Qt -- have you searched for issues in other packages using Qt? There's not much point in our submitting a bug report to them at this point, the version of Qt that you are compiled against is too old to receive any support from upstream.

luzpaz, 2022-12-21:
@marcogiusti can you post what `freecad --verbose --version`  returns for you ?

luzpaz, 2023-01-08:
@marcogiusti please respond to the above 2 questions from chennes and myself. TIA

mosfet80, 2023-01-09:
I think I have the same problem:

This is my configuration (debian 11)

Compiler: GNU, version: 12.2.0
prefix: /usr/local
bindir: bin
datadir: share
docdir: share/doc/FreeCAD
includedir: include
libdir: lib
cmake: 3.25.1
Compiling with Qt 5
PyCXX found:
  Headers:  /home/r/software/FreeCAD/src
  Sources:  /home/r/software/FreeCAD/src/CXX
  Version:  7.1.7
-- Found OCE/OpenCASCADE version: 7.6.3
-- OCE/OpenCASCADE include directory: /usr/include/opencascade
-- OCE/OpenCASCADE shared libraries directory: /usr/lib/x86_64-linux-gnu
Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
CMake Deprecation Warning at /usr/lib/x86_64-linux-gnu/cmake/vtk-9.1/vtk-config.cmake:75 (message):
  The new name for the 'vtkCommonCore' component is 'CommonCore'
Call Stack (most recent call first):
  cMake/FreeCAD_Helpers/SetupSalomeSMESH.cmake:29 (find_package)
  CMakeLists.txt:57 (SetupSalomeSMESH)

VTK components: CommonCore;CommonDataModel;FiltersVerdict;IOXML;FiltersCore;FiltersGeneral;IOLegacy;FiltersExtraction;FiltersSources;FiltersGeometry;IOMPIParallel;ParallelMPI;hdf5;RenderingCore;InteractionStyle;RenderingFreeType;RenderingOpenGL2
Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so  found components: OpenGL GLX 
Check for medfile (libmed and libmedc) ...
We guess that libmed was built using hdf5-openmpi version
Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.65") found components: filesystem program_options regex system thread date_time 
Set up to compile with Qt 5.15.7
Shiboken2Config: Using default python: .cpython-311-x86_64-linux-gnu
SHIBOKEN_PYTHON_INCLUDE_DIRS computed to value: '/usr/include/python3.11'
SHIBOKEN_PYTHON_LIBRARIES computed to value: ''
libshiboken built for RelWithDebInfo
PYTHON_CONFIG_SUFFIX: .cpython-311-x86_64-linux-gnu
libshiboken built for RelWithDebInfo
PySide 5.13.2 Python module found at /home/r/.local/lib/python3.11/site-packages/PySide2.

Found Matplotlib: /usr/lib/python3/dist-packages/matplotlib (found version "3.6.2")
Platform is 64-bit, set -D_OCC64
Performing Test _flag_found
Performing Test _flag_found - Success
Performing Test _flag_found
Performing Test _flag_found - Success
Performing Test _flag_found
Performing Test _flag_found - Success
Performing Test _flag_found
Performing Test _flag_found - Success
[E57] Revison ID: E57Format-2.2.1-x86_64-linux-gcc12
[E57] Building static library
Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: python311 
found Boost: 1_74
boost-incude dirs are: /usr/include
boost-python lib is: 
boost_LIBRARY_DIRS is: /usr/lib/x86_64-linux-gnu
Boost_LIBRARIES is: Boost::python
area module (for Path Workbench) will be installed to: lib
Performing Test _flag_found
Performing Test _flag_found - Success
Performing Test _flag_found
Performing Test _flag_found - Success
Coin3D doc is not installed

==============
Summary report
==============

[... 152 lines cut ...]
#93  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x55d3) [0x7f9375b0a7d3]
#94  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x26a6da) [0x7f9375c6a6da]
#95  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x17e1cf) [0x7f9375b7e1cf]
#96  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0xea) [0x7f9375b7e3ea]
#97  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x3af) [0x7f9375c9637f]
#98  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0xe178) [0x7f9375b13378]
#99  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x3cc73c) [0x7f9375dcc73c]
#100  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyEval_EvalCode+0xc1) [0x7f9375c68421]
#101  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2b6d89) [0x7f9375cb6d89]
#102  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyRun_StringFlags+0x6b) [0x7f9375cb7f3b]
#103  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x268938) [0x7f9375c68938]
#104  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1c9c52) [0x7f9375bc9c52]
#105  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyObject_Vectorcall+0x33) [0x7f9375b7d293]
#106  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x4a63) [0x7f9375b09c63]
#107  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x3cc73c) [0x7f9375dcc73c]
#108  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyEval_EvalCode+0xc1) [0x7f9375c68421]
#109  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2b6d89) [0x7f9375cb6d89]
#110  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyRun_StringFlags+0x6b) [0x7f9375cb7f3b]
#111  0x7f9376463f31 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/r/software/build/Freecad/lib/libFreeCADBase.so+0xd5
#112  0x7f9378ae5a0f in Gui::Application::runInitGuiScript() from /home/r/software/build/Freecad/lib/libFreeCADGui.so+0x3d
#113  0x7f9378ae78ca in Gui::Application::runApplication() from /home/r/software/build/Freecad/lib/libFreeCADGui.so+0x1ea8
#114  ./bin/FreeCAD(+0x133ff) [0x5653caa653ff]
#115  /lib/x86_64-linux-gnu/libc.so.6(+0x2718a) [0x7f937664618a]
#116  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f9376646245]
#117  ./bin/FreeCAD(+0x12621) [0x5653caa64621]

luzpaz, 2023-01-09:
@mosfet80 please remember to put code tags around output. In this case not having the output contained in tags resulted in spamming a whole host of other unrelated closed tickets. TIA

luzpaz, 2023-03-15:
@mosfet80 is this still occurring for you ?

mosfet80, 2023-03-15:
I fixed it by removing the downloaded libraries with the pip-review command. 
In particular I removed the .local/lib/pythonXX/ folder in my home directory

luzpaz, 2023-03-18:
@mosfet80 Can you pinpoint where the issues are specifically ?

marcogiusti, 2023-04-19:
> @marcogiusti can you post what `freecad --verbose --version` returns for you ?
> 
> Also would you consider running the latest development ([Weekly](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)) version appimage to see if you can recreate this problem ?

Sorry to be very late with this. Just downloaded the latest weekly release and I can not reproduce the issue. Before trying the latest weekly, I tried the official Debian testing package and AppImage I had installed before (`FreeCAD_0.20-1-2022-08-20-conda-Linux-x86_64-py310.AppImage`). They both crashed.

This is the output:
```
FreeCAD 0.21.0 Revision: 32867 (Git)

Library versions:
boost    1_74
Coin3D   4.0.0
Eigen3   3.4.0
OCC      7.6.3
Qt       5.15.8
Python   3.10.10
PySide   5.15.8
shiboken 5.15.8
SMESH    9.8.0.2
VTK      9.1.0
xerces-c 3.2.4
```

marcogiusti, 2023-04-19:
> So it's fixed on FreeCAD 0.21.0 Revision: 32867 (weekly AppImage)
> But broken on:
> * Debian testing build (which doesn't have a maintainer AFAIK)
> * FreeCAD_0.20-1-2022-08-20-conda-Linux-x86_64-py310.AppImage
>
> Did I get that right?

Correct

luzpaz, 2023-04-19:
@marcogiusti can you reproduce on v0.20.2 (which is the latest stable) ?

marcogiusti, 2023-04-19:
FreeCAD_0.20.2-2022-12-27-conda-Linux-x86_64-py310.AppImage is also broken.

luzpaz, 2023-04-20:
I assume that FreeCAD_0.20.2-2022-12-27-conda-Linux-x86_64-py310.AppImage is using Qt < v5.15.8