FreeCAD/FreeCAD #8051: [Issue]

rydb, 2022-12-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=74528

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29177 +233 (Git) Snap 245
Build type: Release
Branch: tag: 0.20.1
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

File formats

### Issue description

I am trying to export individual parts of the model linked at the top of my post below:

https://forum.freecadweb.org/viewtopic.php?f=3&t=74528

however, the model export for .dae, .stl, and .obj do not export the models relative to their origins. 

I think the export methods, for these file formats at least, export the model geometry relative to their position in the document, and do not respect the model's FreeCAD origin

is there a setting I can select to fix this or a workaround? Or is this a bug.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rydb, 2022-12-19:
I found a fix to this issue that should be added to the export functions for models, if you export them individually at least..

Add the following line to model exports

__obj__[|index_var|].Placement.Base = FreeCAD.Vector(0, 0, 0)

Models are exported relative to world origin, not relative to their origin. So by setting the exported __obj__'s vector position to world origin, the model is exported relative to both.

I have a macro version of the export for .dae. I made it only export one model at a time. Export a model with the python console open in your preferred file format, see how it exports it, and add the above line to your __obj__[i]] gets exported at:

```
    def export_self_as_dae(model, model_folder, name):
        """
        export self as a .dae model
        """
        dae_folder = "%s%s.dae" % (model_folder, name)
        #documentation of importDAE
        # https://github.com/FreeCAD/FreeCAD/blob/d35400aae339d71d5fc8c7f767e3be17687fae90/src/Mod/Arch/importDAE.py
        try:
            import importDAE
        except:
            print("WARNING: collada likely not installed, attempting fix using freecad.pip util that comes with snap install assuming user installed FreeCAD via snap(where error normally occurs)")
            os.system("freecad.pip install pycollada")
            import importDAE

        __obj__ = model

        #Set the object's placement to be at world origin since models are not exported relative to their origin, but to world origin.
        #by setting the object's position to world origin, the model will be exported relative to it's origin.
        __obj__.Placement.Base = FreeCAD.Vector(0, 0, 0)

        importDAE.export(__obj__, model_folder)
```

My code only exports one model and is part of a larger macro so im not able

luzpaz, 2022-12-20:
@yorikvanhavre you wrote the FreeCAD exporter for blender script (https://gist.github.com/yorikvanhavre/029f6fcce9f4d0e62fb6163804b7f80d). Do you mind weighing in on this ?

onekk, 2022-12-20:
It is not a bug, at least for STL that is not  meant to export in "world coordinates", as it is made to export meshes made for printing, see maybe:

https://en.wikipedia.org/wiki/STL_(file_format)

Probably for DAE files this could have a reason, but for OBJ and STL files exporting in "world coordinates" is more an hassle than an improvement.

Regards

Carlo D.

chennes, 2022-12-20:
This is definitely working as intended -- if you'd like to propose adding a feature to do this procedure automatically under some circumstances, I'd discuss that in the forum.

closed 2022-12-20 by chennes