FreeCAD/FreeCAD #8059: [Feature Request] <title>

SarenT, 2022-12-18:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=10&t=65213

### Subproject(s) affected?

Spreadsheet

### Idea description

Sorry for my ignorance. This issue was probably reported for a long while. Once a file gets bigger with many bodies and features with references to a spreadsheet, each edit to the spreadsheet takes quite some time (to recompute). I use a single spreadsheet. I think that FreeCAD tries to update everything upon changing the spreadsheet. Even if new variables are added to the spreadsheet and of course no object depends on the newly added aliases.
Perhaps, would it be better to use a reactive programming approach? I know this from R Shiny (an R library, a data analysis programming language). But the logic is simple and could be useful for a dependency graph. Let's say we have 5 objects (A, B, C, D and E).

C depends on A and B

E depends on D

A keeps track of C. Once A is updated, A reports this change to C and C gets recomputed.

D remains idle as E doesn't report anything.

(Small note: in R Shiny this is only done if C gets called e.g. displayed and C actually asks A and B if there are any changes. Such calls propagate through the dependency tree. I guess this model wouldn't work very well for FreeCAD.)

So a reactive programming approach should reduce the recompute time... Only relevant objects get recomputed. I wonder, what others think about this?

### Anything else?

More on the Shiny reactive programming: [https://shiny.rstudio.com/articles/reactivity-overview.html](https://shiny.rstudio.com/articles/reactivity-overview.html)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-12-20:
Is this a dupe of #8061 ?Also this ticket is missing About info

SarenT, 2022-12-21:
I did not try with the 0.21. I refer to a more general reoccurring issue in FreeCAD with large projects. Any tiny edit in the spreadsheet leads to long unnecessary recomputations, which can be avoided.

luvtofish, 2023-01-08:
About info added.

adrianinsaval, 2023-05-11:
FreeCAD already does this but at the feature level, not the property level, that's why all objects that depend on the spreadsheet are recomputed.

adrianinsaval, 2023-05-11:
This is not so easy to solve because it's tricky to figure out inter-property relationships, many features modify one property when another is changed but the software currently has no way of knowing this a priori so even if just one property of an object has a dependency on just one property of another it has to be treated as if it's the dependent on the entire object.

SarenT, 2023-12-14:
Out of curiosity, what are properties? e.g. each cell with an alias? For features is it pretty much clear. But I think that it is similarly clear, which feature depends on which spreadsheet cell/alias. The dependency graph can resolve it as well.

luzpaz, 2023-12-24:
bumping for progress