FreeCAD/FreeCAD #8061: [Issue] Freecad stop responding for ~20 seconds when first using a spreadsheet

shaise, 2022-12-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=74554

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30716 (Git)
Build type: Release
Branch: master
Hash: cbd19631428f1d9eb4faae30220551608a648589
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.27
  * sheetmetal 0.2.59
```

### Subproject(s) affected?

None

### Issue description

Possible issue with spreadsheets on FreeCad V0.21:
Seems this is true only for Conda version on Windows platform.

It looks like spreadsheet WB takes huge amount of memory.
How to reproduce:
1. Start FreeCad (v0.21)
    task manager reports 200MB mem usage
[image]

2. Select Spreadsheet WB
3. Create new spreadsheet
4. Click on any cell.
   freeCad hangs for several seconds, after which task manager reports 4GB memory usage.
[image]

When I do the same on V0.20 release, Spreadsheet takes nothing, and no PC hang.

Thanks!

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30716 (Git)
Build type: Release
Branch: master
Hash: cbd19631428f1d9eb4faae30220551608a648589
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.27
  * sheetmetal 0.2.59
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

shaise, 2022-12-20:
The problem is that there are no longer official builds of FreeCad 0.21 with  libpack. My current workaround is to use LinkStage3 version as there is still a libpack version there.

luzpaz, 2022-12-20:
> The problem is that there are no longer official builds of FreeCad 0.21 with libpack

@donovaly :point_up:

donovaly, 2022-12-21:
I will make a LibPack release for 0.21 later today.

donovaly, 2022-12-22:
> I will make a LibPack release for 0.21 later today.

Here is one: https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD-0.21.0dev-31391-Installer.exe

shaise, 2022-12-24:
Confirmed, No issue with spreadsheets on LibPack release. Thanks!

PAS-DRAFTER, 2023-01-11:
This problem is driving me insane.  I can't tell if I have the "fix" installed on my machine or not.  I tried downloading and installing the link above, but I'm not sure if it did anything.

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.5
  * CurvedShapes 1.0.4
  * Curves 0.6.5
  * fasteners 0.4.52
  * sheetmetal 0.2.60
[/code]

luzpaz, 2023-01-12:
What about as a temp workaround you enable '[Skip recomputes](https://wiki.freecadweb.org/Spreadsheet_Workbench#Complex_models_and_recomputes)' ?

PAS-DRAFTER, 2023-01-12:
Well, I gave that a shot however, if I try to Ctrl-C to copy a cell or a group of cells, it still goes into hang mode for 20 or so seconds.  It only seems to happen when I try to copy and paste something.

PAS-DRAFTER, 2023-01-12:
Okay, I figured some more info out.  I was getting so fed up with this problem, I decided to go back to an older version that was working fine on my other computer.  I was running the FreeCAD_weekly-builds-31340-2022-12-18-conda-Windows-x86_64-py310 on that computer and I DID NOT have the copy problem in Spreadsheet when I tried to copy a cell or block of cells.  That computer is running Windows 10.

I then put that same version on my new machine.  It is running Windows 11 and is brand new.  I just installed things on it last week.  I opened the same files and tried the same copy and paste of the cells, and this new computer has the stall when I select a cell or group of cells and hit <Control> C or right click and Copy.  So, it appears to me that there is a problem running on a clean version of Windows 11.

Another thing I noticed.  When I copy the About information in FreeCAD to the clipboard and paste it below from my new computer, it says the OS is Windows 10 Version 2009.  However, when I go to the System>About page, it says I have 

Edition: Windows 11 Pro
Version 22H2
Installed on:  12/‎19/‎2022
OS build: 22621.963

I'm not sure why the FreeCAD info and that do not agree.

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31340 (Git)
Build type: Release
Branch: master
Hash: 856af2a12287a2d7c0d5ab23646be43af26b14b1
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.5
  * CurvedShapes 1.0.4
  * Curves 0.6.5
  * fasteners 0.4.52
  * sheetmetal 0.2.60
[/code]

Above it talks about a LibPack not being provided that is causing the problem.  I tried downloading and running the linked build, but that did not seem to fix the problem.  How do I know the LibPack got installed properly?

Hope this help solve the issue.  I guess I'll run on the old computer for now.  However, that is a pain because it only has one display.  If anyone can come up with a solution, I'll be happy to test.  If you need more information about my system, let me know.

Thanks

donovaly, 2023-01-13:
> Another thing I noticed. When I copy the About information in FreeCAD to the clipboard and paste it below from my new computer, it says the OS is Windows 10 Version 2009. However, when I go to the [...]

We are aware of this. The point is that Windows changed the registry key for the version. There is no solution available that can determine all the Windows versions.
However, for the next FreeCAD version we can have a look since Win 7 and 8 are no longer active (for Microsoft). Therefore there is no big problem, when we switch to the new registry scheme and as result cannot determine the Win 7/8 versions.

PAS-DRAFTER, 2023-01-14:
I just uninstalled all FreeCAD versions including user data.

I installed the suggested version above again.

I rebooted my computer.

I started FreeCAD.

I created a new Spreadsheet and didn't enter any data so it is completely blank.

I selected cells A3:F8 and then hit Control - C.

The program stalled out for 30 seconds.  During this time I heard my computer fan kick in as if it was causing the processor to work hard.  During this time, I was unable to do anything with FreeCAD.  Eventually it came back and I could navigate around and enter data.

I selected a single cell (A1) and right clicked and chose Copy.  FreeCAD and my computer did the same thing.

I am not having this problem on my Windows 10 machine, only on this new Windows 11 machine.  

BTW, the Add On Manager does not work with this version.

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31391 (Git)
Build type: Release
Branch: master
Hash: 0ab2608a429f3a5ee9a63da9b66d13557655d7e4
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

casterle, 2023-01-15:
> > I will make a LibPack release for 0.21 later today.
> 
> Here is one: https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD-0.21.0dev-31391-Installer.exe

Thank you for that - the spreadsheet works properly. However, when I unzip a weekly build into the folder the spreadsheet is once again broken. Would it be possible to make a LibPack exe as part of the weekly build release process?

PAS-DRAFTER, 2023-01-16:
I learned something new again today.  My coworker had a computer with Windows 10 and they upgraded to Windows 11.  I just installed the latest release version (31391) on it.  I tried to copy a cell in Spreadsheet and it appeared to work properly.

So again, it looks like the problem is related to something Windows 11 is not installing on a fresh release that was there in Windows 10 and continued to be there after a Windows 11 upgrade.

jag45, 2023-01-25:
Just to give another view.

I don't see how this has anything to do with the version of windows.
I am running windows 10. The spreadsheet works ok with version 0.20.2, however running version 0.21.31641 causes the spreadsheet to become unresponsive when clicking on a cell.

I've tried with an existing spreadsheet and also created a new blank spreadsheet, both become unresponsive as above.

Jon

FREECAD 0.21.0.31641
CODE: [SELECT ALL]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31641 (Git)
Build type: Release
Branch: master
Hash: d28d63b87b60161419c6c0c532fbbfaed96926b8
Python 3.10.8, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * A2plus 0.4.60i
  * Assembly4 0.12.5
  * ExplodedAssembly
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * RemBench
  * sheetmetal 0.2.60
  * Woodworking 0.21.30486
FREECAD 0.20.2.29177 +426
CODE: [SELECT ALL]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * A2plus 0.4.60i
  * Assembly3 0.11.4
  * Assembly4 0.12.5
  * ExplodedAssembly
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * RemBench
  * sheetmetal 0.2.60
  * Woodworking 0.21.30486
[Top](https://forum.freecadweb.org/viewtopic.php?f=10&t=74004&p=655556&hilit=weekly+build+menu#top)

shaise, 2023-01-25:
And, for some reason, only in Conda version.

PAS-DRAFTER, 2023-01-25:
After looking back above, I see that it is possible that these may be two entirely different issues.  I have attached my issue to this one because it has similar symptoms where the Spreadsheet WB stalls out.  I now realize that could be a false assumption.  What was causing the stall on this original ticket and what is causing the stall out for me are two separate events.

The original post is about the problem you encounter where Spreadsheet WB stalls when you click on a cell after opening one, and it isn't Windows version specific.  It appears that they may have solved that issue by adding the LibPack in the release they gave above.  I am unsure if they have yet added that to the full release.  Have you tried the version provided in the link above?[https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD-0.21.0dev-31391-Installer.exe?](url)

I may have run across this original issue with the Windows version initially (I'm afraid it has been too long for me to remember the entire chain of events).  However, even with the linked version above, I still have a problem with the Spreadsheet WB stalling out when I try to copy cells (and occasionally when I just try to copy the alias).  I also have tested the same version of the software on my brand new Windows 11 machine and my older Windows 10 machine.  The Windows 11 machine has the copy delay, and the Windows 10 does not.  I've also tested on a Windows 10 machine upgraded to Windows 11, and the problem did NOT happen there.  That is what has me thinking the problem I'm having is related to the Windows version.

It occurs to me that maybe I should be opening a separate ticket on the issue I'm that I am having.  There is already a discussion on the forum with the symptoms I'm having [https://forum.freecadweb.org/viewtopic.php?t=73520](url).  Please let me know if I should and I will.

adrianinsaval, 2023-01-26:
> I have created the new issue that describes the alternative way the stall is started at: [https://github.com/FreeCAD/FreeCAD/issues/8265](url)

your link is broken, instead you can refer to other issues within the same repository like this #8265

PAS-DRAFTER, 2023-01-26:
> your link is broken, instead you can refer to other issues within the same repository like this #8265

@adrianinsaval, thank you for letting me know.  I did not format them properly.  I have never used them in github so I had to go find the documentation.  I have them fixed in my original post now.  Always learning new things.  :-)

jag45, 2023-01-28:
Tried the 0.21.31391 with lib pack as above, it does solve the problem for me. I can work with the spreadsheets as normal, but the weekly builds are still a problem as stated.

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31391 (Git)
Build type: Release
Branch: master
Hash: 0ab2608a429f3a5ee9a63da9b66d13557655d7e4
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * A2plus 0.4.60i
  * Assembly3 0.11.4
  * Assembly4 0.12.5
  * ExplodedAssembly
  * QuickMeasure 2022.10.28
  * sheetmetal 0.2.60
  * Woodworking 0.21.30486
[/code]

adrianinsaval, 2023-05-11:
can't reproduce anymore on latest conda weekly
```yaml
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.33110 (Git)
Build type: Release
Branch: master
Hash: 18e40941ceb36c52701a7c74ea26043ef3385701
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C) [ OS: Spanish/Paraguay (es_PY) ]
Installed mods: 
```

can you reproduce @shaise ?

PAS-DRAFTER, 2023-05-11:
I have not seen a delay in spreadsheet for quite some time in either the loading spreadsheet WB or when I try to copy instances.  I even went back and turn "Suggested actions" on and it still works.  It has been mostly a joy to use since the fixes were incorporated.

`
OS: Windows 11 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.33110 (Git)
Build type: Release
Branch: master
Hash: 18e40941ceb36c52701a7c74ea26043ef3385701
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.5
  * CurvedShapes 1.0.4
  * Curves 0.6.5
  * fasteners 0.4.53
  * OpticsWorkbench 1.0.9
  * sheetmetal 0.2.60
`

adrianinsaval, 2023-05-11:
I assume this was fixed by #8320 then? @chennes and @acpopescu IIRC that PR disabled some accessibility features in order to avoid this problem, should we create an issue to track if we can ever re enable those features?

chennes, 2023-05-11:
A "[Problem] Spreadsheet cannot use Windows Accessibility Features" would be reasonable.

closed 2023-05-11 by chennes

shaise, 2023-06-27:
Sorry for the late response: Confirmed the issue is fixed for the latest 0.21 releases.