FreeCAD/FreeCAD #8062: [Feature Request] TechDraw White on Black for Visually Impaired

WandererFan, 2022-12-19:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://www.forum.freecadweb.org/viewtopic.php?f=8&t=74222

### Subproject(s) affected?

Techdraw

### Idea description

Black text on white background is reported to be difficult to work with for some visual impairments.   White text on black background is better for some. 

### Anything else?

Description of Work:
Text colour is already a preference (Normal colour).
Page colour is set in QGVPage drawbackground.
New preference required for page colour.
Template colours are set in the Svg code and not controllable by program code.  Will need matched pairs of Svg templates, one with light text and one with dark.  Alternative would be to require templates to have a text colour property.

Template would need to be switched at print time to get dark on light print.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-12-19:
> Template would need to be switched at print time to get dark on light print.

maybe this should be optional?

WandererFan, 2022-12-20:
"maybe this should be optional?"  
For a PDF, maybe.  My printer doesn't have white ink. :)

luzpaz, 2022-12-20:
This is also related to #6862

WandererFan, 2023-01-04:
implemented by PR #8123 / d63ac0f7f6.

closed 2023-01-04 by WandererFan