FreeCAD/FreeCAD #8064: [Issue]

mhindi2, 2022-12-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Fedora Linux 35 (Workstation Edition) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.20.1.Unknown
Build type: Unknown
Python 3.10.6, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.3, OCC 7.5.0
Locale: English/United States (en_US)
Installed mods: 
  * Pyramids-and-Polyhedrons
  * fasteners 0.4.25
  * GDML 2.0.0
  * Manipulator 1.4.9
  * BIM 2021.12.0
  * Assembly4 0.12.4
[/code]
```

### Subproject(s) affected?

Path

### Issue description

[image]
One of the letters in a vcarve of a Shape string is not rendering at all (the "e"  in Kitchen, please see attached image.)
The font is urw-base35/Z003-MediumItalic.otf; the size is 10.00 mm high. I've also observed a "B" missing in a different font, but not this one.

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-12-20:
@mhindi2 can you please zip the .FCStd file and attach it to the ticket. Can you x-post this to the Path forum.freecadweb.org to make sure this isn't a 3rd party dependency or user-error ?

mhindi2, 2022-12-20:
[missing_letter.zip](https://github.com/FreeCAD/FreeCAD/files/10269803/missing_letter.zip)
Here it is. The tool is a 60 degree V-bit, with a 3 mm diameter.

mhindi2, 2022-12-21:
Discussions about this issue in the Path forum:
https://forum.freecadweb.org/viewtopic.php?f=15&t=74600&p=648804#p648804

mlampert, 2022-12-21:
First off, when I tested this last night vcarve on the latest master the letter `e` is carved correctly, however the letter `h` is missing. As already indicated by Gene in the forum discussion, whatever happens here is data driven.

The issue seems to be with the voronoi diagram construction. For a regular letter of the sample the `finite` `primary` edges of the constructed diagram look like this for `c`:
[image]
and like this for the letter `e`:
[image]
For the letter `h` I get this however:
[image]
All of those edges get discarded afterwards in cleaning up the diagram - they are presumed to be `exterior` edges, which are either `infinite` or linked to an `infinite` edge.

At this point I suspect that either I intruduced some memory corruption in the binding or the voronoi algorithm itself runs into some instability.

mhindi2, 2022-12-22:
What height are using for the letters? I'd like to be able to test the 'h' myself, but I won't be able to look at anything until I return from travels in 10 days or so. I can give the algorithm a look after I get back, but have to tend to some open issues that I am helping on in the GDML workbench first.

mlampert, 2022-12-22:
I've been using your file as is - the only change I made was to remove your `Job` and create a new one - there was an incompatible refactor between your version of FC and `master`.

Not sure which version of `libboost-dev` was used to build your version. I'm on `libboost1.74-dev`.

mlampert, 2022-12-22:
The letter having the issue depends on the origin for the `Job`. If the origin is the _top left_ corner the letter `e` is missing and if the origin is the _top right_ the letter `h` is missing.
If I use either of the _bottom_ corners I don't get any output at all.

mlampert, 2022-12-22:
Internally I scale all coordinates by `1000` - IIRC the reasoning was it scales 1um to an integer, being precise enough. It turns out that scaling has a major impact on the construction of the voronoi diagram. Using a scale of `973` it seems to generate the proper voronoi edges - but then one of the edges in the `n` results in an _"Unknown C++ exception"_ - which might not be related to the issue at hand but represent a bug in it's own right. More testing is needed.
[image]

mhindi2, 2022-12-22:
Wow! Obviously the median line should not depend on the origin. Looks like
there is a test for parallel lines somewhere whose result might depend
sensitively on the numerics (and hence the origin). I will attempt to look
at the code when I get a chance.

mlampert, 2022-12-23:
The voronoi diagram creates 2 types of artifacts, straight edges and parabolas. Parabolas are defined by a straight segment and a point. In order to get the shape of the parabola we have to find the orthogonal projection of the point on the segment. We get the _Unknown C++ exception_ when the point is on the line - this is the reconstruction of one of those instances:
[image]
Either this is a degenerated case and the _parabola_ should actually be a straight edge - or - the segment we're getting here is perpendicular to what the segment should be.

I'll have to ignore this case and isolate the edge in question in order to figure out what _should_ happen here.

@mhindi2 in case you want to play with the implementation - you set the scale factor in the constructor `Path.Voronoi.Diagram()` in `Vcarve.py`. The part with the exception (a result of doing arithmetic with `nan` values) happens in `VoronoiEdgePyImp.cpp` in the function `toShape()`.

mhindi2, 2022-12-23:
My recollection is that all conics (circles, ellipses, parábolas and
hyperbolas) can be defined as the set of points the ratio of whose
distances from a line (I believe that is called the directrix) and a point
is some constant. For a parabola, that ratio is one. The ellipses and
hyperbolas correspond to the ratio less than one and greater than one,
respectively, or may be vice versa, I don't remember).  I think the case of
the point being on the line itself corresponds to a line perpendicular to
the directrix, because that gives the distance to the point equal to the
distance to the line, hence a constant ratio ( the case of a plane passing
through the vertex of the cone). So upshot is I believe the desired
"parabola" in that case is a line perpendicular to the given line.

mhindi2, 2022-12-23:
To expand a little on the geometry, the point of the parabola where the
derivative is zero "the lowest point" passes mid way between the line and
the point. As the point gets close to the line the parabola becomes sharper
and narrower, and when the point is on the line, the parabola becomes a
line perpendicular to the directrix. (I am recollecting this stuff from my
highschool days, more than 50 years ago; that was our starting point on the
study of conics - will double check the terminology on Wikipedia when I get
a chance.)

mhindi2, 2022-12-23:
So, an obvious suggestion, when the point is closer to the line than some
threshold small distance, skip the parabola and generate a line
perpendicular to the segment.

br4243, 2022-12-28:
This sounds suspiciously like something I fixed for a similar problem last night, and was just investigating how to report as a bug with an attached solution.  For the problem I had, the implementation of orthognalProjection() in src/Mod/Path/App/VoronoiEdgePyImp.cpp does not test for a zero length segment before the orthogonal projection of the point, resulting in a divide by zero.  In my case, the segment was very short from Python's perspective in Path/Op/Vcarve.py's insert_many_wires(), and ended up being a zero length segment in orthognalProjection().  I added a check for that:
$ git diff
diff --git a/src/Mod/Path/App/VoronoiEdgePyImp.cpp b/src/Mod/Path/App/VoronoiEdgePyImp.cpp
index efe68022bd..e5fa98d2c2 100644
--- a/src/Mod/Path/App/VoronoiEdgePyImp.cpp
+++ b/src/Mod/Path/App/VoronoiEdgePyImp.cpp
@@ -67,7 +67,9 @@ namespace {
     // calculate the orthogonal projection of p onto s
     // ((p dot s) / (s dot s)) * s (https://en.wikibooks.org/wiki/Linear_Algebra/Orthogonal_Projection_Onto_a_Line)
     // and it back by original offset to get the projected point
-    double proj = (p.x() * s.x() + p.y() * s.y()) / (s.x() * s.x() + s.y() * s.y());
+    double proj = 0;
+    if (length(segment) != 0)
+      proj = (p.x() * s.x() + p.y() * s.y()) / (s.x() * s.x() + s.y() * s.y());
     Voronoi::point_type pt;
     {
       pt.x(offset.x() + proj * s.x());
This resolved the issue I had, which was an infinite loop while generating the path for a ShapeString.

mlampert, 2022-12-29:
That does seem to be a different issue. Applying above patch does not fix this issue. If you have a sample document that exposes your issue could you file a separate issue here and describe what's going on please?

mhindi2, 2022-12-29:
If it is not too much trouble, could you please tell me how you generated
those figures? If it is, I'll figure something out.

mlampert, 2022-12-30:
@mhindi2  There are 2 parts to it, modify Vcarve.py to keep track of the voronoi diagrams:

```diff
markus@minzla:~/projects/FreeCAD/dev/FreeCAD$ git diff src/Mod/Path/Path/Op/Vcarve.py
diff --git a/src/Mod/Path/Path/Op/Vcarve.py b/src/Mod/Path/Path/Op/Vcarve.py
index 15d98d7017..dbf16f347c 100644
--- a/src/Mod/Path/Path/Op/Vcarve.py
+++ b/src/Mod/Path/Path/Op/Vcarve.py
@@ -317,6 +317,8 @@ class ObjectVcarve(PathEngraveBase.ObjectOp):
                 wires = _sortVoronoiWires(wires)
             voronoiWires.extend(wires)
 
+            self.vd.append(vd)
+
         if _sorting == "global":
             voronoiWires = _sortVoronoiWires(voronoiWires)
                                                                                                
@@ -335,6 +337,8 @@ class ObjectVcarve(PathEngraveBase.ObjectOp):                               
         """opExecute(obj) ... process engraving operation"""                                   
         Path.Log.track()                                                                       
                                                                                                
+        self.vd = []                                                                           
+                                                                                               
         if not hasattr(obj.ToolController.Tool, "CuttingEdgeAngle"):                           
             Path.Log.error(                                                                    
                 translate(                                                                     
markus@minzla:~/projects/FreeCAD/dev/FreeCAD$ 
```

Once you did a recompute you can then inspect the resulting cells, edges and vertices of each voronoi diagram. There is one for each face, so there's an extra voronoi diagram for the dot of the 'i'.

Displaying all primary edges I did with this function:
```python
def vd_show(vd):
    for i, l in enumerate("KiItchen"):
       for j, e in enumerate([s for s in vd[i].Edges if s.Color == 0]):
           Part.show(e.toShape(), f"{l}{j}")
```

This'll run for a while - it's not very efficient. The `Color` values are defined in `Vcarve.py` in case you want to look at edges that have already been discarded (colored). There is a post processing stage after this which filters out all edges that are deemed to be "artifacts of linear approximation" for the op. So don't worry if it gets noisy. The main thing is that the edges that _should_ come out for the operation are there in the first place.

mhindi2, 2022-12-30:
Great! Thanks a lot. I am finally back from my travels and will try to
understand what's going on.

mhindi2, 2022-12-30:
@mlampert 
I am trying to compile the latest FreeCAD. I am encountering many misisng PathToolController.xxxx modules after launching the newly compiled FreeCAD. The file https://github.com/FreeCAD/FreeCAD/tree/master/src/Mod/Path/Path/Tool)/__init__.py is 0 bytes long. Is this right?

mhindi2, 2022-12-30:
Typical message is
12:00:46  Traceback (most recent call last):
  File "/usr/lib64/python3.10/site-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'PathScripts.PathJob'
12:00:46  Traceback (most recent call last):
I am assuming some directory setup problems on my system.

br4243, 2022-12-30:
I was curious about the numeric instability/memory corruption suggested by @mlampert, so I printed the segments from which the Voronoi diagrams are constructed, v.getSegments(), for each face of the ShapeString.  The vertices all looked reasonable and I don't see any reason for VB to be numerically unstable working with them.  Interestingly enough, the segments from which the VDs are built form closed polygons for all but the problematic letter, and I'm wondering if that is somehow confusing VoronoiBuilder, or maybe the lambda for the EXTERIOR2 rejection?  VB has no closure requirement, and should be able to build a diagram from arbitrary [non-overlapping segments](https://www.boost.org/doc/libs/1_81_0/libs/polygon/doc/voronoi_main.htm), as far as I know.  Here are the starts and ends of the segment lists for each face, in case this is useful:
Face 0
<code>
(Vector (-16.15, -21.061666666666667, 0.0), Vector (-16.512955729166666, -21.555432942708336, 0.0))
(Vector (-16.512955729166666, -21.555432942708336, 0.0), Vector (-16.83958333333333, -21.960234375000002, 0.0))
snip
(Vector (-15.866666666666667, -23.17, 0.0), Vector (-16.14999999999998, -21.061666666666817, 0.0))
(Vector (-16.14999999999998, -21.061666666666817, 0.0), Vector (-16.15, -21.061666666666667, 0.0))
</code>
ends up with 13 wires

Face 1
<code>
(Vector (-12.625, -17.003333333333334, 0.0), Vector (-13.058333333333334, -17.245, 0.0))
(Vector (-13.058333333333334, -17.245, 0.0), Vector (-14.132682291666665, -17.848776041666667, 0.0))
snip
(Vector (-12.724999999999998, -17.745, 0.0), Vector (-12.625, -17.00333333333336, 0.0))
(Vector (-12.625, -17.00333333333336, 0.0), Vector (-12.625, -17.003333333333334, 0.0))
</code>
ends up with 14 wires

Face 2
<code>
(Vector (-13.599999999999998, -3.961666666666668, 0.0), Vector (-14.577083333333334, -4.721041666666668, 0.0))
(Vector (-14.577083333333334, -4.721041666666668, 0.0), Vector (-15.366666666666667, -5.286666666666667, 0.0))
snip
(Vector (-13.441666666666666, -4.295000000000002, 0.0), Vector (-13.599999999999998, -3.9616666666666713, 0.0))
(Vector (-13.599999999999998, -3.9616666666666713, 0.0), Vector (-13.599999999999998, -3.961666666666668, 0.0))
</code>
ends up with 3 wires

Face 3
<code>
(Vector (-4.891666666666666, -16.845000000000002, 0.0), Vector (-6.152213541666669, -17.592916666666667, 0.0))
(Vector (-6.152213541666669, -17.592916666666667, 0.0), Vector (-7.051041666666666, -18.084583333333335, 0.0))
snip
(Vector (-4.94166666666667, -17.486666666666668, 0.0), Vector (-4.891666666666673, -16.84500000000005, 0.0))
(Vector (-4.891666666666673, -16.84500000000005, 0.0), Vector (-4.891666666666666, -16.845000000000002, 0.0))
</code>
ends up with 12 wires

Face 4:
<code>
(Vector (3.616666666666667, -16.736666666666668, 0.0), Vector (2.9147135416666643, -17.252649739583333, 0.0))
(Vector (2.9147135416666643, -17.252649739583333, 0.0), Vector (2.265104166666667, -17.70046875, 0.0))
snip
(Vector (3.56666666666667, -17.595000000000002, 0.0), Vector (3.616666666666667, -16.736666666666682, 0.0))
(Vector (3.616666666666667, -16.736666666666682, 0.0), Vector (3.616666666666667, -16.736666666666668, 0.0))
</code>
ends up with 11 wires

Face 5:
<code>
(Vector (4.266666666666666, -20.103333333333335, 0.0), Vector (6.083333333333336, -19.361666666666668, 0.0))
(Vector (6.083333333333336, -19.361666666666668, 0.0), Vector (7.591666666666669, -13.395000000000001, 0.0))
snip
(Vector (6.725000000000001, -9.386666666666668, 0.0), Vector (4.266666666666758, -20.103333333332955, 0.0))
(Vector (4.266666666666758, -20.103333333332955, 0.0), Vector (4.266666666666666, -20.103333333333335, 0.0))
</code>
ends up with 12 wires

Face 6 - the one that doesn't produce output:
<code>
(Vector (23.275000000000006, -16.470000000000002, 0.0), Vector (22.549169921875006, -17.095537109375, 0.0))
(Vector (22.549169921875006, -17.095537109375, 0.0), Vector (21.902734375, -17.619609375000003, 0.0))
snip
(Vector (18.653743489583334, -14.717233072916667, 0.0), Vector (17.741666666666703, -15.344999999999978, 0.0))
(Vector (17.741666666666703, -15.344999999999978, 0.0), Vector (17.741666666666667, -15.345, 0.0))
</code>
ends up with 0 wires, hence no generated medial line

Face 7
<code>
(Vector (26.275, -19.478333333333335, 0.0), Vector (27.791666666666664, -13.311666666666667, 0.0))
(Vector (27.791666666666664, -13.311666666666667, 0.0), Vector (27.87083333333333, -13.259583333333335, 0.0))
snip
(Vector (24.358333333333334, -20.236666666666668, 0.0), Vector (26.274999999999935, -19.47833333333336, 0.0))
(Vector (26.274999999999935, -19.47833333333336, 0.0), Vector (26.275, -19.478333333333335, 0.0))
</code>
ends up with 16 wires

mhindi2, 2022-12-30:
@br4243 
I am not 100% sure what you mean by closed polygon - I am assuming topologically without a hole. In the discussions in the forum (https://forum.freecadweb.org/viewtopic.php?f=15&t=74600&p=648804#p648804)  I had pointed out that the letter "a", which topologically is the same as the "e", i.e., has one hole produces a correct medial path, so I doubt that is the issue.

mhindi2, 2022-12-31:
@mlampert 
So I think I can now access the code and begin testing.  Atached is a plot showing the vd edges with s.Color==0 and s.Color==2. I deleted the previous job and created a new one. The current FreeCAD compiled from source is:

[code]
OS: Fedora Linux 35 (Workstation Edition) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31452 +3 (Git)
Build type: Unknown
Branch: Vcarve-tests
Hash: aa25f0565a07d0bf58c16e92136e0aa4b78bd02d
Python 3.10.8, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.3, OCC 7.5.0
Locale: English/United States (en_US)
Installed mods: 
  * Pyramids-and-Polyhedrons
  * fasteners 0.4.25
  * GDML 2.0.0
  * Manipulator 1.4.9
  * BIM 2021.12.0
  * Assembly4 0.12.4
[/code]

(Note I made a new branch locally to do the testing).

[image]
[image]
[image]

Note that the "Kitchen" now carves correctly, but vd_show() is not showing the "n" in the s.Color=0 edges.
Note also that in your original vd_show(vd), the string is "KiItchen",  and not "Kitchen" (or may be t was "Kitch en", so the "h" might have matched the edges of the "e" in vd.Edges.

I don't know why all the edges are not showing in vd_shiw(vd), but they seem all to be vcarved. Will have to repeat this tomorrow and test agian with the stock Fedora35 FreeCAD version that I had before as well.

mhindi2, 2022-12-31:
Another test. The strings have all 26 letters, upper case and lower case. Note that the "I" in "BLITZ" does not have a medial line, while the "I" in "QUICK" in the same string does.
[image]
[Kitchen-Vcarve-test.zip](https://github.com/FreeCAD/FreeCAD/files/10326879/Kitchen-Vcarve-test.zip)

mhindi2, 2022-12-31:
Sorry, zipped the wrong file for the above test.
[Kitchen-Vcarve-test-pangrams.zip](https://github.com/FreeCAD/FreeCAD/files/10326889/Kitchen-Vcarve-test-pangrams.zip)

luzpaz, 2022-12-31:
@mhindi2 you can edit the post you made and correct it (instead of keeping the faulty test)

mlampert, 2022-12-31:
Sounds like you're trying to load a Path file from before the big
refactoring. You'll have to delete the Job and create a new one.

see https://forum.freecadweb.org/viewtopic.php?f=15&t=71610

On Fri, 30 Dec 2022 12:06:02 -0800
Munther Hindi ***@***.***> wrote:

> Typical message is
> 12:00:46  Traceback (most recent call last):
>   File
> "/usr/lib64/python3.10/site-packages/shiboken2/files.dir/shibokensupport/__feature__.py",
> line 142, in _import return [...]

br4243, 2022-12-31:
@mhindi2
I was referrring to the last point of the polygon not matching the first, but your point about the circle in the center of the letter is a good one.  I looked back at the vertices of the segments for that letter and there are actually two closed polygons, one for the outside of the letter and one for the inner loop.  The last segment I printed closes one of them, but not the one the first segments I printed opened.
Updating for completeness the segments for that letter opening and closing both polygons are:
<code>
(Vector (23.275000000000006, -16.470000000000002, 0.0), Vector (22.549169921875006, -17.095537109375, 0.0))
(Vector (22.549169921875006, -17.095537109375, 0.0), Vector (21.902734375, -17.619609375000003, 0.0))
snip
(Vector (23.250000000000007, -17.245, 0.0), Vector (23.275000000000006, -16.470000000000013, 0.0))
(Vector (23.275000000000006, -16.470000000000013, 0.0), Vector (23.275000000000006, -16.470000000000002, 0.0))
closes the exterior polygon
opens the interior polygon
(Vector (17.741666666666667, -15.345, 0.0), Vector (17.9390625, -14.302210286458333, 0.0))
(Vector (17.741666666666667, -15.345, 0.0), Vector (17.9390625, -14.302210286458333, 0.0))
snip
(Vector (18.653743489583334, -14.717233072916667, 0.0), Vector (17.741666666666703, -15.344999999999978, 0.0))
(Vector (17.741666666666703, -15.344999999999978, 0.0), Vector (17.741666666666667, -15.345, 0.0))
closes the interior polygon
</code>
Definitely not a topological problem with the input.

mhindi2, 2022-12-31:
Yes, thank you. I realized that after a while; I should have re-edited the
comment.

mhindi2, 2023-01-01:
Here is the clearest and simplest manifestion of the problem. The string "II" (two consective "I",s in the same font as I reported in the FC files above) produces no medial line for the second "I".
[image]

To test if the input to the vd=Path.Voronoi.Diagram() object is getting the correct input edges, I printed just the x coordinates of the input segments, after shifting by the CenterOfGravity of the face. The results in the attached file shows the two letters have the same edge coordinates (up to the overall shift of the second letter); print output:
15:57:43  ('Mod/Pyramids_and_polyhedrons', 'Mod/Pyramids-and-Polyhedrons', 'Mod/FreeCAD-Pyramids-and-Polyhedrons', 'Mod/Pyramids-and-polyhedrons')
15:58:05  <Face object at 0x8240ca0> num segments in vd: 22)
15:58:05  Shifted x coordinate of segment start
15:58:05  -5.145
15:58:06  -3.072
15:58:06  -0.153
15:58:06  0.871
15:58:06  1.792
15:58:06  -0.588
15:58:06  -0.047
15:58:06  0.784
15:58:06  1.755
15:58:06  2.714
15:58:06  4.198
15:58:06  5.095
15:58:06  4.224
15:58:06  1.178
15:58:06  -0.793
15:58:06  -1.689
15:58:06  0.615
15:58:06  -0.483
15:58:06  -2.329
15:58:06  -2.713
15:58:06  -4.223
15:58:06  -5.145
15:58:06  face0 num Edges in vd: 222
15:58:06  <Face object at 0x7faa9d0> num segments in vd: 22)
15:58:06  Shifted x coordinate of segment start
15:58:06  -5.145
15:58:06  -3.072
15:58:06  -0.153
15:58:06  0.871
15:58:06  1.792
15:58:06  -0.588
15:58:06  -0.047
15:58:06  0.784
15:58:06  1.755
15:58:06  2.714
15:58:06  4.198
15:58:06  5.095
15:58:06  4.224
15:58:06  1.178
15:58:06  -0.793
15:58:06  -1.689
15:58:06  0.615
15:58:06  -0.483
15:58:06  -2.329
15:58:06  -2.713
15:58:06  -4.223
15:58:06  -5.145
15:58:06  face1 num Edges in vd: 230

_Notice_, however,  that the number of edges produced by the vd.construct() call (len(vd.Edges)) is _different_ for the two letters: it is 222 for the first letter and 230 for the second letter. So clearly the overall shift in letter position is producing a different result in the VD construction. So I believe the problem is on the C++ side of the Voronoi algorithm.

I'll have to look into that next :(

mlampert, 2023-01-01:
@mhindi2 could you attach the test file with the 2 IIs please

mhindi2, 2023-01-01:
[II_test.zip](https://github.com/FreeCAD/FreeCAD/files/10328181/II_test.zip)

mhindi2, 2023-01-01:
and for completeness:

def debug_input(vd, face):
    print(f'{face} num segments in vd: {vd.numSegments()})')
    segs = vd.getSegments()  # a list of pairs of Vectors
    shifted = [s[0] - face.CenterOfGravity for s in segs]
    print("Shifted x coordinate of segment start")
    for pt in shifted:
        print(f'{pt.x:0.3f}')

and in buildPathMedia loop:

            debug_input(vd, f)
            vd.construct()
            print(f'face{i} num Edges in vd: {len(vd.Edges)}')

mhindi2, 2023-01-01:
I am just beginning to look at Voronoi.cpp. The correct way to get an angle from a tangent is to use the atan2(y,x) function, so I changed that:

`@@ -226,6 +226,7 @@ double Voronoi::diagram_type::angleOfSegment(int i, Voronoi::diagram_type::angle
   if (!angle || a == angle->end()) {
     Voronoi::point_type p0 = low(segments[i]);
     Voronoi::point_type p1 = high(segments[i]);
+	/*
     double ang = 0;
     if (p0.x() == p1.x()) {
       if (p0.y() < p1.y()) {
@@ -236,6 +237,8 @@ double Voronoi::diagram_type::angleOfSegment(int i, Voronoi::diagram_type::angle
     } else {
       ang = atan((p0.y() - p1.y()) / (p0.x() - p1.x()));
     }
+	*/
+	double ang = atan2(p0.y() - p1.y(), p0.x() - p1.x());
     if (angle) {
       angle->insert(angle_map_t::value_type(i, ang));
     }`

Still does not solve the problem, but I'll keep looking.

mlampert, 2023-01-01:
I created above PR as a discussion point. Note that  skipping 0-length segments has no impact on the behaviour of the voronoi diagram, it's really just the rounding of the scaled coordinates which leads to (more or less) correct results.
I'm hesitant to merge above changes since I don't see _why_ rounding the coordinates should fix the issue.

luzpaz, 2023-01-01:
@mhindi2 please familiarize yourself with [github markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code-and-syntax-highlighting) to learn how to put code in code tags, TIA!

mhindi2, 2023-01-01:
I was not suggesting the rounding as a modification. I was simply pointing
out how I was  testing whether there was something wrong with the input to
the second letter.

mhindi2, 2023-01-01:
@mlampert 

> I created above PR as a discussion point. Note that skipping 0-length segments has no impact on the behaviour of the voronoi diagram, it's really just the rounding of the scaled coordinates which [...]

I am not exactly sure which changes you are referreing to, (0-length segments). I was suggesting a change to Voronoi.cpp, but not to Vcarve.py.

br4243, 2023-01-01:
In mhindi's display of the EXTERIOR2 edges for "II", the curved segment that enters the bottom left side of the right "I" is interesting.  Rather than having a vertex on the left sloped edge of that I, the curve's vertex1 is in the interior of the I.  Voronoi::colorExterior(edge, colorValue) removes it (colors it as EXTERIOR2) as part of the incident_edge() processing, along with all the other interior edges to which it is connected as it follows the rot_next() loop.  I 'm sure we'd like it to stop removing edges when it goes inside the letter's outline.
If you display all the edges of both "I"s you'll see there's nothing like that in the bottom left corner of the left "I"; the exterior edges there are clearly exterior.
I extracted the segment list from the outline of the right "I" and built a standalone program that passes those directly to construct_voronoi() to try different bindings for the polygons' coordinates as well as starting offsets and scaling, and I get what looks to be the same diagram every time.  The scaling is important; I get NaNs back from boost if I drop the scaling too low.  Here is that in case it's of any interest.
 ./rightI | head -16 leaves that curved segment as the last output line.
[rightI.zip](https://github.com/FreeCAD/FreeCAD/files/10329647/rightI.zip)

mhindi2, 2023-01-02:
@br4243 
Bravo! I must admit I am still struggling with understanding the call to colorExterior from python to the Voronoi.cpp code, as it seems to be passing through some intermediate code in VoronoiPy.cpp. But definitely edges that start outside and end inside should not all be marked as exterior.
Note that since the scaling is relative to the origin, the second "I" will have coordinates in the "voronoi coordinate space" that are even more shifted to the right, so perhaps it is not surprising that one gets a different number of segments for the second "I" than the first. On the other hand, I tried a string with three I's ("III") and the third I has the same number of segments as the first and displays fine. I don't understand that.

@mlampert 
In case my suggestions to the modifications to Voronoi.cpp were not clear from the diff I quoted, may I state them here in words: getting an angle from $\rm{atan}(y/x)$ in angleOfSegment() may give the wrong angle, since, for example, $y=-1, x=1$, will give an angle that is the same as $y=1, x=-1$, however the correct angle for the first segment is -45 deg, while for the second it should be 135 deg, definitely not the same angle, eventhough the tangent  $(y/x)$ is the same for both. The function $\rm{atan2}(y,x)$ gives the correct result for both cases, as well as for the case $x=0$, so the extra tests for $x==0$ are not needed. May be for the colinearity test, for which angleOfSegment is being called, the ultimate outcome might not matter, but certainly one would not want for a navigation app, for example, to tell one to head 45 degrees south of east, whereas it should be telling one to head 45 degrees west of north (exaclty the opposite direction!) (Note that for some libraries the first argument to the atan2 is the x and the second the y, so one should actually check!)

Another suggestion: calculating how close in direction two segments are by taking the difference between their angles relative to some direction (as is done in colorColinear() is tricky; for example segments with angles 1 deg and 359 degrees are quite close to each other, but the difference is large; subtracting a PI from the difference does not solve the problem. The safe way is to take the dot product of unit vectors along the two segments (cos(thet) = u_1.dot( u_2) and comparing to the cosine of some threshold angle.
I apologize for sounding pedantic about this; I am sure you are aware of these points. The code dealing  with angles in Voronoi.cpp, though, is not quite correct.

mlampert, 2023-01-02:
@mhindi2  - I understood your proposed changes, thanks for the details though. The PR I posted is not related to that though, it is the result of some experimentation I did and it fixes the issue(s), I just have no explanation as to _why_ it does so.

mhindi2, 2023-01-02:
Ah, sorry I misunderstood. So, am I to understand that there is a fix now?
How can I test it/use it. The whole thing started when I was trying to
carve something as a Christmas present for my wife and I have not been able
to do that yet!

mlampert, 2023-01-02:
if I'm not mistaken all you have to do is add `round(...)` as in the PR - I have not found an instance where it doesn't generate the correct Path.

mhindi2, 2023-01-02:
@mlampert 
Seems to work for the II with round(...) in addSegments. Wondering if there should als be a riund(..) in addPoint, thouh that does not seem to be called anywhere.
[image]

I am trying with a string that uses all 26 letters of the alphabet. Seem to be running into an infinite loop in cutWire(pWire) about 3/4 into the wires. Will try to find out why. This is a string that used to run before, but skipping one character. The string is "NYMPHS BLITZ QUICK VEX DWARF JOG." should you want to give it a try.

Not sure if it has anything to do with angle changes I made to Voronoi.cpp"
``` diff
@@ -226,16 +228,7 @@ double Voronoi::diagram_type::angleOfSegment(int i, Voronoi::diagram_type::angle
   if (!angle || a == angle->end()) {
     Voronoi::point_type p0 = low(segments[i]);
     Voronoi::point_type p1 = high(segments[i]);
-    double ang = 0;
-    if (p0.x() == p1.x()) {
-      if (p0.y() < p1.y()) {
-        ang = M_PI_2;
-      } else {
-        ang = -M_PI_2;
-      }
-    } else {
-      ang = atan((p0.y() - p1.y()) / (p0.x() - p1.x()));
-    }
+       double ang = atan2(p0.y() - p1.y(), p0.x() - p1.x());
     if (angle) {
       angle->insert(angle_map_t::value_type(i, ang));
     }
@@ -269,15 +262,16 @@ void Voronoi::colorColinear(Voronoi::color_type color, double degree) {
         && it->cell()->contains_segment()
         && it->twin()->cell()->contains_segment()
         && vd->segmentsAreConnected(i0, i1)) {
-      double a0 = vd->angleOfSegment(i0, &angle);
-      double a1 = vd->angleOfSegment(i1, &angle);
-      double a = a0 - a1;
-      if (a > M_PI_2) {
-        a -= M_PI;
-      } else if (a < -M_PI_2) {
-        a += M_PI;
-      }
-      if (fabs(a) < rad) {
+         
+         Voronoi::point_type p0 = low(vd->segments[i0]);
+         Voronoi::point_type p1 = high(vd->segments[i0]);
+         auto u1 = Base::Vector3d(p1.x() - p0.x(), p1.y() - p0.y(), 0).Normalize(); 
+         p0 = low(vd->segments[i1]);
+         p1 = high(vd->segments[i1]);
+         auto u2 = Base::Vector3d(p1.x() - p0.x(), p1.y() - p0.y(), 0).Normalize();
+         double cosAngle = u1.Dot(u2);
+         
+      if (cosAngle > cos(rad)) {
         it->color(color);
         it->twin()->color(color);
       }

```

mhindi2, 2023-01-02:
I am getting a seeming infinite loop when trying to Vcarve the string "NYMPH" in the attached file. This afer using the round(...) after scaling segments.
[post-round-test.zip](https://github.com/FreeCAD/FreeCAD/files/10332671/post-round-test.zip)
The carving works with the string up to the "H", i.e., the string "NYML" processes. I put some debugging prints in Vcarve.py and the infinite loop seems to be enterd after processing wire 30/57 in cutWire. This is with my modified test for colinearity (using the dot product to calculate the angle). With the original way of calculating angles between segments, the infinite loop is entered after processing 34/63 of the vornoiWires. So my calculation of angles seem to produce more colinear segments, but the infinite loop is entered regardless of which version of colinearColor is used. Without the rounding, the character which was causing the infinite loop was not being rendered, as before.

Aside: If the dot product suggested in my previous post is used (which I iterate is the more general way of calculating the angle between vectors, whether in 2D or 3D), then there is no need for the method  angleOfSegment, nor for the angle_type_map angle, so these ought to be removed from Vonronoi.cpp and Vornoi.h.

mhindi2, 2023-01-03:
I know this is probably not the forum to ask this, but could someone please point me to somewhere that explains the difference between Voronoi.cpp and VornoiPyImp.cpp? Seems VoronoiPyImp.cpp is supposed to be the code that gets called from python, but when I make changes to Voronoi.cpp I see the changes when called from Vcarve.py.

mhindi2, 2023-01-03:
Ok, it looks like the python code is interfaced to the code in VoronoiPyImp.cpp and some of the code in that calls code in Voronoi.cpp. The confusing thing is that part of the implementation of methods is in VoronoiPyImp.cpp looks like the code in Voronoi.cpp (other than parsing the arguments coming from the Python side).

mhindi2, 2023-01-03:
I have found out where Vcarve.py seemed to be going into an infinite loop and have found a solution for it.
First try to carve the letter "H" in the attached file. Note the size is 10 mm. There is no infinite loop if the size is 4 mm.
[H_2003_10mm.zip](https://github.com/FreeCAD/FreeCAD/files/10339164/H_2003_10mm.zip)

With the 10 mm letter one of the edges sent to _getPartEdge(edge, depths)  was returning nan for edge.getDistances(). So I modified the code in Vcarve.py to

``` python
def _getPartEdge(edge, depths):
    dist = edge.getDistances()
    if math.isnan(dist[0]) or math.isnan(dist[1]):
        return None
    zBegin = _calculate_depth(dist[0], depths)
    zEnd = _calculate_depth(dist[1], depths)
    return edge.toShape(zBegin, zEnd)
```
and in the calling method _getPartEdges to
``` python
    def _getPartEdges(self, obj, vWire, geom):
        edges = []
        for e in vWire:
            partEdge = _getPartEdge(e, geom)
            if partEdge is not None:
                edges.append(partEdge)
        return edges
```
As to why edge.getDistances() is reutrning nan, I don't know. I am guessing the coloring algorithms in Voronoi.cpp might be passing an edge the is is_inifinte, but I have not checked that.

Note that I used the Voronoi.cpp version with the round() fix and my calculations of angles that I posted before.

I will try to post images of all the characters in that font carved out, but it's taking along time to render all characters!

mhindi2, 2023-01-03:
As far as I can tell, all the characters in the font urw-base35/Z003-MediumItalic.otf now carve with the above changes to Vcarve.py
[image]
The file, in case one want s to test is
[All_characters.zip](https://github.com/FreeCAD/FreeCAD/files/10340332/All_characters.zip)

mhindi2, 2023-01-04:
@mlampert , @br4243 Have either of you had a chance to check whether the full set of characters carves with the modified Vcarve.py code? To summarize, I found that the "II" carves with the round(..) of segment ends that @mlampert suggested, but that FreeCAD was getting into an infinite loop with other charcters. I traced that to a nan that is coming back from edge.getDistances() being called from Vcarve.py. I modified Vcarve.py to bypass the edges with nan and then things work.

mlampert, 2023-01-06:
@mhindi2 sorry about the late reply, I just got bounce msgs to my emails.

If you have a change that fixes all problems please create a PR. It does feel a bit like a game of whack-a-mole tbh but your changes are making the world a better place (or at least vcarve ;) ).
One question - does your change also fix @br4243 's original issue? It has no impact on your PR, just wondering.

mhindi2, 2023-01-06:
Aaah! The issue brought up by @br4243 might have been the cause of the nan. I had not payed attention to that issue, because of your comment:

`That does seem to be a different issue. Applying above patch does not fix this issue. If you have a sample document that exposes your issue could you file a separate issue here and describe what's going on please?
`
I will try to make the modification @br4243  suggested and temporarily comment out my skipping the nan's to double check that it resolves the infinite loop issue. In any case, the additional test for nan in Vcarve.py does not do any harm for the rest of the non-nan cases, so might as well be left there.

I am not sure I am the best person to do a PR. Although I do contribute to the GDML Workbench, my github skills are still developing, and I am not sure how I would do it correctly for the full FreeCAD project yet. After I confirm @br4243 fix, may be you/him can put in a new PR. It is still a mystery, though, why the round(...) fixed some the letters being skipped.

mhindi2, 2023-01-06:
@mlampert , @br4243 
> One question - does your change also fix @br4243 's original issue? It has no impact on your PR, just wondering.

I can confirm that that the infinite loop I was encountering was due to the division by zero that @br4243 pointed to. My changes to Vcarve.py did not _fix_ the issue per se, but bypassed it when it happened. So definitely @br4243 's fix should be incorporated. Although my change to Vcarve.py does not hurt, it is probably better to leave it out, in case there is some other source of a nan. Although if a nan happens Vcarve would go into an infinite loop, it is probably better to know about it and fix its cause, rather than simply bypassing the result.

We still don't know why round() fixed the original issue (which is separate, I believe, from the issuse pointed to by @br4243 ). Looking at the code in VoronoiEdgeImp.cpp, I noticed that when constructing the parabola, if the point was on the segment, a line is constructed (I presume perpendicular to the line segment, but I will double check); however the test for the point being on the segment seems to be a test of whether the point is on either end, not anywhere along the line. I will double check and fix if that is really the test.

mhindi2, 2023-01-06:
@mlampert , @br4243 
> The voronoi diagram creates 2 types of artifacts, straight edges and parabolas. Parabolas are defined by a straight segment and a point. In order to get the shape of the parabola we have to find the [...]

So, I am now looking at the code in VoronoiPyImp.cpp:toShape(), and I see a test isPointOnSegment. If the test is true, an edge that is a line is created. I see two problems with this:
(1) The test isPointOnLine seems to be testing for the point being at either end of the line segment, and not anywhere on the line (the test should be _anywhere_ on the line)
(2) The constructed line (instead of the parabola) is a copy of the original line. It should be a line _perpendicular_ to the line segment, passing through the point. I am not sure what the length should be at this point.
Could either of you please confirm the above observations before I make the needed changes. Thanks.

mhindi2, 2023-01-07:
@mlampert 
> The voronoi diagram creates 2 types of artifacts, straight edges and parabolas. Parabolas are defined by a straight segment and a point. In order to get the shape of the parabola we have to find the [...]

I now fully understand VoronoiEdgePyImp.cpp and how the parabolas are generated, and have a possible construction for the case the point is on the segment.  However, I am not sure how to generate the point/edges to test that case. Could you please provide me with the file that had segments in the attached picture? I suppose I could write a test python script that calls the VoronoiDiagram with the above edges/points.

mhindi2, 2023-01-11:
from @mlampert 
> First off, when I tested this last night vcarve on the latest master the letter `e` is carved correctly, however the letter `h` is missing. As already indicated by Gene in the forum discussion, [...]

I have conducted many tests on this issue to try to understand the cause of the error and the behavior for cases of a point on the same line as a segment. 

(1) If a point is on the same line as a segment (but ouside the segment), the vornoi algorithm produces the correct edge ( a line midway between the segment and the end closest to the input segment; not a parabola). 
(2) If a point is on the segment itself, the vornoi algorithm produces a line with one end a nan and the other and infinite.  Mathematically, the "parabola" in that case should become a two lines perpendicular to the segment, on both sides of the segment. There is no limit to these lines. However, none of the letters tested pose this problem.

I suspect, as @mlampert indicates above, the problem is with the vornoi algorithm (from the boost library) and not from anything done on the FreeCAD side. The following images show the vornoi segments, with and without rounding the input coordinates:

With rounding:

[image]

Without rounding (the current FreeCAD code):
[image]

Note that without rounding the vornoi algorithm is producing edges that cross each other, without them sharing a common vertex. That should never happen. I believe these edges are produced by the vornoi algorithm itself and not by the post processing done in VornoiEdgePyImp.cpp.

I am currently trying to compile boost1.81.0 and build FreeCAD with it to see if the problem is there too. If it is, I believe this is something that should be brought to the attention of the boost vornoi developers.

mhindi2, 2023-01-11:
The problem is still there with boost1.81.0: without round(input coordinates), you get crossing edges from voronoi for the second "I", and with rounding you don't. I am giving up doing other investigations at this point. Seems to me a problem for the boost voronoi library developers.

br4243, 2023-01-11:
The input to VoronoiBuilder is slightly different for the two Is.  This expanded (and sadly, somewhat ugly) test program for experimenting with coordinate shifts shows there is a 1 abscissa unit X shift in the last point, shared by the last two segments, of the input going to boost between the good and bad diagram.  
The Voronoi diagram oscillates between the two states shown by @mhindi2 above depending on the shift in the X axis; I have not tried shifts in the Y axis, but have been looking at one of the transition points for the X axis shift.  The "-x" values below are specified as FreeCAD coordinates, so mm in this case.
[vori.zip](https://github.com/FreeCAD/FreeCAD/files/10395669/vori.zip)
<code>
$ ./vori -o -x -.606000000000028 > close0.txt
$ ./vori -o -x -.606000000000029 > close1.txt
</code>
The above produces output suitable for [Voronoi Visualizer](https://www.boost.org/doc/libs/1_81_0/libs/polygon/example/voronoi_visualizer.cpp) matching the scaling used by Vcarve for the I shifted by two slightly different values.  Meaning, these are the values Vcarve passes to boost for these two instances of the "I".  In this case "close1.txt" is the good diagram.

<code>
$ diff close0.txt close1.txt
23,24c23,24
< -24554 6769 -25475 6130
< -25475 6130 -25476 6130
---
> -24554 6769 -25476 6130
> -25476 6130 -25476 6130
$ tail -2 close0.txt
-24554 6769 -25475 6130
-25475 6130 -25476 6130
</code>
The two input files are short enough to hopefully not be obnoxious to include here.
<code>
$ cat close0.txt
0
22
-25476 6130 -23402 6155
-23402 6155 -20484 6130
-20484 6130 -19460 6130
-19460 6130 -18538 6769
-18538 6769 -20919 6948
-20919 6948 -20378 9294
-20378 9294 -19546 12640
-19546 12640 -18575 16404
-18575 16404 -17616 20002
-17616 20002 -16132 20155
-16132 20155 -15236 20796
-15236 20796 -16106 20796
-16106 20796 -19152 20770
-19152 20770 -21124 20796
-21124 20796 -22020 20155
-22020 20155 -19716 20002
-19716 20002 -20813 15526
-20813 15526 -22660 8305
-22660 8305 -23044 6948
-23044 6948 -24554 6769
-24554 6769 -25475 6130
-25475 6130 -25476 6130
$ cat close1.txt
0
22
-25476 6130 -23402 6155
-23402 6155 -20484 6130
-20484 6130 -19460 6130
-19460 6130 -18538 6769
-18538 6769 -20919 6948
-20919 6948 -20378 9294
-20378 9294 -19546 12640
-19546 12640 -18575 16404
-18575 16404 -17616 20002
-17616 20002 -16132 20155
-16132 20155 -15236 20796
-15236 20796 -16106 20796
-16106 20796 -19152 20770
-19152 20770 -21124 20796
-21124 20796 -22020 20155
-22020 20155 -19716 20002
-19716 20002 -20813 15526
-20813 15526 -22660 8305
-22660 8305 -23044 6948
-23044 6948 -24554 6769
-24554 6769 -25476 6130
-25476 6130 -25476 6130
</code>

mhindi2, 2023-01-11:
I have built my own Visualizer based on what's in Vcarv.py; I will take a look at your input files. The issue, though, is not that the inputs are slighly different for the two I's - that is expected. The issue is that Vornoi diagram for the second I is producing a voronoi diagram that violates the definition of a Voronoi diagram. The voronoi diagram is supposed to be made of distinct non-overlapping cells. The cells are supposed to share edges and vertices, not areas. That is, the edges of a cell can meet at a common vertex, but are not supposed to cross. But the edges for second "I" clearly cross where I indicated.
Afterthought: of course, the above assumes that the input is made up of segments and points that do not cross. Are you saying that after the scaling some segments overlap? I will check later.

luzpaz, 2023-01-12:
(note: converted code tags to markdown code tags)

[18 more replies]

closed 2024-04-27 by sliptonic