FreeCAD/FreeCAD #8065: [Issue] AddonManager GUI tests stuck in a localized environment

0penBrain, 2022-12-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu 22.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31347 +72 (Git)
Build type: Debug
Branch: MyBranch
Hash: bb613ced7e4aef68c78e08bf89968d8d6cf00826
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: French/France (fr_FR)

NB : the +72 commits don't touch the source code. :)
```

### Subproject(s) affected?

Addon Manager

### Issue description

In a localized (non-english) environment, AddonManager GUI tests will stuck on the below modal dialog : 
[image]

### Anything else?

@chennes Quickly looking at the code, it looks like it's betting on the button text to find and virtually click the button.
I think it would be more robust to use the [button role](https://doc.qt.io/qt-5/qdialogbuttonbox.html#ButtonRole-enum) instead.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2022-12-20:
Yeah, I had wondered if that would be an issue -- you're right, button role would work, or just make the dialog fire it's accepted/rejected signal would probably work too.

0penBrain, 2022-12-20:
> just make the dialog fire its accepted/rejected signal would probably work too.

I guess you're thinking more to something like calling the accept/reject slot of the dialog. ;)

closed 2022-12-20 by chennes

0penBrain, 2022-12-20:
Can confirm it works fine now. Thanks @chennes

PR #8066 (2022-12-20): Addon Manager: Fix test language dependency
Some incorrect translation contexts for dialog title strings are fixed, and the button search is switched to search for the Qt constants rather than the actual button text.

Fixes #8065
commit 4db68877 (2022-12-20): Addon Manager: Fix test language dependency

Fixes #8065