FreeCAD/FreeCAD #8067: [Issue]

dajt1725, 2022-12-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Rocky Linux 8.7 (Green Obsidian) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

Sketcher solver says sketch contains redundant constraints when it does not.

Run the following script to create a body containing a sketch with two lines, two arcs, and some constraints.
Note that the solver says the sketch has a redundant constraint (10).
Disable constraint 10.
Now note that the solver says the sketch has a redundant constraint (12).
Disable constraint 12.
Note the sketch is now under-constrained.
Move the two lines.
Activate constraint 10.
Note that one of the lines is moved back to its original position.
Activate constraint 12.
Note the other line is moved back to its original position.
Note that the sketch is now fully constrained, even though it has the same set of constraints that was "over constrained" originally.

-----
import math
import Part
import Sketcher

doc = App.getDocument('Unnamed')
vOrigin = App.Vector(0,0,0)
vNormal = App.Vector(0,0,1)

stepName = 'Step0'
sketchName = 'Step0step'
rLeading = 0
rTrailing = math.radians(21.1765)
mmInnerEdge = 999
mmOuterEdge = 2001
mmStepWidth = 1002
vLI = App.Vector(mmInnerEdge * math.cos(rLeading), mmInnerEdge * math.sin(rLeading),0)
vLO = App.Vector(mmOuterEdge * math.cos(rLeading), mmOuterEdge * math.sin(rLeading),0)
vTI = App.Vector(mmInnerEdge * math.cos(rTrailing), mmInnerEdge * math.sin(rTrailing),0)
vTO = App.Vector(mmOuterEdge * math.cos(rTrailing), mmOuterEdge * math.sin(rTrailing),0)

def CreateSketch(body, name, sup, zoffset, geometries, constraints):
    body.newObject('Sketcher::SketchObject', name)
    sketch = doc.getObject(name)
    sketch.Support = sup
    sketch.AttachmentOffset.Base.z = zoffset
    sketch.MapMode = 'FlatFace'
    for o in geometries:
        sketch.addGeometry(o, False)
    for o in constraints:
        sketch.addConstraint(o)
    return sketch

doc.addObject('PartDesign::Body', 'Step0Body')
body = doc.getObject('Step0Body')
sup = (doc.getObject(body.Origin.OriginFeatures[3].Label), [''])
sketch = CreateSketch(body, sketchName, sup, 0,
[Part.LineSegment(vLI, vLO),
 Part.LineSegment(vTI, vTO),
 Part.ArcOfCircle(Part.Circle(vOrigin, vNormal, mmInnerEdge), rLeading, rTrailing),
 Part.ArcOfCircle(Part.Circle(vOrigin, vNormal, mmOuterEdge), rLeading, rTrailing)],

[Sketcher.Constraint('Coincident',0,1,2,1), Sketcher.Constraint('Coincident',0,2,3,1),
 Sketcher.Constraint('Coincident',1,1,2,2), Sketcher.Constraint('Coincident',1,2,3,2),
 Sketcher.Constraint('Coincident',-1,1,2,3), Sketcher.Constraint('Coincident',-1,1,3,3),

 Sketcher.Constraint('Angle', -1, 1, 0, 1, rLeading),
 Sketcher.Constraint('Angle', -1, 1, 1, 1, rTrailing),

 Sketcher.Constraint('Distance',-1,1,0,1, mmInnerEdge), Sketcher.Constraint('Distance',-1,1,0,2, mmOuterEdge),
 Sketcher.Constraint('Distance', 0, mmStepWidth), Sketcher.Constraint('Distance', 1, mmStepWidth),
 ])

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-09:
Hi @abdullahtahiriyo can you weigh-in on this ticket at your convenience? TIA

abdullahtahiriyo, 2023-01-17:
First things first, thanks for providing the Python code to make it reproducible.

The bug report is licit. The underlying problem is known. But I save the ticket as means to reproduce it.

The issue arises because the gradient of a constraint becomes zero in a specific position, so the constraint is detected as redundant. However, when activated the second time, it does not converge to exactly that specific position, and the gradient is very small, but not zero. Therefore it is not detected as redundant.

No "buts" there. Yes, it is a bug. But some "Howevers" :).

However, as @PLChris would say, that sketch does not comply with good sketching practice. One basic rule is that geometric constraints (the ones not having an associated value) should be preferred to dimensional ones (the one having an associated value, i.e. the magnitude).

So this:
[image]

Should rather be something like this:
[image]

There are many different alternative (good) ways of constraining. The one making the sketch usually understands better what needs to be dimensioned, as he or she knows best what needs to be achieved. For example the lines defining the sector could also had been made just "equal".

Following these rules, issues like the one described are less likely to appear.

closed 2025-01-12 by github-actions[bot]