FreeCAD/FreeCAD #8074: [Issue]

hwa59, 2022-12-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Debian GNU/Linux 11 (bullseye) (KDE/plasma)
Version: 0.21.0.31271 (Git) AppImage
Build type: Release
Branch: master
Hash: 1189be75d044e87cb55c74344405f4e484d57ddc
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * SelectorToolbar
  * kicadStepUpMod 10.16.5
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * Curves 0.6.5
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

Mesh workbench 
if you export a Mesh to a STL-File the file-extension (.stl) will not be added automatic.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-15:
Hi @ianrrees LTNS! Hope you don't mind but would you have a moment to tackle this issue ? If not, where would this be located in the source code ?

ianrrees, 2023-01-16:
Sorry, no, I believe it has been years since I have compiled FreeCAD.  From memory the export dialog is the standard Qt file save one, and some formats have multiple extensions, so I'd guess that would be a substantial amount of work to make something with a nicer design.

hwa59, 2023-01-16:
After a crash i had to reinstall my system. Now the export is ok but i don't know why.

closed 2023-01-16 by hwa59