FreeCAD/FreeCAD #8075: [Feature Request] Broken out section views

FEA-eng, 2022-12-22:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=35&t=74105

### Subproject(s) affected?

Techdraw

### Idea description

The new Complex Section tool added support for several advanced section types. However, broken out sections are not yet supported and have to be created manually with workarounds.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MakerFC, 2023-01-08:
please also add Forums discussion:
[Techdraw broken-out section]
(https://forum.freecadweb.org/viewtopic.php?f=35&t=74441)

luzpaz, 2023-01-08:
Added to OP

NXTler, 2023-08-26:
It's used a lot in the professional space, we should implement it.

luzpaz, 2023-11-06:
Is this actually a duplicate of #5694 ?

FEA-eng, 2023-11-06:
> Is this actually a duplicate of #5694 ?

No, despite similar naming, those are different kinds of views. The one discussed here is basically about looking locally inside the structure while the other (older) one is about depicting long parts by cutting a segment from the middle and putting the pieces close to each other.

FEA-eng, 2024-04-10:
@pierreporte In SW, you can either specify the depth value or select the geometry in a drawing view up to which the section will be made.

[image]

I wouldn't mind if it was only possible to define the depth value in FreeCAD initially but geometry selection would be nice too. Fortunately, this feature should be possible to implement in a similar way as other advanced sections.

pure-machinery, 2024-09-21:
Perhaps the creation should stay consistent with the other methods of defining cross sections. The distance is a no-no from my side.

On the other hand this has more to do with rendering in my opinion then the cross section code, since the cross section code is already here the only problem is how to display the local area of a section cut while all else is preserved. 

Multiple local cross sections are also possible and useful.

3nt3, 2025-04-10:
If the PR about this was merged, why is the issue sti open ^^

FEA-eng, 2025-04-10:
> If the PR about this was merged, why is the issue still open ^^

It’s explained in the description of this issue - the merged PR implemented other types of advanced section views, not broken out ones discussed here.

pierreporte, 2025-04-29:
A **150 € reward** will be given to the contributor that writes a pull request, solving this issue, that gets merged. It will be paid by the FreeCAD Project Association. It is to be considered an incentive. Grant applications are welcome, but the total amount will exclude the reward.