FreeCAD/FreeCAD #8076: [Feature Request] Rotated and removed section views

FEA-eng, 2022-12-22:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=35&t=74105

### Subproject(s) affected?

Techdraw

### Idea description

The new Complex Section tool added support for several advanced section types (#6309). However, rotated and removed sections are not yet supported and have to be created manually with workarounds.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-08-21:
Was this recently addressed ?

FEA-eng, 2023-08-21:
> Was this recently addressed ?

Nope, no development in this regard so far.

franklincg, 2026-08-17:
Hi, I'm interested in working on this bounty. Is it still available for a new contributor? I’m currently getting familiar with the TechDraw section-view code and would like to work on it if it’s still open.