FreeCAD/FreeCAD #8079: Arch: module 'inspect' has no attribute 'getargspec'

AchimStuy, 2022-12-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Fedora Linux 37 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.1.Unknown
Build type: Unknown
Python 3.11.0rc1, Qt 5.15.5, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.0
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly3 0.11.4
[/code]
```

### Subproject(s) affected?

Arch

### Problem description

I just created a new project and wanted to start by creating a wall. There is an exception during defining the first point of the wall:

```
14:50:17  AttributeError: module 'inspect' has no attribute 'getargspec'. Did you mean: 'getargs'?
14:50:31  Traceback (most recent call last):
14:50:31    File "/usr/lib64/freecad/Mod/Draft/DraftGui.py", line 1481, in validatePoint
14:50:31      self.pointcallback(num_vec, self.relativeMode)
14:50:31    File "/usr/lib64/freecad/Mod/Draft/draftguitools/gui_snapper.py", line 1412, in getcoords
14:50:31      accept()
14:50:31    File "/usr/lib64/freecad/Mod/Draft/draftguitools/gui_snapper.py", line 1430, in accept
14:50:31      if len(inspect.getargspec(callback).args) > 1:
14:50:31             ^^^^^^^^^^^^^^^^^^
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-12-23:
What is the step by step actions you took? (so it can be reproduced reliably)

AchimStuy, 2022-12-23:
As I already described above:

0. Clean FreeCAD installatioin (no BIM addon installed)
1. Start new project
2. Create a wall. During picking the first point of the wall the error occurs.

ztx-lyghters, 2022-12-24:
Same problem for me on Void Linux. Same steps.

luzpaz, 2022-12-29:
@paullee0 can you reproduce ?

paullee0, 2022-12-29:
FreeCAD built-in repository on Fedora right ?  I am on Fedora 36, always using AppImage.

Tested the built-in repository 0.20.1 version, drawing a Wall works as usual.

[image]

PR #8101 (2022-12-29): Draft: update inspect.getargspec to inspect.getfullargspec
Fixes #8079.

`getargspec` has been removed from the `inspect` module in Python 3.11.

See: https://docs.python.org/3/whatsnew/3.11.html

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR
closed 2022-12-29 by Roy-043