FreeCAD/FreeCAD #8082: Navigation Cube - breaks upon larger size.

bstott, 2022-12-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
OS: Windows 10 Version 2004
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * A2plus 0.4.60g
  * Assembly4 1.20.6
  * boltsfc 2022.11.5
  * CfdOF 1.20.6
  * CurvedShapes 1.0.4
  * Curves 0.12.5
  * Defeaturing 1.2.0
  * Design456 0.0.1
  * DynamicData 2.46.0
  * EM
  * Estimate 0.1.2
  * ExplodedAssembly
  * fasteners 0.4.27
  * fcgear 1.0.0
  * FEMbyGEN 1.2.0
  * FEM_FrontISTR 0.1.0
  * freecad.gears 1.0.0
  * Help 1.0.3
  * InventorLoader 1.3.0
  * lattice2 1.0.0
  * Launcher
  * Manipulator 1.4.9
  * offline-documentation 1.0.0-alpha
  * OpticsWorkbench 1.0.9
  * parts_library
  * pcb
  * PieMenu 1.2.4
  * POV-Ray-Rendering
  * QuickMeasure 2022.10.28
  * Silk 1.0.0
  * slic3r-tools
  * SlopedPlanesMacro
  * symbols_library
  * ThreadProfile 1.84.0
  * toSketch 1.0.1
```

### Subproject(s) affected?

None

### Issue description

Edit > Preferences > Display > Navigation.
Cube size: 132  <default>
Cube size: 142 <breaks>

Navigation Cube Breaks on enlarging size for the reverse spot in upper right.

If you adjust size of cube up from default the reverse side button won't work.

 Example: Default size = 132; Change size = 142.  Cube will not work when the reverse spot is selected.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-12-29:
I can't reproduce this issue at all

bstott, 2022-12-29:
> 

Sorry to hear that. It is repeatable for me. Vs 0.20.1 and 0.20.2.. But, works on RealThunder.

adrianinsaval, 2022-12-29:
Do you have AMD graphics? In the past some people had issues with the navicube when using AMD graphics

bstott, 2022-12-29:
This problem is programmatic. I think. The problem isn't rendering the cube or featured. A specific program feature stops working 9n resizing the navicube. When ising RealThunder version, the feature works as expected.

adrianinsaval, 2022-12-29:
when asked something in the interest of identifying the bug please answer, it may not seem relevant to you but it could be. Help us help you, otherwise you might as well report this bug to a wall.

adrianinsaval, 2022-12-29:
And FYI in the case I mentioned the navicube was displayed correctly but clicking on it not always worked, so it seems very similar to me.

adrianinsaval, 2022-12-29:
Ok I was now also able to reproduce by going from 132 to 200:
[image]
Notice however that this only happens on a 3d view that was already active before changing the configuration, when you open new 3d views they work normally, when reporting you should include this type of detail, the issue seems to be that the bounding region of the cube is not updated in the active view, it is not only the reverse side button but any clickable face that is outside the previous bounds of the navicube as can be seen in the gif

bstott, 2022-12-29:
Good Job! 

Sorry, I wasn't going to fully bug test the feature. That's the progmer's task. Eh?

You went too deep for me. Congratulations, you found the problem is worse. You now know its function is more fully degraded than I noticed.  I was only hindered and annoyed about the reverse 'dot/circle' not working.

I didn't see you even mouse over the dot in the upper right.

Have a Good New Year!

bstott, 2022-12-30:
> And FYI in the case I mentioned the navicube was displayed correctly but clicking on it not always worked, so it seems very similar to me.

Not sure what sequence you are responding to at this time.

But, Congratulations!  You found details that show more a problem than I reported.

I am not a programmer. I am also not professional with FreeCAD. I am struggling through tutorials, which appear basic. However, FreeCAD is so buggy and quirky I am having quite a tough time. It actually will not work for consistent design to production.  But, it's all some newbs like me have access to.

You went to opening and closing projects and clicking all around. I am just trying to get a model done with an interface I can use. BUT - it isn't too cooperative. 

Happy New Year - Again!

I'm done. I reported. You found the inactivity.  You can close me out of the communications.

adrianinsaval, 2022-12-30:
If you mean that you don't want to receive further email notifications, that is not something we control, that's GitHub doing it, so you have to click on the "Unsubscribe" link at the bottom of the email

donovaly, 2023-01-22:
Is this still an issue with the new naviCube in master? At least i cannot reproduce it there or need a better recipe. Changing the size makes no visible problems for me.

adrianinsaval, 2023-01-23:
Yes I can reproduce on master too
```
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31641 (Git)
Build type: None
Branch: master
Hash: d28d63b87b60161419c6c0c532fbbfaed96926b8
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
Installed mods: 
```
repro steps:
-open freecad
-create or open a file (this is important)
-open preferences and increase navicube size (recommended 2x or 3x for more obvious effect), click ok
on a previously existing 3d view some buttons of the navicube are not clickable, only the buttons within the bounds of the previous navicube size measured from the lower left corner are clickable.
on any newly created 3d view navicube works as expected

adrianinsaval, 2023-01-23:
in this example I went from 150 to 300, you can see how the buttons don't highlight on hover as soon as I pass the half point of the cube

https://user-images.githubusercontent.com/36372335/214013356-83f0c0ed-109a-4da0-8e5d-dc0221824e5b.mp4

donovaly, 2023-01-23:
Thanks, I see now. With my attempt to fix the font issue in Windows 11, this would bug would be fixed too.
However, I am stuck since I don't know how to delete an existing NaviCube to recreate it properly from scratch: https://forum.freecadweb.org/viewtopic.php?p=654779#p654779

I need the help of @Rexbas

Rexbas, 2023-01-23:
I tried to reproduce it but instead I somehow managed to make the NaviCube completely disappear xD. I don't have much time next two weeks so I cannot test many things. I am not sure how to delete the existing NaviCube. From what I can see, a pointer to the NaviCube is stored in View3DInventorViewer.
https://github.com/FreeCAD/FreeCAD/blob/23af228b8790d74e03eab58e302bb95cbc82a34a/src/Gui/View3DInventorViewer.h#L440
and created here:
https://github.com/FreeCAD/FreeCAD/blob/23af228b8790d74e03eab58e302bb95cbc82a34a/src/Gui/View3DInventorViewer.cpp#L496
NaviCube itself has then a pointer to a NaviCubeImplementation
https://github.com/FreeCAD/FreeCAD/blob/23af228b8790d74e03eab58e302bb95cbc82a34a/src/Gui/NaviCube.h#L53
and is created here in NaviCube:
https://github.com/FreeCAD/FreeCAD/blob/23af228b8790d74e03eab58e302bb95cbc82a34a/src/Gui/NaviCube.cpp#L235

If you can access the View3DInventorViewer and delete the NaviCube there then NaviCubeImplementation will also be deleted by the destructor. And after that just create a new one like here, I guess: https://github.com/FreeCAD/FreeCAD/blob/23af228b8790d74e03eab58e302bb95cbc82a34a/src/Gui/View3DInventorViewer.cpp#L496

NaviCube pointer in View3DInventorViewer is private so maybe you need to make a public method in View3DInventorViewer that deletes and creates a new NaviCube. And call that method.

donovaly, 2023-01-24:
Many thanks! Good pointer with View3DInventorViewer . I will try this the next days and report back.

donovaly, 2023-01-26:
I succeeded. Here is the PR that should also fix the issue reported here: #8266
@adrianinsaval can you please try it out?

closed 2023-01-27 by donovaly

bstott, 2023-01-27:
Problem still exists - Is worse.  1/26/2023 with latest bundle.

[code]
OS: Windows 10 Version 2004
Word size of FreeCAD: 64-bit
Version: 0.21.0.31695 (Git)
Build type: Release
Branch: master
Hash: 64caea207394bb05587c8ffb29b7a3e8696d24df
Python 3.10.8, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * A2plus 0.4.60g
  * Assembly4 0.12.5
  * boltsfc 2022.11.5
  * CfdOF 1.20.6
  * CurvedShapes 1.0.4
  * Curves 0.6.5
  * Defeaturing 1.2.0
  * Design456 0.0.1
  * DynamicData 2.46.0
  * EM
  * Estimate 0.1.2
  * ExplodedAssembly
  * fasteners 0.4.50
  * fcgear 1.0.0
  * FEMbyGEN 1.2.0
  * FEM_FrontISTR 0.1.0
  * freecad.gears 1.0.0
  * Help 1.0.3
  * InventorLoader 1.3.0
  * lattice2 1.0.0
  * Launcher
  * Manipulator 1.4.9
  * offline-documentation 1.0.0-alpha
  * OpticsWorkbench 1.0.9
  * parts_library
  * pcb
  * PieMenu 1.2.4
  * POV-Ray-Rendering
  * QuickMeasure 2022.10.28
  * Silk 1.0.0
  * slic3r-tools
  * SlopedPlanesMacro
  * symbols_library
  * ThreadProfile 1.84.0
  * toSketch 1.0.1
[/code]

adrianinsaval, 2023-01-27:
Fix is in version 21699 and above, wait for the next bundle

bstott, 2023-02-02:
Thanks. I've version 0.21.0.31739. It sure does work.

PR #8266 (2023-01-27): [Gui] allow to specify font for NaviCube
- the problem is that depending on the OS, the font of the NaviCube is hardly readable. For example under Windows 11, there is no Helvetica font and therefore an ugly replacement font is used. On some laptop screens the font is too large or too small etc.
- as solution this PR add the change the NaviCube's font and font size

- the PR also fixes an issue that the position (corner) of the NaviCube was not respected when the NaviCube is recreated. This fix is necessary for the PR therefore included

- as by-product the PR fixes #8082 since every change in [... cut]
commit 588f2d50 (2023-01-27): [Gui] allow to specify font for NaviCube

- the problem is that depending on the OS, the font of the NaviCube is hardly readable. For example under Windows 11, there is no Helvetica font and therefore an ugly replacement font is used. On some laptop screens the font is too large or too small etc.
- as solution this PR add the change the NaviCube's font and font size

- the PR also fixes an issue that the position (corner) of the NaviCube was not respected when the NaviCube is recreated. This fix is necessary for the PR therefore included

- as by-product the PR fixes #8082 since every change i [... cut]