FreeCAD/FreeCAD #8086: [Bug] [crash] Filleting certain edges in part design crashes FreeCAD

eblanton, 2022-12-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (custom)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.6, Qt 5.15.7, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.2
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Issue description

This may be related to #6625, but if I understand the comments there, my version of FreeCAD (0.20.1) should have the fix linked on that issue already applied.  I don't understand the tech stack completely, however, so I could be mistaken.

When selecting the edges in the attached image/CAD file from the Part Design workbench and pressing the Fillet button, FreeCAD crashes with this backtrace:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x3f0c0) [0x7f0f4eab10c0]
#1  0x7f0e7b1f90db in TopLoc_Location::Multiplied(TopLoc_Location const&) const from /app/lib/libTKMath.so.7+0x1b
#2  0x7f0e881364b9 in BRep_Tool::Surface(TopoDS_Face const&, TopLoc_Location&) from /app/lib/libTKBRep.so.7+0x39
#3  0x7f0e8813a4fe in BRep_Tool::Parameters(TopoDS_Vertex const&, TopoDS_Face const&) from /app/lib/libTKBRep.so.7+0x4e
#4  0x7f0e88b64249 in ChFi3d_IsInFront(TopOpeBRepDS_DataStructure&, opencascade::handle<ChFiDS_Stripe> const&, opencascade::handle<ChFiDS_Stripe> const&, int, int, int, int, double&, double&, TopoDS_Face&, bool&, int&, int&, bool&, TopoDS_Vertex const&, bool, bool) from /app/lib/libTKFillet.so.7+0x6b9
#5  0x7f0e88c03896 in ChFi3d_FilBuilder::PerformTwoCorner(int) from /app/lib/libTKFillet.so.7+0x986
#6  0x7f0e88b561c6 in ChFi3d_Builder::PerformFilletOnVertex(int) from /app/lib/libTKFillet.so.7+0x3e6
#7  0x7f0e88b5704a in ChFi3d_Builder::Compute() from /app/lib/libTKFillet.so.7+0xd0a
#8  0x7f0e88cdce56 in BRepFilletAPI_MakeFillet::Build(Message_ProgressRange const&) from /app/lib/libTKFillet.so.7+0x36
#9  0x7f0e89c58afc in PartDesign::Fillet::execute() from /app/freecad/lib/_PartDesign.so+0x7fc
#10  0x7f0f50adcbf3 in App::DocumentObject::recompute() from /app/freecad/lib/libFreeCADApp.so+0x5d3
#11  0x7f0e897cb700 in Part::Feature::recompute() from /app/freecad/lib/Part.so+0x10
#12  0x7f0f50a5891c in App::Document::_recomputeFeature(App::DocumentObject*) from /app/freecad/lib/libFreeCADApp.so+0x2ac
#13  0x7f0f50a73809 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /app/freecad/lib/libFreeCADApp.so+0x549
#14  0x7f0f50b2e0a7 in App::DocumentPy::recompute(_object*) from /app/freecad/lib/libFreeCADApp.so+0x2f7
#15  0x7f0f50b2e1d8 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /app/freecad/lib/libFreeCADApp.so+0x28
#16  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x125fc8) [0x7f0f50379fc8]
#17  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x9c) [0x7f0f5038c2ac]
#18  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x8be1) [0x7f0f502ca811]
#19  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c62e5) [0x7f0f5041a2e5]
#20  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0x96) [0x7f0f5041a956]
#21  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c69d4) [0x7f0f5041a9d4]
#22  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x22144b) [0x7f0f5047544b]
#23  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyRun_StringFlags+0x7c) [0x7f0f5048486c]
#24  0x7f0f506f053e in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /app/freecad/lib/libFreeCADBase.so+0x6e
#25  0x7f0f5140453c in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /app/freecad/lib/libFreeCADGui.so+0x9c
#26  0x7f0f514046dd in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /app/freecad/lib/libFreeCADGui.so+0xed
#27  0x7f0f514048d9 in Gui::Command::updateActive() from /app/freecad/lib/libFreeCADGui.so+0x49
#28  0x7f0e790c0db5 in finishFeature(Gui::Command const*, App::DocumentObject*, App::DocumentObject*, bool, bool) from /app/freecad/lib/PartDesignGui.so+0x3b5
#29  0x7f0e790c1d43 in finishDressupFeature(Gui::Command const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Part::Feature*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, bool) from /app/freecad/lib/PartDesignGui.so+0x663
#30  0x7f0e790c8ded in makeChamferOrFillet(Gui::Command*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) from /app/freecad/lib/PartDesignGui.so+0x10d
#31  0x7f0e790c8efd in CmdPartDesignFillet::activated(int) from /app/freecad/lib/PartDesignGui.so+0x4d
#32  0x7f0f51406a6a in Gui::Command::_invoke(int, bool) from /app/freecad/lib/libFreeCADGui.so+0x2aa
#33  0x7f0f51406f53 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /app/freecad/lib/libFreeCADGui.so+0x133
#34  /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x305adc) [0x7f0f4f1e6adc]
#35  0x7f0f4fd067d2 in QAction::triggered(bool) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x42
#36  0x7f0f4fd093fb in QAction::activate(QAction::ActionEvent) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xab
#37  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x26f12a) [0x7f0f4fe0112a]
#38  0x7f0f4fe01277 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf7
#39  0x7f0f4fefa6ba in QToolButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa
#40  0x7f0f4fd4ecf8 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1a8
#41  0x7f0f4fd0cfce in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#42  0x7f0f4fd15562 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x13c2
#43  0x7f0f513c52f8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x88
#44  0x7f0f4f1aef98 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#45  0x7f0f4fd1366e in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1be
#46  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1d5b28) [0x7f0f4fd67b28]
#47  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1d8eb0) [0x7f0f4fd6aeb0]
#48  0x7f0f4fd0cfce in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#49  0x7f0f513c52f8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x88
#50  0x7f0f4f1aef98 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#51  0x7f0f4f5ecb3d in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x80d
#52  0x7f0f4f5bfeec in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#53  /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x6a71a) [0x7f0f49be471a]
#54  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a1) [0x7f0f4b1ccd71]
#55  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xb0028) [0x7f0f4b223028]
#56  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f0f4b1ca4a3]
#57  0x7f0f4f2072f6 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x66
#58  0x7f0f4f1ada0b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#59  0x7f0f4f1b5b76 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x96
#60  0x7f0f5131846c in Gui::Application::runApplication() from /app/freecad/lib/libFreeCADGui.so+0x198c
#61  FreeCAD(+0x5d33) [0x559d8224fd33]
#62  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x2954a) [0x7f0f4ea9b54a]
#63  /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7f0f4ea9b60b]
#64  FreeCAD(+0x6065) [0x559d82250065]
```

This is currently 100% reproducible for me.

Edges:

[image]

Source file (sorry that it's zipped, it's the only way GitHub would let me attach it):

[crashes.zip](https://github.com/FreeCAD/FreeCAD/files/10298926/crashes.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EdmundHernandezStudent, 2022-12-26:
Here's what I got "13:33:58  Part::Fillet: Link(s) to object(s) 'Pocket004' go out of the allowed scope 'Fillet'. Instead, the linked object(s) reside within 'Body Body'."

adrianinsaval, 2022-12-26:
the report mentions occt 7.6.3 as having the fix while your occt version is 7.6.2, the flatpak version should have been updated with freecad 0.20.2 and occt 7.6.3 recently so can you try updating and trying again?

EdmundHernandezStudent, 2022-12-26:
[image]
Did you mean do select this two edges. The largest one fillets fine but the one on the edge will have trouble if its a 1mm fillet.

adrianinsaval, 2022-12-26:
whether the fillet works or not or how to achieve it is besides the point here, for advice on those ask in the forum. Here what is meaningful is the crash

adrianinsaval, 2022-12-27:
@eblanton the update is now available on flathub, please update and test.

luzpaz, 2022-12-28:
@eblanton ping

eblanton, 2022-12-29:
(Sorry for the delay, holiday schedule!)

I can confirm that the updated flatpak (0.20.2, rev 29603) behaves as expected; no crash, but the corner cannot fillet because of the adjacent geometry.

adrianinsaval, 2022-12-29:
Thanks for testing, that would be a limitation of occt (the geometric kernel) which is out of FreeCAD's control. IMO this issue can then be closed

luzpaz, 2022-12-29:
@adrianinsaval is there an open OCCT fillet ticket we can link to regarding the OCCT limitation?

closed 2022-12-29 by luzpaz