FreeCAD/FreeCAD #8091: [Issue] FreeCad gives error when entering dimensions as a fraction

CrashedAgain1859, 2022-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=74764

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu 22.04.1 LTS (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31391 (Git) AppImage
Build type: Release
Branch: master
Hash: 0ab2608a429f3a5ee9a63da9b66d13557655d7e4
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * toSketch 1.0.1
  * ThreadProfile 1.84.0
  * A2plus 0.4.60g
  * freecad.gears 1.0.0
```

### Subproject(s) affected?

None

### Issue description

Sketcher: Entering imperial dimensions as a fraction (e.g. 1/4 in) or using a fractional entry to edit an existing dimension (eg. editing existing "0.123 in" by entering "/2" ("0.123 in /2")) gives an incorrect result. 
Part, Part Design, and other workbenches: Entering a fraction as a dimension in a data dialog box gives an incorrect calculation e.g. attempting to extrude a sketch to 1/2 inch length by entering "1/2" or "1/2 in" in the dialog box gives an extrusion of 0.2", not 0.500". 
It appears that it is now necessary to be very specific in making the entry, making sure to add a " symbol in just the right place in the entry.....for example entering 1"/2 works but 1/2" does not. 

At best this is just a workaround. 
This used to work but now does not.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-12-28:
most likely introduced by #7124

webmite, 2022-12-30:
It appears [#7124](https://github.com/FreeCAD/FreeCAD/pull/7124) affects far more than just sketcher.

On 'part workbench' adding a cube and attempting to modify Length dimension with unit math in input text box fails spectacularly.

CrashedAgain1859, 2022-12-30:
Yes it does. There is some discussion of this in the forum conversation
referenced.

adrianinsaval, 2023-01-02:
it affects most gui spinboxes

> attempting to modify Length dimension with unit math in input text box fails spectacularly.

be more specific please, nothing can be one if we don't know what issue you are facing.

webmite, 2023-01-02:
> On 'part workbench' adding a cube and attempting to modify Length dimension with unit math in input text box fails spectacularly.

I entered '1 + 1 in' and got a '1 mm' result. Sorry for not being explicit.

adrianinsaval, 2023-01-03:
what unit do you have configured? I cannot reproduce, when I have mm configured `1 + 1 in` results in 26.4 mm and when I configure US customary the result is 2 "

adrianinsaval, 2023-01-03:
> Version: 0.20.1.29410 (Git)

ok I've just now seen this, the issue we are discussing here is exclusive to 0.21.30752 and above, whatever you may experience in 0.20.1 is unrelated, please test on the latest 0.21 available for your platform https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2023-03-05:
https://github.com/FreeCAD/FreeCAD/issues/8091#issuecomment-1366624059  
CC @ConnorWorrell

ConnorWorrell, 2023-03-06:
Thanks for the ping!

Confirmed this was introduced by https://github.com/FreeCAD/FreeCAD/pull/7124

This is an issue on if the parser should assume 1/2in means (1/2)in or 1/(2in). It seems the former is always the correct assumption.

@CrashedAgain1859 I pushed an update to this branch, if you're able to give it a shot.
https://github.com/ConnorWorrell/FreeCAD/tree/SpinboxFractionInput

This modifies the input box text before its passed to the expression parser to give the requested behavior. Also improves the behavior when handling units that are part of division, multiplication, and exponentiation operations.

I think this is the requested and correct behavior, this assumption does lead to some interesting math:
W/ US Customary units:
- 1/2 -> .5in
- 1/2in -> .5in
- 1/2mm -> 19.685 thou
- 1/(1in+1) -> .5in
- ((1in+1in)/(1in*1in))*2^(1in+1in) -> 8in

CrashedAgain1859, 2023-03-07:
I think that will fix the problem, it looks like most of the potential
problem inputs are covered.
Will try to check it once it makes it into a weekly build appimage.

Thanks

adrianinsaval, 2023-03-22:
these are ok and are the desired behavior:
> * 1/2 -> .5in
> * 1/2in -> .5in
> * 1/2mm -> 19.685 thou

these do not make sense:
> * 1/(1in+1) -> .5in
> * ((1in+1in)/(1in*1in))*2^(1in+1in) -> 8in

and should rightfully result in error

CrashedAgain1859, 2023-03-23:
Even though "((1in+1in)/(1in*1in))*2^(1in+1in)" is a very odd way to enter
the expression, it must be admitted that the editor is evaluating the
expression correctly (ans = 8), (or at least evaluating it in accordance
with what might reasonably be expected to be the user's intentions) so I
don't think it should generate an error.
 What would the error be?
 "Units mismatch" does not make sense, the only units listed are inches
which are a match for the setting in preferences. "Unable to parse the
expression" makes sense but does not give the user any information to make
a correction.

Question: after the changes you are working on, is the editor still able to
cope with mixed units?
Does "3 in + 6 mm" still work? Result used to be 3.236" in the old (ie
currently stable) version - but only if entered in though the formula
editor. If interred directly into a properties box, the answer would be 3".

adrianinsaval, 2023-04-21:
Is this solved now?

CrashedAgain1859, 2023-04-21:
Yes, is it solved now?

Have the revisions been incorporated into a weekly build appimage? That's
the only way I can check....

adrianinsaval, 2023-04-21:
yes they are included in latest weekly

ryan-pebk, 2023-04-22:
Hi, there appears to be a new bug related to this fix.  It occurs when a dimension is less than 1 foot.  

For example: -1'2" is calculated as -1' + 2" = -10" instead of -14". 

This happens when editing an existing dimension using the spinboxes but does not seem to affect the calculator.  If this should go in as a new bug, please accept my apologies but it appears related due to timing.

ConnorWorrell, 2023-04-22:
> Hi, there appears to be a new bug related to this fix. It occurs when a dimension is less than 1 foot.
> 
> For example: -1'2" is calculated as -1' + 2" = -10" instead of -14".

Good catch!, there are always so many edge cases.

I wasn't able to reproduce this, but still see a similar issue. Setting units to Building-US and inputting -1'-1", works correctly, but when editing the dimension again, the input box shows 0 (when it should show -1'-1"). It appears this was introduced in https://github.com/FreeCAD/FreeCAD/pull/7124.

This behavior is caused by the expression parer not being able to parse the #'#" format units. 
[image]

Previously this input was handled in sketcher, and does not work anywhere else, ex: spreadsheets.
[image]

If the #'#" format is a valid input then this should be handled in the expression parser to maintain consistency.

ConnorWorrell, 2023-04-26:
It looks like a bug for this exists here. https://github.com/FreeCAD/FreeCAD/issues/5909

adrianinsaval, 2023-04-26:
> but does not seem to affect the calculator

does anybody know what parser is used there?

I think we can close this issue now as we have a dedicated issue for the multiunit input problem and the fraction problem described here has been solved. I'm not sure about the calculator but if there is a problem there a new issue should be created about that.

closed 2023-04-26 by adrianinsaval

CrashedAgain1859, 2023-04-27:
I have tested the latest appimage and the parser seems to work as expected
with only one (very minor) bump. Sketcher displays to 6 decimal places even
though it is set to show only 3. (Screenshot attached).

And (I suspect this is not related) - I have custom keys ctrl-k set to open
a new sketch. It used to work on all workbenches, now it only works when
you are already in the sketcher workbench, which defeats the purpose of
having it. Any idea why this might be?

adrianinsaval, 2023-04-27:
Those are unrelated reports, head to the help forum for these please, the number of decimals in sketcher is known to not be affected by the preference, IIRC sketcher has it's precision hardcoded sure to the solver.  The shortcut thing is hard to know but might just be some conflicting shortcut or you need to configure sketcher to autoload in startup

CrashedAgain1859, 2023-04-27:
Thanks for the reply. The sketches decimals issue is not really important,
it still works fine.
I will take up the other thing elsewhere. Sketches is already autoloader,
BTW.