FreeCAD/FreeCAD #8093: [Issue] ProDark: Light text background colors result in illegible text

life5ign, 2022-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=34&t=55134&p=649728#p649728

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * Rocket 2.8.5
  * fasteners 0.4.27
  * Help 1.0.3
  * parts_library
  * lattice2 1.0.0
  * Curves 0.6.5
  * LCInterlocking
  * A2plus 0.4.60g
  * FreeCAD-library-eac93dcfa999af5c1b2682eda999fbac9f4d63df.zip
[/code]
```

### Subproject(s) affected?

None

### Issue description

Excellent theme! I've noticed on my system that certain text, when highlighted, has a white or a yellow background, which, with the white text, makes it illegible:

# Active Part or Body in Combo View tree:

[image]

# Annotation in TechDraw:

[image]

# my system
[image]

### Anything else?

I tried resetting all in my Prefereneces, as well as uninstalled all preference packs, restarted freecad, and it didn't fix the issue.  This is how it looks with no style sheet:

[image]

[image]

It's a bit of a drag to not be able to read the name of the active part and body; the annotations is a bit less of an annoyance.  Please let me know if there's any other information you need, and thanks for all the work you do to help us build and create.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-12-27:
this is not directly related to stylesheet as it is either hardcoded or a preference, please see https://github.com/FreeCAD/FreeCAD/issues/7623#issuecomment-1287544999
https://forum.freecadweb.org/viewtopic.php?p=635039#p635039
and
https://forum.freecadweb.org/viewtopic.php?f=34&t=70672

life5ign, 2022-12-27:
Hi, thanks for the reply--I can't believe I didn't find any of those in my google, forum, github searches.  Thanks, looks like the team is aware of it.

MisterMakerNL, 2022-12-27:
Hey thanks for reporting.
The first issue is fixed in my extremeprodark, but you need too restart Freecad. And apply the preference pack as it's not in the theme file.
The second issue I was sure we fixed it, or looked into it but I can't find where it was discussed.

life5ign, 2022-12-27:
Thanks!  I just installed extremeprodark (great theme), and the first is definitely fixed, but I still get the issue with annotations in techdraw: 
[image]

adrianinsaval, 2022-12-27:
yeah that's not themable IIRC, IMO the real issue is that we are using the wrong widget type for this, we are using a text editor to edit a list and adding some workarounds to have line highlight, instead of just using QListWidget that is made for lists and would have highlighting that is themable.

WandererFan, 2023-01-04:
TechDraw Annotation fixed by PR #8123 / d63ac0f7f6

luzpaz, 2023-01-08:
Can this be closed ?

WandererFan, 2023-01-08:
> Can this be closed ?

No.  The PropertyEditor still uses the old QLineEdit based editor for PropertyStringList, which doesn't respect themes.

luzpaz, 2023-01-08:
@WandererFan should the ticket then be renamed ?

WandererFan, 2023-01-09:
No need to rename the ticket,  the PropertyEditor has the same illegible text in dark mode.

Intuos5, 2023-01-25:
There's another window I came across that has inelligible text:
[image]
This is in the Dark-Orange stylesheet.

I also modified the Darker-Green stylesheet and had the same issue there:
[image]
See: https://forum.freecadweb.org/viewtopic.php?style=2&p=655842#p655842

MisterMakerNL, 2023-03-02:
All of these are pretty much fixed by now. Although some are a bit of a workaround like adding 
```
          <FCParamGroup Name="Spreadsheet">
            <FCText Name="AliasedCellBackgroundColor">#9b4de6</FCText>
          </FCParamGroup>

```
Too the Stylesheet CFG file.
Techdraw dev did a great job on fixing it on his part!
[image]
Can go further in how to fix all the old style-sheets.... here https://github.com/FreeCAD/FreeCAD/issues/7623

luzpaz, 2023-09-17:
bumping for status

MisterMakerNL, 2024-07-22:
Pro dark has been removed from main to the theme, repro. 
This can be moved or closed, for the other theme's we have other topics.

luzpaz, 2024-07-22:
>> Can this be closed ?

> No.  The PropertyEditor still uses the old QLineEdit based editor for PropertyStringList, which doesn't respect themes. 

@WandererFan checking in

obelisk79, 2024-08-17:
@maxwxyz this should either be moved to the FreeCAD-themes repo or closed.