FreeCAD/FreeCAD #8098: [Issue] AddonManager Gui tests fail with 'Qt has caught an exception thrown from an event handler.'

0penBrain, 2022-12-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu 22.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31390 +92 (Git)
Build type: Debug
Branch: MyBranch
Hash: 02c5bcc0b2bff1a3a6811faa816d3ec3a325c201
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: French/France (fr_FR)
```

### Subproject(s) affected?

Addon Manager

### Issue description

AddonManager Gui tests fail with a Qt error message.
In some circumstances it can just stuck the unit tests (which is a problem for the CI at least).
To reproduce :
 * Start FC
 * Got to Test Framework WB
 * Click "Self-test"
 * Choose "TestAddonManagerGui"
 * Start

At some point it fails with some
`Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.`

### Anything else?

@chennes I didn't investigated much on this, but I'm not sure it's actually directly linked to test code, but may show an underlying issue in the AddonManager itself.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2022-12-28:
It's a shame it doesn't say what the exception was! It would be easier to track down then. I'll spool up a Linux instance and give it a test, I cannot reproduce on Windows.

0penBrain, 2022-12-28:
IMO that won't be so easy to debug because I guess the bug happens when using multiple threads.
If it can help, it not always stuck FC (I'd say 50% of times here), but when it do it seems to be always on same dialog (below 2 example runs):
[image]
[image]

chennes, 2022-12-28:
Doubly difficult because I cannot reproduce at all: I just ran the tests ten times on my Linux VM and never had an issue. I even tried running in French, and still could not reproduce. I'm on:
```
OS: Ubuntu 22.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31446 (Git)
Build type: Release
Branch: master
Hash: af3c057f995ab911c4eae24d158e514ddd21f778
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.0.dev
Locale: English/United States (en_US)
Installed mods: 
  * BaseTest 0.0.1
  * Curves 0.6.5
  * 3DfindIT 1.2.0
```

0penBrain, 2022-12-28:
From what I see from the CI I'm working on, it's 100% reproducible on Ubuntu 20.04. Not really on Ubuntu 22.04.

chennes, 2022-12-30:
Even on Ubuntu 20.04 LTS I cannot reproduce this exact problem, but I do see an issue where the correct window is not being automatically closed (probably I'm not giving it enough time to open). I'll see if fixing that issue resolves this one.

chennes, 2023-01-02:
OK, I've just pushed a commit that I am hoping addresses this issue: when you have a minute, can you re-test?

0penBrain, 2023-01-09:
> OK, I've just pushed a commit that I am hoping addresses this issue: when you have a minute, can you re-test?

Hi @chennes . I was AFK last week. :) Just tested. It'sa bit better as now the tests sometimes are passing and sometimes are still stuck. When the tests are passing, I still see these warnings messages, but they appear at a different (later) place in the log.
They appear between list of available addons and lines starting with `Addon added ...`. Below the log in this case if it can help:
```
10:01:42  Optional dependency on disallowed_package ignored because it is not in the allow-list
10:01:42  Nothing really failed, this is a test of the dialog box
10:01:42  The following Python packages are allowed to be automatically installed:
10:01:42    * aiofiles
10:01:42    * pyoptools
10:01:42    * geomdl
10:01:42    * pydocx
10:01:42    * blinker
10:01:42    * autobahn
10:01:42    * gmsh-dev
10:01:42    * hausdorff
10:01:42    * etabs_api
10:01:42    * numba
10:01:42    * qtrangeslider
10:01:42    * pynastran
10:01:42    * olefile
10:01:42    * openpyxl
10:01:42    * gmsh
10:01:42    * xlwt
10:01:42    * ply
10:01:42    * awkward
10:01:42    * pillow
10:01:42    * xlutils
10:01:42    * markdown
10:01:42    * pandas
10:01:42    * py-slvs
10:01:42    * nodeeditor
10:01:42    * nine
10:01:42    * qtpy
10:01:42    * rhino3dm
10:01:42    * xlrd
10:01:42    * lxml
10:01:42    * scikit-image
10:01:42    * docutils
10:01:42    * pygit2
10:01:42    * matplotlib
10:01:42    * openexr
10:01:42    * requests
10:01:42    * ocp
10:01:42    * pyyaml
10:01:42    * trimesh
10:01:42    * capytaine
10:01:42    * comtypes
10:01:42    * pycollada
10:01:42    * ifcopenshell
10:01:42    * scipy
10:01:42    * msgpack
10:01:42    * numpy
10:01:42    * ezdxf
10:01:42    * beautifulsoup4
10:01:42  The following Python packages are allowed to be automatically installed:
10:01:42    * aiofiles
10:01:42    * pyoptools
10:01:42    * geomdl
10:01:42    * pydocx
10:01:42    * blinker
10:01:42    * autobahn
10:01:42    * gmsh-dev
10:01:42    * hausdorff
10:01:42    * etabs_api
10:01:42    * numba
10:01:42    * qtrangeslider
10:01:42    * pynastran
10:01:42    * olefile
10:01:42    * openpyxl
[... 71 lines cut ...]
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.

10:01:45  Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.

10:01:45  Addon added: 3D_Printing_Tools
10:01:45  Addon added: 3DfindIT
10:01:45  Addon added: A2plus
10:01:45  Addon added: AirPlaneDesign
10:01:45  Addon added: Alternate_OpenSCAD
10:01:45  Addon added: animation
10:01:45  Addon added: AnimationFreeCAD
10:01:45  Addon added: ArchTextures
10:01:45  Addon added: assembly2
10:01:45  Addon added: Assembly3
10:01:45  Addon added: Assembly4
10:01:45  Addon added: Autoload
10:01:45  Addon added: BCFPlugin
[Continues listing without error]
```

chennes, 2023-01-17:
Resolved by d6b3efc7a84588174666efd1ab664f2fbf769377

closed 2023-01-17 by chennes