FreeCAD/FreeCAD #8099: [Problem] Error when trying to create a sketch on the Part Design workbench

PX2Z3676, 2022-12-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Portuguese/Brazil (pt_BR)
Installed mods: 
  * 3D_Printing_Tools
  * CurvedShapes 1.0.4
  * fasteners 0.4.50
  * fcgear 1.0.0
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

Dear all,

First of all I used the workbench Image to load a file as reference for the sketch, it´s all right until now.

I changed to the Part Design workbench and I created a sketch, so I started to "follow" the lines in the image using the PolyLine tool. After I finished the draw, as a continuous polygon I clicked on the CLOSE button to create inded the vectored lines, but in the report view I received a lot of error messages like it:  **Updating geometry: Error build geometry(34): Both points are equal**

I did a lot of tests, with different sequences to create the polylines and I realize if I draw up to 35 lines, when I clicked on the Close button I got success...

I did another test, I started the process without "import an image as reference", and I created a sketch and it is possible to draw more than 35 lines, in fact, a lot of lines and if I click on the Close button, again I got success.

I hope to help you with this issue, because I like a lot of the FreeCad, it help me a lot with 3d Printed process and CNC routering.

If you need more information about the issue, please, contact me.

Thanks in advance.

André Luiz Guimarães - Sáo Paulo - Brazil

### Anything else?

not for the moment

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PX2Z3676, 2022-12-29:
[image]

luzpaz, 2022-12-29:
Please zip the FCStd demo file and upload it.

adrianinsaval, 2022-12-29:
Most likely an incorrectly constrained sketch

PX2Z3676, 2022-12-29:
[NovoLogoLexos.zip](https://github.com/FreeCAD/FreeCAD/files/10320961/NovoLogoLexos.zip)

PX2Z3676, 2022-12-29:
@luzpaz , good afternoon. Above, I uploaded the zip file with the sketch. Thanks in advance for the interest to help me. A big hug.

adrianinsaval, 2022-12-29:
The problem is in constraint18 which is trying to constraint both points of edge35 as coincident which results in an invalid 0 length line. I don't know how this constraint came to be, do you know how to reproduce this and can you make a screencast of it?

PX2Z3676, 2022-12-29:
Hi @adrianinsaval , I will check about the contraint 18 and try to reproduce that, and attach the screencast here. Thanks in advance. A big hug!

PX2Z3676, 2022-12-29:
@luzpaz & @adrianinsaval , finally I found the origin of the my issue. In the Image workbench, when "openning" the file, with the Create Image Plane button, after to choose the orientation, I was keeping the offset as zero, I saw a video in the Youtube and the instructor put an negative number on that field, -5mm for example, I tried the same steps and using a negative number on the offset, when I changed to the PartDesign workbench I did the same procedures and now I got success, as the image below:

PX2Z3676, 2022-12-29:
[image]

adrianinsaval, 2022-12-29:
I highly doubt that has any effect on autoconstraints but ok if it's not reproduceable I think we can assume it was just a missclick or something like that. Please close the issue

PX2Z3676, 2022-12-29:
[image]

PX2Z3676, 2022-12-29:
@adrianinsaval , sorry, but I tried more than 10 times to create it, and after I used the -5mm as offset when creating the plane image It worked in a first time. But, anyway I would like to thank you and @luzpaz in this issue, and I will close it now. Best regards.

PX2Z3676, 2022-12-29:
Thanks @luzpaz and @adrianinsaval !

closed 2022-12-29 by PX2Z3676