FreeCAD/FreeCAD #8100: [Problem] Helix vanishes when "turns" parameter is set to certain values, reappears when turns is changed to other values.

dajt1725, 2022-12-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Rocky Linux 8.7 (Green Obsidian) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

1. Open the attached part
2. Click on the LowerInnerHandrail helix
3. Note that the "turns" parameter is .15, which corresponds to the short handrail attached to the wall.
4. Change the "turns" parameter to .20
5. Note that instead of the handrail changing length, it vanishes.
6. Change the "turns" parameter to .30
7. Note that the handrail reappears, longer.
[ShowHelixBug.zip](https://github.com/FreeCAD/FreeCAD/files/10317167/ShowHelixBug.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dajt1725, 2022-12-29:
This bug is new in 0.20.2  It displays correctly in 0.20.1

luzpaz, 2022-12-29:
@donovaly can you reproduce (I can't test ATM)

dajt1725, 2022-12-29:
Yes.  The steps to reproduce are in the description.

donovaly, 2022-12-30:
There is no bug in FC but in your design. You attached the helix directly onto the surface. Since a curve is always a numerical approximation, with this method, the Helix calculation algorithm can return a geometric point for the helix outside the existing part. And then OCC cannot calculate further. Therefore assure you start a helix always a tiny bit inside the existing part so that numerical deviations cannot have an influence. I have done this here:
[ShowHelixBug-correct.zip](https://github.com/FreeCAD/FreeCAD/files/10322609/ShowHelixBug.zip)

For the next time please first report potential bugs at the forum that others can verify there is a real bug.

closed 2022-12-30 by donovaly