FreeCAD/FreeCAD #8102: [Issue] FreeCAD hangs generating Vcarve path for a ShapeString

br4243, 2022-12-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Fedora Linux 37 (Workstation Edition) (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31445 (Git)
Build type: Release
Branch: master
Hash: 8b0df1dc936b544091f6a2d68df1c1a14ae3de5b
Python 3.11.0, Qt 5.15.7, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.0
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Path

### Issue description

Path::orthognalProjection() generates nan values for the "proj" variable when the segment length is zero, causing FreeCAD to hang.  To reproduce this:
Start FreeCAD; open "carvetest.FCStd"
Activate Path workbench
In the Model tree select Job...Model...Model-31/64
Select the Engraving...Vcarve tool
Click OK
FreeCAD hangs (with a spinning cursor on my system)

[carvetest.zip](https://github.com/FreeCAD/FreeCAD/files/10322578/carvetest.zip)

Attaching to the hung process:

$ gdb -p 1789953
<snip>
0x00007f788436a8b3 in gp_XYZ::Reverse (this=<optimized out>, this=<optimized out>) at /usr/src/debug/opencascade-7.5.0-7.fc37.x86_64/src/gp/gp_XYZ.lxx:228
228       x = - x;
Missing separate debuginfos, use: dnf debuginfo-install qt5-qtwebengine-freeworld-5.15.10-4.fc37.x86_64
(gdb) where
#0  0x00007f788436a8b3 in gp_XYZ::Reverse()
    (this=0x7ffeb1432070, this=<optimized out>)
    at /usr/src/debug/opencascade-7.5.0-7.fc37.x86_64/src/gp/gp_XYZ.lxx:228
#1  gp_Dir::Reverse() (this=0x7ffeb1432070, this=<optimized out>)
    at /usr/src/debug/opencascade-7.5.0-7.fc37.x86_64/src/gp/gp_Dir.lxx:225
#2  Geom_Conic::Reverse() (this=0x61de110)
    at /usr/src/debug/opencascade-7.5.0-7.fc37.x86_64/src/Geom/Geom_Conic.cxx:29
#3  0x00007f7884387e71 in Geom_TrimmedCurve::Reverse() (this=0x61de210)
    at /usr/src/debug/opencascade-7.5.0-7.fc37.x86_64/src/Standard/Standard_Handle.hxx:130
#4  0x00007f7867d06fed in Part::GeomArcOfParabola::setRange(double, double, bool) ()
    at /data/build/FreeCAD/build/Mod/Part/Part.so
#5  0x00007f78ecf12ff0 in Path::VoronoiEdgePy::toShape(_object*) ()
    at /data/build/FreeCAD/build/Mod/Path/PathApp.so
#6  0x00007f78ecf0d884 in Path::VoronoiEdgePy::staticCallback_toShape(_object*, _object*) () at /data/build/FreeCAD/build/Mod/Path/PathApp.so
#7  0x00007f79193d8228 in cfunction_call
    (func=0x7f78c9184450, args=<optimized out>, kwargs=<optimized out>)
    at /usr/src/debug/python3.11-3.11.0-1.fc37.x86_64/Objects/methodobject.c:553
#8  0x00007f79193b9b13 in _PyObject_MakeTpCall
    (tstate=0x7f79197200d8 <_PyRuntime+166328>, callable=0x7f78c9184450, args=<optimize--Type <RET> for more, q to quit, c to continue without paging--q
Quit
(gdb) frame 5
#5  0x00007f78ecf12ff0 in Path::VoronoiEdgePy::toShape(_object*) ()
   from /data/build/FreeCAD/build/Mod/Path/PathApp.so
(gdb) x/i $pc
=> 0x7f78ecf12ff0 <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6124>:     lea    -0x5b0(%rbp),%rax
(gdb) x/20i $pc-60
   0x7f78ecf12fb4 <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6064>:     cmp    %bh,%ah
   0x7f78ecf12fb6 <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6066>:     (bad)  
   0x7f78ecf12fb7 <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6067>:     decl   -0x73(%rax)
   0x7f78ecf12fba <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6070>:     test   %edx,-0x6(%rax)
   0x7f78ecf12fbd <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6073>:     (bad)  
   0x7f78ecf12fbe <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6074>:     decl   -0x77(%rax)
   0x7f78ecf12fc1 <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6077>:     (bad)  
   0x7f78ecf12fc2 <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6078>:     mov    %rax,%rdi
   0x7f78ecf12fc5 <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6081>:
    call   0x7f78ece09940 <_ZN4Part17GeomArcOfParabola9setHandleERKN11opencascade6handleI13Geom_ParabolaEE@plt>
   0x7f78ecf12fca <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6086>:     movsd  -0x40(%rbp),%xmm0
   0x7f78ecf12fcf <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6091>:     mov    -0x38(%rbp),%rdx
   0x7f78ecf12fd3 <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6095>:     lea    -0x5b0(%rbp),%rax
   0x7f78ecf12fda <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6102>:     mov    $0x0,%esi
   0x7f78ecf12fdf <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6107>:     movapd %xmm0,%xmm1
   0x7f78ecf12fe3 <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6111>:     movq   %rdx,%xmm0
   0x7f78ecf12fe8 <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6116>:     mov    %rax,%rdi
   0x7f78ecf12feb <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6119>:     call   0x7f78ece23470 <_ZN4Part17GeomArcOfParabola8setRangeEddb@plt>
=> 0x7f78ecf12ff0 <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6124>:     lea    -0x5b0(%rbp),%rax
   0x7f78ecf12ff7 <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6131>:     mov    %rax,%rdi
   0x7f78ecf12ffa <_ZN4Path13VoronoiEdgePy7toShapeEP7_object+6134>:     call   0x7f78ece09e70 <_ZNK4Part17GeomArcOfParabola6handleEv@plt>
(gdb) x/fg $rbp-0x40
0x7ffeb14327e0: nan(0x8000000000000)
(gdb) x/fg $rbp-0x38
0x7ffe50ad27b8: nan(0x8000000000000)
(gdb) q

Path::VoronoiEdgePy::toShape() is passing nans to Part::GeomArcOfParabola::setRange()

Further investigation revealed the nan comes from orthognalProjection() in Path/App/VoronoiEdgePyImp.cpp
when the 2nd argument, the segment, has zero length.  Applying the following patch resolves the problem:
```
$ git diff
diff --git a/src/Mod/Path/App/VoronoiEdgePyImp.cpp b/src/Mod/Path/App/VoronoiEdgePyImp.cpp
index efe68022bd..e5fa98d2c2 100644
--- a/src/Mod/Path/App/VoronoiEdgePyImp.cpp
+++ b/src/Mod/Path/App/VoronoiEdgePyImp.cpp
@@ -67,7 +67,9 @@ namespace {
     // calculate the orthogonal projection of p onto s
     // ((p dot s) / (s dot s)) * s (https://en.wikibooks.org/wiki/Linear_Algebra/Orthogonal_Projection_Onto_a_Line)
     // and it back by original offset to get the projected point
-    double proj = (p.x() * s.x() + p.y() * s.y()) / (s.x() * s.x() + s.y() * s.y());
+    double proj = 0;
+    if (length(segment) != 0)
+      proj = (p.x() * s.x() + p.y() * s.y()) / (s.x() * s.x() + s.y() * s.y());
     Voronoi::point_type pt;
     {
       pt.x(offset.x() + proj * s.x());
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-12-30:
Thank you for your efforts. In future, please be extra mindful to put output in the code tags, in this case gdb output spammed a whole bunch of unrelated tickets because Github thought you were referencing them. TIA

br4243, 2022-12-30:
My apologies for the spam; I won't make that mistake again.
I should have mentioned above that the zero length segment(s) causing that problem are generated in Vcarve.py's ObjectVcarve.buildPathMedial.insert_many_wires() by the points from wire.discretize().  The points are distinct as generated, but some of them are close enough so that doing arithmetic on them makes at least one of the segments' length zero.

PR #10915 (2023-10-12): [Path] Prevent division by zero
Closes #8102.

If one follows the steps described by @br4243, `Voronoi::point_type orthognalProjection` dividing by zero, returns NaN. This NaN is fed to `Part::GeomArcOfParabola::setRange` causing freeze. This pull request fixes the issue.
commit 5e5f000c (2023-10-12): [Path] Prevent division by zero

Closes #8102.
closed 2023-10-12 by sliptonic