FreeCAD/FreeCAD #8104: [Issue] Segmentation fault when right clicking Sketcher constraint list

ion201, 2022-12-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Arch Linux (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: makepkg
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.7, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

Steps to reproduce:

1. Create a new empty project
2. Open the Sketcher workbench
3. Create a new sketch
4. Click "Create Line"
5. Right click on the empty constraints list
6. FreeCAD has now segfaulted

Sorry I do not have debug symbols, but hopefully this is easy to reproduce. It happens every time for me.  

Backtrace:
```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x38a00) [0x7fc19fa51a00]
#1  0x7fc1571688ca in SketcherGui::ConstraintView::contextMenuEvent(QContextMenuEvent*) from /usr/lib/freecad/lib/SketcherGui.so+0x8ca
#2  0x7fc1a0fafab9 in QWidget::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0xe99
#3  0x7fc1a105d823 in QFrame::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0x23
#4  0x7fc1a028cc02 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x92
#5  0x7fc1a0f78b0c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x8c
#6  0x7fc1a0f7d342 in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x3b2
#7  0x7fc1a2648cb1 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x91
#8  0x7fc1a028cf98 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x128
#9  /usr/lib/libQt5Widgets.so.5(+0x1cddf4) [0x7fc1a0fcddf4]
#10  /usr/lib/libQt5Widgets.so.5(+0x1cf15e) [0x7fc1a0fcf15e]
#11  0x7fc1a0f78b1c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x9c
#12  0x7fc1a2648cb1 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x91
#13  0x7fc1a028cf98 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x128
#14  0x7fc1a073f15c in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/libQt5Gui.so.5+0x6ec
#15  0x7fc1a07289a5 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Gui.so.5+0xb5
#16  /usr/lib/libQt5XcbQpa.so.5(+0x64710) [0x7fc19a9e6710]
#17  /usr/lib/libglib-2.0.so.0(g_main_context_dispatch+0x19b) [0x7fc19e71687b]
#18  /usr/lib/libglib-2.0.so.0(+0xacc89) [0x7fc19e76dc89]
#19  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x32) [0x7fc19e715132]
#20  0x7fc1a02d7c4c in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x6c
#21  0x7fc1a028573c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x12c
#22  0x7fc1a0290269 in QCoreApplication::exec() from /usr/lib/libQt5Core.so.5+0x99
#23  0x7fc1a25c34f4 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0x2064
#24  freecad(+0x3990) [0x556a2bb64990]
#25  /usr/lib/libc.so.6(+0x23290) [0x7fc19fa3c290]
#26  /usr/lib/libc.so.6(__libc_start_main+0x8a) [0x7fc19fa3c34a]
#27  freecad(+0x4a55) [0x556a2bb65a55]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-10:
@adrianinsaval do you mind testing this bug?

adrianinsaval, 2023-01-11:
confirmed and it's also reproducible on 0.21 appimage so I think this is not archlinux specific
```
OS: Linux 5.15.85-1-lts (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31328 (Git) AppImage
Build type: Release
Branch: master
Hash: 9e702c016bcc6318ea35a69437241ae67f92524d
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
Installed mods: 
```

luzpaz, 2023-01-11:
@AjinkyaDahale any idea what is generating this problem in Sketcher?

AjinkyaDahale, 2023-01-11:
I'll take a look. Also paging @PaddleStroke since he often works om the taskview.

AjinkyaDahale, 2023-01-12:
The crash appears to happen in the line https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Sketcher/Gui/TaskSketcherConstraints.cpp#L454, but I'm not able to track with gdb how this is happening. The `clearSelection` can probably be removed since this is already done when releasing any handlers (which has to happen if the line is to be reached).

AjinkyaDahale, 2023-01-12:
@abdullahtahiriyo @wwmayer any idea what might be happening?

PR #8178 (2023-01-13): [Sketcher] Prevent dereferencing `nullptr` in `ConstraintView`
Fixes #8104.

As described in that issue, a crash happens under the following conditions:
1. A `DrawSketchHandler` is active.
2. No constraint is selected.
3. Context menu on the `ConstraintView` is triggered.
commit e47147ea (2023-01-13): [Sketcher] Prevent dereferencing `nullptr` in `ConstraintView`

Fixes #8104.

As described in that issue, a crash happens under the following
conditions:
1. A `DrawSketchHandler` is active.
2. No constraint is selected.
3. Context menu on the `ConstraintView` is triggered.
closed 2023-01-13 by wwmayer