FreeCAD/FreeCAD #8105: FreeCad 0.20 locking up generating circular parts

imtheman0313, 2022-12-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 1909
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3D_Printing_Tools
  * FCGears
[/code]
```

### Subproject(s) affected?

Part

### Issue description

I'm not sure exactly if the issue lies with FreeCad or something in Windows, actually; but I figure I would start here.  

Issue::  Any part generated in Part Workbench that includes any type of circle (ie sphere, toroid, cone, cylinder) will lock FreeCad up for about 5-10 minutes.  During the lock, memory usage *crawls* up at about 5MB/s, until the part is finally placed into the workspace.  Moving the view around is very sluggish with any of these objects on the screen, but seems fine with rectangles and the like.  The issue arose last weekend on this particular machine, on 0.18.  I installed 0.20 hoping that the issue was something that had been fixed with the update, with no joy. The problem does NOT affect the desktop.  The only item that has changed since the last time FreeCad worked without issue was a RAM stick.  The machine with the issue is a laptop (specs below) that shipped with 2x8GB sticks.  One of the slots recently failed, so I moved it to 1x16GB.  I'm not sure if that might cause this or not.  I swear FreeCad worked just fine on this machine just a few months ago, but I have no evidence to verify that.  The laptop has had it a bit rough, as it used to be my automotive tuning laptop, and recently got upgraded to print server.  Any more information required will have to be requested in baby talk, as I'm most decidedly NOT a software person.

Specs:
Laptop::  ASUS TUF706IU
CPU:  Ryzen 7 4800H with Radeon Graphics, 2900 Mhz, 8 Core(s), 16 Logical Processor(s)
RAM:  1x16GB (Corsair Vengence I believe)
Video:  Nvidia 1660Ti Mobile
Windows:  Pro 10.0.18363 Build 18363

Desktop::  Custom Built
CPU:  FX-8350 4.8GHz  "8"core
RAM:  32GB Ripjaws 
Video:  ASUS Strix 1060 6GB
Windows:  Same as above

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-12-31:
Please attach example file(s)

imtheman0313, 2022-12-31:
There's no file to example.  I simply open a new document, click any of the buttons to add a part in Part Workbench, and the program locks up.  Windows and other programs continue to function normally.  It's quite strange, actually.  Also, I have a video, but the file is enormous (nearly a GB for under 10 min).  I'm currently fighting with Windows to get the thing modified and down converted, but it's fighting me every step of the way.  Is there anywhere I might upload such a file?  I do not have any accounts with any kind of file sharing server, nor do I desire to create any new accounts anywhere, so the solution would have to be a P2P, FTP transfer, or the like (which is it's own set of issues).

luzpaz, 2022-12-31:
You can use [handbrake](https://handbrake.fr/), but the clip isn't necessary if it's as simple to reproduce as you mention. Can you use try using the [latest windows stable v0.20.2](https://github.com/FreeCAD/FreeCAD/releases/tag/0.20.2)

imtheman0313, 2022-12-31:
Result is the same after the update.  I'm beginning to suspect it may actually be a Windows issue.  I have been thinking of reinstalling the OS to simply wipe all the old things from it, and begin anew.  Is there any method of watching exactly what FreeCad is doing, via a terminal or some such thing?  Perhaps there may be a clue there?  Also, is there a way to change what hardware FreeCad uses to render?  Perhaps the RAM change has done something funny with the on-board video, and I need to render on CPU instead?  Or vice versa.

imtheman0313, 2023-01-01:
Small update, I did try enabling OpenGL rendering, with no change.  I opened the inbuilt console, and all that shows is generation of the object.  I guess what I would need to find is a debug console?

imtheman0313, 2023-01-08:
Update::  I opened the console, and did a little digging (a tiny tiny bit).  The part create line works fine; the part make active line is fine.  The issue arises during the recompute.  I also deleted the object this way, one line at a time, with the same result.  Somehow, recompute is causing only this machine to lock up.

imtheman0313, 2023-01-08:
Unsure exactly what I have done, but everything seems to be functioning normally, now.  I have uninstalled FreeCAD twice, and restarted Windows a few times (normally this machine stays up and running for months on end), and clicking any of the previously problematic parts now performs the requested action, as intended.  I'm going to close the ticket, and chalk it up to the oddities of Windows.

closed 2023-01-08 by imtheman0313