FreeCAD/FreeCAD #8107: [Issue] Subtractive Primitive tool Part Design does not work

fvdhom, 2022-12-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
[/code]
```

### Subproject(s) affected?

None

### Issue description

I followed this lesson on Youtube.  From 00:14:28 about Substractive Primitive does not work. Nut is not going to be cut in half. See this Youtube lesson: 
https://www.youtube.com/watch?v=3a74uw3Jskc

Frans van den Hombergh
Frans@Kingfield.nl

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-12-31:
Can you upload a zipped .FCStd file with a step-by-step way to recreate the bug ?

luzpaz, 2023-01-10:
@fvdhom please upload example file

Syres916, 2023-03-01:
This looks like user error to me, I've tested the subtraction in using the following successfully:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * CurvedShapes 1.0.4
  * freecad.gears 1.0.0
  * Curves 0.6.6
  * PieMenu 1.2.4
  * Plot 2022.4.17
  * AirPlaneDesign 0.4.0
  * fasteners 0.4.54
```

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32146 (Git)
Build type: Release
Branch: master
Hash: 04d494cb75e180a4805feedc6d9c96e2ec306f97
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.3.0
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * CurvedShapes 1.0.4
  * freecad.gears 1.0.0
  * Curves 0.6.6
  * PieMenu 1.2.4
  * Plot 2022.4.17
  * AirPlaneDesign 0.4.0
  * fasteners 0.4.54 
```

luzpaz, 2023-03-01:
Thanks @Syres916  
Closing since we haven't heard back from @fvdhom.  
BTW. @donovaly can you try to reproduce on windows ?

closed 2023-03-01 by luzpaz