FreeCAD/FreeCAD #8116: Illegal storage access! in Part.makeRuledSurface

more10, 2023-01-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Swedish/Sweden (sv_SE)
[/code]
```

### Subproject(s) affected?

None

### Issue description

```
    surface = Part.makeRuledSurface(wire.Shape, wire_1.Shape)
<class 'RuntimeError'>: FreeCAD exception thrown (Illegal storage access! Please save your work under a new file name and restart the application!)
```

I am making a ruled surface between two b-splines (Draft.makeBSpline). 

It works in 0.18, but not in 0.19 (and not in 0.20).

I have tried to output a log file, using the --write-log switch, but I was unsuccessful. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-01:
Can you attach an example file please

more10, 2023-01-02:
I will make a simple one. In a few days.

more10, 2023-01-03:
[image]
[image]
[bug.zip](https://github.com/FreeCAD/FreeCAD/files/10340063/bug.zip)

Roy-043, 2023-08-14:
The problem is caused by the Draft code. During the V0.19 dev cycle it was decided to remove calls to `recompute` from the Draft `make` functions. As a programmer you now have to explicitly call `recompute` yourself.

See for example:
https://wiki.freecad.org/Draft_BSpline#Scripting

Closing this.

closed 2023-08-14 by Roy-043