FreeCAD/FreeCAD #8117: [Issue] Detecting contiguous object in Part Design fails if linking object is replaced by a new one

ma261065, 2023-01-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * ExtremeProDark 2.6.4
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

In Part Design, do the following:
1) Create a sketch with a simple rectangle (Call it A)
2) Use the pad tool to make it into a solid.
3) Create a new sketch with a second rectangle (call it B) that is separated from the first (so there are now two non-contiguous objects)
4) Try and use the pad tool to make the second rectangle into a solid. As expected, it fails because there are now multiple solids.
5) Create a new sketch with a third rectangle (call it C) that bridges/joins the first two.
6) Use the pad tool to make C into a solid. 
7) This then allows you to use the pad tool on B to make it into a solid as expected, because there is now one contiguous object.
8) We now have a solid consisting of A+B+C.
8) Create a new sketch with a fourth rectangle (call it D) that also joins the original two rectangles (A & B).
9) Use the pad tool to make D into a solid. As expected, we now have four solids (A+B+C+D), all joined together.
10) Delete the third pad & sketch (C), so that the first and second objects (A & B) are now only joined by the fourth rectangle (D).
11) The pad on B now shows an error about multiple solids, even though A & B are joined together by D.
12) The only way to fix this (that I have found) is to delete and re-create the pad on B which then seems to "notice" that D exists.

The expected behaviour would be for any subsequent "bridges" which join two non-contiguous parts to be taken into account if the original "bridge" is deleted.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-03:
Please add an example file for testers to quickly recreate the issue, TIA

FlachyJoe, 2023-01-03:
Hi,
A PartDesign feature depends on creation order, this is not an issue.
You can solve your problem by moving up the D pad in the tree.

closed 2023-01-06 by luzpaz