FreeCAD/FreeCAD #8118: [Issue] visual bug in some std base menu buttons

HeronFlight, 2023-01-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=650783#p650783

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31432 (Git)
Build type: Release
Branch: master
Hash: 63a2855751d638bf27ed969dc60654bda2b54baf
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: Spanish/Spain (es_ES) ]
Installed mods: 
  * Assembly4 0.12.4
  * BIM 2021.12.0
  * Curves 0.6.1
  * DynamicData 2.46.0
  * fasteners 0.4.24
  * fcgear 1.0.0
  * Help 1.0.3
  * InvGears 0.1.2
  * lattice2 1.0.0
  * Manipulator 1.4.9
  * ratchetWB 0.1.0
  * Render 2022.2.0
  * sheetmetal 0.2.59
  * Woodworking 0.21.30486
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

[image]

The buttons marked in the image above, when clicking on them turn highlighted and stays in that way until clicked on them again.
The all buttons functionality is ok, each press triggers the command, it's just a minor visual inconvenience.

The "Tree view actions" button seems to work fine, but the button's highlighting doesn't actually correspond to the actions being checked on/off.

The SelBoundingBox button [image] has the same behaviour, but in this case, everything is OK.

In FreeCAD 0.20.2 all works fine.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

HeronFlight, 2023-01-11:
hey @luzpaz, thanks for your corrections.

luzpaz, 2023-09-17:
@HeronFlight can you also test this on the latest development release 0.22 ?

HeronFlight, 2023-09-18:
> can you also test this on the latest development release 0.22 ?
Hello @luzpaz, I have just tested on the latest weekly build, the issue is still here.
`
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34310 (Git)
Build type: Release
Branch: master
Hash: 3e789dfc85c84e5e6295df908795db5830d28d43
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US) [ OS: Spanish/Spain (es_ES) ]
Installed mods: 
  * Assembly4 0.50.2
  * BIM 2021.12.0
  * Curves 0.6.13
  * DynamicData 2.46.0
  * ExtremeProDark 2.9.0
  * fasteners 0.4.65
  * freecad.gears 1.0.0
  * Help 1.0.3
  * InvGears 0.1.2
  * kicadStepUpMod 10.19.4
  * lattice2 1.0.0
  * Manipulator 1.5.0
  * PartCentroid 0.2.0
  * ratchetWB 0.1.0
  * Render 2023.8.18
  * sheetmetal 0.2.63
  * Wacky
  `

luzpaz, 2025-01-16:
@HeronFlight any chance you could reproduce on 1.0.0 ?

closed 2025-06-09 by github-actions[bot]