FreeCAD/FreeCAD #8126: [Issue] Addon Manager - Connection Failed / Unable to read data from GitHub...

sam1am, 2023-01-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?f=3&t=74990&p=651243&hilit=connection+failed&sid=f7710cc5821c918e569a396711991997#p651243

### Version

0.20 (Release)

### Full version info

```shell
0.20.2 
29177 +426 (Git)
2022/12/07 02:35:37
```

### Subproject(s) affected?

Addon Manager

### Issue description

Addon Manager fails to start with error: 

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sam1am, 2023-01-04:
This is a known issue listed here: https://forum.freecadweb.org/viewtopic.php?f=3&t=30573 

However, the instructions do not provide a workaround. I ran the test suggested and was able to get a result, yet the issue persists. 

[image]

FEA-eng, 2023-01-04:
It occurs in 0.20.2, I never had this issue in 0.20.1 so some change in the source code must be causing it.

chennes, 2023-01-04:
Two things: first, please make sure you are posting your full version information here (including OS information). Otherwise it's hard to know what these installs all have in common. Second, can you please look at the Addon Manager settings in the FreeCAD Preferences page and ensure that the proxy did not get messed up in the upgrade?

donovaly, 2023-01-04:
I fixed the issue in the Windows installer 2 weeks ago. Please download the latest installer and try again:
https://github.com/FreeCAD/FreeCAD/releases/download/0.20.2/FreeCAD-0.20.2-WIN-x64-installer-1.exe

LVAero, 2023-01-05:
@donovaly I tried your download; no resolution. Here is the current setup of what I just downloaded from you:

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

Roy-043, 2023-01-05:
> ... Second, can you please look at the Addon Manager settings in the FreeCAD Preferences page and ensure that the proxy did not get messed up in the upgrade?

I have two 0.20.2 versions (using the same preferences) and only one shows this problem:

The issue occurs with this version (FreeCAD-0.20.2-WIN-x64-installer-1.exe):
```
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

The issue does not occur with this version (note: same hash):
```
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

ningjiazun, 2023-01-05:
The issue occurs with this version (FreeCAD-0.20.2-WIN-x64-installer-1.exe):
[code]
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Chinese/China (zh_CN)
Installed mods: 
  * freecad.gears 1.0.0
[/code]

tankmayvin, 2023-01-05:
I can also confirm getting this issue installing freecad on a system that has never seen the software before (installed just today). I tried both the build currently hosted on the website, and the one linked in this thread:

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

I am running 21H2 19044.2364

I have only Python 3.1 installed (Anaconda3)..

The system does not use a proxy server, and the OS is set to "automatically detect settings". I've tried both "no proxy" and "user system proxy" and both throw up the same issue.

davthomaspilot, 2023-01-06:
Same issue, tried the website and the one linked in this post.

Is there a work-around?  Maybe install an earlier version?

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

chennes, 2023-01-06:
@donovaly what was the fix? Was this about which version of OpenSSL the Windows installer used? Where is that installed?

chennes, 2023-01-06:
Out of curiosity: is anyone experiencing this on a totally new system, that has never had FreeCAD before? Or are you all upgrading from a previous installation? If a previous installation, was it the previous 0.20.2 installer, or something else?

dubbySwords, 2023-01-06:
I upgraded from 0.19 and having same problem.

tankmayvin, 2023-01-06:
> Out of curiosity: is anyone experiencing this on a totally new system, that has never had FreeCAD before? Or are you all upgrading from a previous installation? If a previous installation, was it the previous 0.20.2 installer, or something else?

For me this issue appeared on a totally new system using the 0.20.2 installer current hosted on the main website. I then unistalled/re-installed 0.20.2 using the version @donovaly mentioned was fixed, but the issue remained.

bmw-martin, 2023-01-07:
hallo zusammen hab genau das selbe problem 
hab freecad 0.20.2 gestern das erstemal instaliert über die hauptseite 
hat schon jemand eine lösung für das problem herausgefunden lg martin

donovaly, 2023-01-07:
I made a new installer hoping it fixes the issue. Can you please test it and report back:
https://github.com/FreeCAD/FreeCAD/releases/download/0.20.2/FreeCAD-0.20.2-WIN-x64-installer-2.exe

PerpetualMotivation, 2023-01-07:
I had the same issue. This system did not have FreeCAD previously, just installed a few days ago.
Uninstalled:
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
Installed (from @donovaly's latest) :
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

Add-on manager now loads but returns these errors:
10:31:45  Failed to execute Git Python command: check installation of GitPython and/or git
10:32:26  Traceback (most recent call last):
10:32:26    File "C:\Program Files\FreeCAD 0.20\Mod\AddonManager\addonmanager_workers.py", line 916, in run
10:32:26      self.macro.fill_details_from_wiki(url)
10:32:26    File "C:\Program Files\FreeCAD 0.20\Mod\AddonManager\addonmanager_macro.py", line 315, in fill_details_from_wiki
10:32:26      self.clean_icon()
10:32:26    File "C:\Program Files\FreeCAD 0.20\Mod\AddonManager\addonmanager_macro.py", line 341, in clean_icon
10:32:26      with open(constructed_name, "wb") as f:
10:32:26  OSError: [Errno 22] Invalid argument: 'C:\\Users\\USER\\AppData\\Local\\Temp\\FreeCAD\\Cache\\AddonManager\\MacroIcons\\120px-FCBmpImportScreenshot.png?20210923163201'
10:32:33  Syntax error while reading __files__ from macro AeroFoil
10:32:33  __Files__         = "AeroFoil_UI_Files/AeroFoil_Initial_Dialog.ui, AeroFoil_UI_Files/AeroFoil_NACA4Digit_Dialog.ui, \

10:32:38  Traceback (most recent call last):
10:32:38    File "C:\Program Files\FreeCAD 0.20\Mod\AddonManager\addonmanager_workers.py", line 916, in run
10:32:38      self.macro.fill_details_from_wiki(url)
10:32:38    File "C:\Program Files\FreeCAD 0.20\Mod\AddonManager\addonmanager_macro.py", line 315, in fill_details_from_wiki
10:32:38      self.clean_icon()
10:32:38    File "C:\Program Files\FreeCAD 0.20\Mod\AddonManager\addonmanager_macro.py", line 341, in clean_icon
10:32:38      with open(constructed_name, "wb") as f:
10:32:38  OSError: [Errno 22] Invalid argument: 'C:\\Users\\USER\\AppData\\Local\\Temp\\FreeCAD\\Cache\\AddonManager\\MacroIcons\\Macro_makeCube.png?20110801163602'
10:32:39  Traceback (most recent call last):
10:32:39    File "C:\Program Files\FreeCAD 0.20\Mod\AddonManager\addonmanager_workers.py", line 916, in run
10:32:39      self.macro.fill_details_from_wiki(url)
10:32:39    File "C:\Program Files\FreeCAD 0.20\Mod\AddonManager\addonmanager_macro.py", line 315, in fill_details_from_wiki
10:32:39      self.clean_icon()
10:32:39    File "C:\Program Files\FreeCAD 0.20\Mod\AddonManager\addonmanager_macro.py", line 341, in clean_icon
10:32:39      with open(constructed_name, "wb") as f:
10:32:39  OSError: [Errno 22] Invalid argument: 'C:\\Users\\USER\\AppData\\Local\\Temp\\FreeCAD\\Cache\\AddonManager\\MacroIcons\\Macro_Site_From_Contours.png?20190704172250'
10:32:41  Traceback (most recent call last):
10:32:41    File "C:\Program Files\FreeCAD 0.20\Mod\AddonManager\addonmanager_workers.py", line 916, in run
10:32:41      self.macro.fill_details_from_wiki(url)
10:32:41    File "C:\Program Files\FreeCAD 0.20\Mod\AddonManager\addonmanager_macro.py", line 315, in fill_details_from_wiki
10:32:41      self.clean_icon()
10:32:41    File "C:\Program Files\FreeCAD 0.20\Mod\AddonManager\addonmanager_macro.py", line 341, in clean_icon
10:32:41      with open(constructed_name, "wb") as f:
10:32:41  OSError: [Errno 22] Invalid argument: 'C:\\Users\\USER\\AppData\\Local\\Temp\\FreeCAD\\Cache\\AddonManager\\MacroIcons\\Centericon.png?20210619192700'

donovaly, 2023-01-07:
The error can be ignored. The addon manager tries to load images from the Wiki and for 3 images this fails. Besides this the addon manager should works now for you as expected, right?

PerpetualMotivation, 2023-01-07:
Yes I just finished testing that. Thank you!
Also installed the latest versions of git and python and that first error went away

donovaly, 2023-01-07:
Thanks for testing. closing as fixed.

closed 2023-01-07 by donovaly

Intuos5, 2023-01-25:
@donovaly I ran into this today in the latest dev build:  FreeCAD-0.21.0dev-31391 from 21 december. Could the dev build be updated?

`[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31391 (Git)
Build type: Release
Branch: master
Hash: 0ab2608a429f3a5ee9a63da9b66d13557655d7e4
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Launcher
[/code]
`

donovaly, 2023-01-25:
I plan to make a new build this weekend.

Intuos5, 2023-01-25:
Perfect!
As a workaround: I could also use 0.20.2's addon manager to get an addon in the development build (without manually fiddling with files). Might be worth it for people who also face this same issue still.

donovaly, 2023-02-07:
> I plan to make a new build this weekend.

So just 2.5 weeks later ;-), here it is: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Intuos5, 2023-02-07:
No worries, just confirming the addon manager is working again!

wearysapper, 2023-04-06:
Sorry for messing with the closing .....
I had this problem on Debian_Linux with 20.1 (BULLSEYE)  and again still now wit 20.2 (BOOKWORM  deb) updating  python3-git and git to the last package:  nope.
It is quite annoying because also the help system comes trough the addon-manager.
Some suggestion on how to investigate the behavior of the call to github?
I cannot figure how this simple step could be so unstable depending on build.
Thanks in advance for the attention
David

wearysapper, 2023-04-06:
Hi, I beg your pardon for the previuos post,
I went digging on the src tree for the addon manager and realized that  is definetly a no so simple task ...
But the strange thing is that it seems not able to connect with the server.
Again thanks to all people working on this huge program
d.

chennes, 2023-04-06:
This error message is generic: there are many reasons you might receive it, and there is not a particular "bug" to fix. Rather, we need to troubleshoot your connection. I suggest posting on the Forum, the Facebook group, or on Reddit to try to get help with that.

wearysapper, 2023-04-06:
Thank you chennes,
seems you're  right. I tried the last AppImage build too, but with the same report result:
"Addon Manager: Risposta 0 inattesa dal server" (replay 0 unexpected from server) .
.I'll tranfer the question on the forum. 
Anyway, on the programming side (being my connection without other manifest problems), the connection chain less weak and more flexibile would be suitable (i posted here because I saw this issue was not only mine).
Again, thanks for the answer
david

wearysapper, 2023-04-06:
For people with  problem like mine (on Debian Bullseye/Bookworm):
not connecting to github  -  "Addon Manager: Unexpected 0 response from server"
Installing   pyside2-tools    solved for me the problem.

wearysapper, 2023-04-06:
A precisation: solved only for the AppImage build not for deb pachage  ...

Mannshoch, 2023-11-23:
I have this Dialog on Windows.
If I install FreeCAD, after Installing the Addon Manager only works on the first FreeCAD Start.
I tested several cases:
Install -> Start FreeCAD -> Open Addon Manager -> Close and Reopen FreeCAD -> AddonManager not work
Install -> Start FreeCAD -> Close and Reopen FreeCAD -> AddonManager not work

```
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Switzerland (de_CH)
```

MantaRayDeeJay, 2023-11-27:
on 2023-10-29 :
- I met this issue with the last FreeCAD release FC v0.21.1.
- by using FC v0.21.0 I didn't meet this issue.
I thought it was a new bug with the last version.

on 2023-11-27 :
- I met this issue **both** with the FC v0.21.0 and FC v0.21.1 releases.
- by using FC v0.20.2 I didn't meet this issue.

I work on Windows 10 platform.

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: French/France (fr_FR)

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: French/France (fr_FR)
```

medisoft, 2024-02-01:
On Ubuntu 22.04, and Freecad 0.21.2 (latest stable), got the same error, but in the console found these errors:
`FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

connect failed: No such file or directory
The `.FCStd#` backup format is deprecated as of v0.21 and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
(qt.network.ssl) QSslSocket: cannot resolve OPENSSL_init_ssl
(qt.network.ssl) QSslSocket: cannot resolve OPENSSL_init_crypto
(qt.network.ssl) QSslSocket: cannot resolve ASN1_STRING_get0_data
(qt.network.ssl) QSslSocket: cannot resolve EVP_CIPHER_CTX_reset
(qt.network.ssl) QSslSocket: cannot resolve EVP_PKEY_up_ref
(qt.network.ssl) QSslSocket: cannot resolve EVP_PKEY_param_check
(qt.network.ssl) QSslSocket: cannot resolve RSA_bits
(qt.network.ssl) QSslSocket: cannot resolve OPENSSL_sk_new_null
(qt.network.ssl) QSslSocket: cannot resolve OPENSSL_sk_push
(qt.network.ssl) QSslSocket: cannot resolve OPENSSL_sk_free
(qt.network.ssl) QSslSocket: cannot resolve OPENSSL_sk_num
(qt.network.ssl) QSslSocket: cannot resolve OPENSSL_sk_pop_free
(qt.network.ssl) QSslSocket: cannot resolve OPENSSL_sk_value
(qt.network.ssl) QSslSocket: cannot resolve DH_get0_pqg
(qt.network.ssl) QSslSocket: cannot resolve SSL_CTX_set_options
(qt.network.ssl) QSslSocket: cannot resolve SSL_CTX_get_security_level
(qt.network.ssl) QSslSocket: cannot resolve SSL_CTX_set_security_level
(qt.network.ssl) QSslSocket: cannot resolve SSL_CTX_set_ciphersuites
(qt.network.ssl) QSslSocket: cannot resolve SSL_set_psk_use_session_callback
(qt.network.ssl) QSslSocket: cannot resolve SSL_SESSION_is_resumable
(qt.network.ssl) QSslSocket: cannot resolve SSL_get_client_random
(qt.network.ssl) QSslSocket: cannot resolve SSL_SESSION_get_master_key
(qt.network.ssl) QSslSocket: cannot resolve SSL_session_reused
(qt.network.ssl) QSslSocket: cannot resolve SSL_set_options
(qt.network.ssl) QSslSocket: cannot resolve TLS_method
(qt.network.ssl) QSslSocket: cannot resolve TLS_client_method
(qt.network.ssl) QSslSocket: cannot resolve TLS_server_method
(qt.network.ssl) QSslSocket: cannot resolve X509_up_ref
(qt.network.ssl) QSslSocket: cannot resolve X509_STORE_CTX_get0_chain
(qt.network.ssl) QSslSocket: cannot resolve X509_getm_notBefore
(qt.network.ssl) QSslSocket: cannot resolve X509_getm_notAfter
(qt.network.ssl) QSslSocket: cannot resolve X509_get_version
(qt.network.ssl) QSslSocket: cannot resolve X509_STORE_set_ex_data
(qt.network.ssl) QSslSocket: cannot resolve X509_STORE_get_ex_data
(qt.network.ssl) QSslSocket: cannot resolve OpenSSL_version_num
(qt.network.ssl) QSslSocket: cannot resolve OpenSSL_version
(qt.network.ssl) Incompatible version of OpenSSL
(qt.network.ssl) QSslSocket::connectToHostEncrypted: TLS initialization failed
Addon Manager: Unexpected 0 response from server`

Upgraded the libssl-dev with
`sudo apt install libssl-dev`

That installed libssl-dev_3.0.2, and after that, the addon manager worked

d4mer, 2024-02-05:
Hi All, 

Same issue on macOS

[code]
OS: macOS 14.0
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.9.22
  * InventorLoader 1.4.0
  * parts_library
  * Render 2024.1.23
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * kicadStepUpMod 10.23.0
  * AirPlaneDesign 0.4.0
  * sheetmetal 0.4.0
  * MakerWorkbench 1.0.1
[/code]