FreeCAD/FreeCAD #8127: [Issue] Printing changes original dimensions

mjapps, 2023-01-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
I sent the model to a printer, directly from FreeCAD and it changed/increased dimensions a lot (something like when the "fit to page" option is somewhere selected).

Alternatively, I exported the model to a PDF and sent that PDF to the printer. The result was much better, but the dimensions are still different from the original. While it is known to me from some other software that exporting to a PDF and printing alters the dimensions, printing directly from the application usually works well, but it was not the case here.
```

### Subproject(s) affected?

Techdraw

### Issue description

I drew just a simple layout of a PCB with a patch on top, a substrate below it and the ground plane at the bottom.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-05:
Edit the above to add your full [About info](https://www.freecadweb.org/About) :point_up:

WandererFan, 2023-01-05:
Can we get a file that displays the problem?

mjapps, 2023-01-05:
Hi!

Sure, the file is attached. Thanks for responding.

Regards,
Miroslav Joler

WandererFan, 2023-01-06:
<<Sure, the file is attached. Thanks for responding.>>

Don't see the file.  Is it attached somewhere else?

mjapps, 2023-01-06:
Sorry, I must have missed it.
How about now ?
Thanks!

WandererFan, 2023-01-07:
I still don't see the file.

mjapps, 2023-01-07:
Sorry abou. Yesterday, I attached it for sure. I guess OneDrive messes with
me because that's where I had the file stored and attached from.
Now I copied it into a local folder on my drive and hope this time it will
work.
Again, sorry about the inconvenience. It appeared to me the file was
attached each time.
Let me just mention that I prepared and printed out the same layout within
LibreCAD and the dimensions printed out faithfull there. It is just for a
comparison, but I would like to know that it also works in FreeCAD.
I really hope this time the file will be delivered to you. Here is the
screenshot of it being attached.
[image: image.png]

Thanks!
Miroslav Joler

luzpaz, 2023-01-07:
To attach file to issue:. 
1. Click the link to GitHub.
2. Zip file locally.  
3. Drag and drop zip file in to comment box

mjapps, 2023-01-07:
[design.zip](https://github.com/FreeCAD/FreeCAD/files/10367374/design.zip)

Thanks for the instructions. I hope I did it right.

luzpaz, 2023-01-07:
Please also add your full [About](https://wiki.freecad.org/About) info

mjapps, 2023-01-08:
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners
[/code]

WandererFan, 2023-01-08:
design.FCStd doesn't have any TechDraw pages in it.  Are you printing a TD Page or are you printing from the 3d window?

mjapps, 2023-01-08:
Before I actually reply, it is somewhat confusing to read that you say the FCStd file *doesn't have an TechDraw pages in it,* then to ask me if I was printing a TD page?!
Anyhow, upon opening the new FreeCAD file, I started the TechDraw option and drew that simple layout on the initial canvas that was there. Then I chose File>Print and that was it. I was looking for an option to adjust the scale of the layout, but I didn't find anything but zoom in and out. Although I am a novice to FreeCAD, I have enough general experience with all sorts of other software and I believe I would have found my way out had there been some intuitive and obvious option on how to adjust the view to be printed out. Now, what is a TD page and what is a 3D page... I am quite unfamiliar with it. I just know the GUI let me model that layout on that initial canvas and I naturally expected it to let me print it out keeping the original dimensions.
Now, if you can figure out or give some effective advice how to mitigate it, I will appreciate it. If you think everything is fine with the printing feature, ok. Thanks!

WandererFan, 2023-01-08:
With a little luck, here are some pictures of 3d window vs TechDraw Page.
First the 3d window which you are familiar with...
[image]

... and a TechDraw Page: 
[image]

WandererFan, 2023-01-08:
If I press print while in the 3d window I get something that is of no particular scale.  If I make a TD Page and print it, the image is correctly sized within the limits of my eyesight.

WandererFan, 2023-01-10:
AFAICT, this is not about TechDraw, but about trying to print the contents of the 3d window in scale.  I would think the print size depends on the zoom level, not the size of the objects in the model.

Is any promise made about the size of a print from the 3d window?  If not, then this is not a bug.

luzpaz, 2023-01-15:
> Is any promise made about the size of a print from the 3d window? If not, then this is not a bug.

@WandererFan  a possible solution is that we grey out 'Print' in the drop down menu so users aren't confused ?

WandererFan, 2023-01-15:
> > Is any promise made about the size of a print from the 3d window? If not, then this is not a bug.
> 
> @WandererFan a possible solution is that we grey out 'Print' in the drop down menu so users aren't confused ?

There is nothing wrong with printing the 3d window, so I don't think we should prevent it.   

The only issue is that the OP was expecting a 1:1 print from the 3d window and didn't get it.    I added a note to the wiki for Std_Print to say that no particular scale is guaranteed.

luzpaz, 2023-01-15:
@WandererFan thanks for the clarification. Shall we close this ticket thne?

WandererFan, 2023-01-16:
Not a bug. Closing.

closed 2023-01-16 by WandererFan